analogdevicesinc/meta-adi #12: Make yocto generic

ckoehler, 2019-04-16:
Could you make a recipe for your kernel, similar to linux-xlnx? Maybe linux-adi? That way I could just include yours. As it is, I cannot  include your meta-adi-xilinx layer because I don't use petalinux.

nunojsa, 2019-04-17:
Hi,

If you are not using petalinux, it does not make much sense for now in using this layer. It was only tested under petalinux.
What are you trying to accomplish here? Are you using [https://github.com/Xilinx/meta-xilinx](meta-xilinx) at all?
Regarding your request, it is something that we will be adding eventually...

ckoehler, 2019-04-17:
I am using meta-xilinx, yes. I am creating a BSP and image for a Zynq board + custom daughterboard, and want to decouple Xilinx tools from the embedded side, so that our firmware guys don't have to worry about it.
So I can use meta-xilinx just fine in Yocto without petalinux, and would like to do the same with yours.

nunojsa, 2019-04-17:
To be honest, this layer was not tested at all using plain yocto. Eventually, we want to add a linux-adi recipe for the kernel but for now maybe you can leave with this. Could you tell me which issues do you have when you include this layer? Are you using the machine configuration files from meta-xilinx or are you redefining them? Also note, that our linux-xlnx.bbappend recipe is just enforcing that our adi kernel is used... So at the end you are not using Xilinx kernel...

ckoehler, 2019-04-17:
Since your layer overrides the petalinux recipe (with a bbappend), and I don't have that recipe (because I don't include that layer from meta-xilinx), it errors out.

nunojsa, 2019-05-23:
I finally got the time to add a linux adi recipe to include ADI kernel on plain yocto builds. Most likely you already moved on with a fix but you can take the new meta-adi-bsp layer into consideration for the future...

closed 2019-05-23 by nunojsa