analogdevicesinc/meta-adi #14: ZC706 ad9361Unable to parse input tree

gtarim, 2019-04-17:
Hi ,

I have an issue that i followed your instructure at https://github.com/analogdevicesinc/meta-adi/tree/master/meta-adi-xilinx

and when i started my petalinux-build after a long building time. petalinux returning a problem that contains lots of red texts are at below.

| Error: /home/Workspace/PetalinuxWorkspace/xilinx_templates/peta_zc706_bsp/petalinux/fmcomms/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/device-tree/xilinx+gitAUTOINC+b7466bbeee-r0/system-user.dtsi:1.1-3 syntax error
| FATAL ERROR: Unable to parse input tree

when i am going to open that file (system-user.dtsi) i saw my selected board KERNEL_DTB, and i had remove that text at top of the file it is going well. 
(-e /include/ "pl-delete-nodes-zynq-zc706-adv7511-ad9361-fmcomms2-3.dtsi" after remove )

i couldnt understand the problem i think this line has problem to resolve from petalinux do you have an idea about this problem.

Thanks

Gokhan

nunojsa, 2019-04-17:
Hi,

I just tried to reproduce your issue and it worked fine for me...
I'm not sure if I understood you correctly, did it worked after you removed the `/include/ pl-delete-nodes-zynq-zc706-adv7511-ad9361-fmcomms2-3.dtsi` in the system_user.dtsi file? Is it possible for you to show me the state of your system_user.dtsi and   pl-delete-nodes-zynq-zc706-adv7511-ad9361-fmcomms2-3.dtsi?

Thanks!

closed 2019-05-20 by nunojsa

bkzshabbaz, 2019-06-16:
I'm experiencing the exact same issue.

nunojsa, 2019-06-28:
Hi,

I'm still unable to reproduce this on my machine. For me it works just fine. Note, that this layer was only tested against Petalinux-v2018.3. You should not have to edit system-user.dtsi. The only file that should be edited is device-tree.bbappend to select the proper devicetree.

bkzshabbaz, 2019-06-28:
Can this issue be re-opened?  I tried the build on a fresh Ubuntu 16.04.06 laptop and it's the exact same behavior.  I changed the device-tree.bbappend file to specify the zc706 "zynq-zc706-adv7511-ad9361-fmcomms2-3".  I am using Petalinux-v2018.3.
What are we doing wrong?

bkzshabbaz, 2019-06-29:
Found the problem.  I was using dash instead of bash.  '$ sudo dpkg-reconfigure dash' fixed the issue.

nunojsa, 2019-06-29:
NIce. That explains it

thecodeforzj, 2020-07-20:
I'm experiencing the exact same issue. I use it '$ sudo dpkg-reconfigure dash' , but it wasn't sliven it.  I am using Petalinux-v2018.3

nunojsa, 2020-07-20:
Petalinux-v2018.3 is supported on the 2019_R1 branch. Is that the one you are using? If so, make sure you have a proper environment to build petalinux. Alternatively, you can use the petalinux dockerfile to build your project...