analogdevicesinc/meta-adi #26: ERROR: Parse failure with the specified layer added

bkzshabbaz, 2019-08-19:
I'm trying to build an image offline using the BB_NO_NETWORK option.  After I add the meta-adi layers to Yocto Settings and check 'Enable BB NO NETWORK', then save the config, I get the following output:
```
INFO: Getting hardware description...
INFO: Rename system_top.hdf to system.hdf
[INFO] generating Kconfig for project
[INFO] menuconfig project
configuration written to /home/sl31870/zc706_sdr/project-spec/configs/config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

[INFO] sourcing bitbake
[INFO] generating plnxtool conf
[INFO] generating meta-plnx-generated layer
[INFO] generating machine configuration
[INFO] generating bbappends for project . This may take time ! 
ERROR: Parse failure with the specified layer added
[INFO] generating u-boot configuration files
[INFO] generating kernel configuration files
[INFO] generating kconfig for Rootfs
[INFO] oldconfig rootfs
[INFO] generating petalinux-user-image.bb
```
The error does not appear when BB NO NETWORK is disabled.

nunojsa, 2019-08-19:
This layer was never tested with `BB_NO_NETWORK`, so this might be possible. I will have to do some checking when i have the time.
Thanks for reporting this!

bkzshabbaz, 2019-08-30:
Here's the verbose output:
```
INFO: Getting hardware description...
INFO: Rename system_top.hdf to system.hdf
[INFO] generating Kconfig for project
INFO: [Hsi 55-1698] elapsed time for repository loading 0 seconds
[INFO] menuconfig project
configuration written to /home/sl31870/zc706_sdr/project-spec/configs/config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

[INFO] sourcing bitbake
[INFO] generating plnxtool conf
[INFO] generating meta-plnx-generated layer
[INFO] generating machine configuration
[INFO] generating bbappends for project . This may take time ! 
NOTE: Starting bitbake server...
Parsing recipes...ERROR: ExpansionError during parsing /home/sl31870/projects/meta-adi/meta-adi-core/recipes-core/libad9361-iio/libad9361-iio_0.2.bb
Traceback (most recent call last):
  File "/opt/petalinux-v2018.3/components/yocto/source/arm/layers/core/bitbake/lib/bb/data_smart.py", line 412, in DataSmart.expandWithRefs(s="${@bb.parse.BBHandler.vars_from_file(d.getVar('FILE', False),d)[1] or '1.0'}+git${SRCPV}", varname='PV'):
                 try:
    >                s = __expand_var_regexp__.sub(varparse.var_sub, s)
                     try:
  File "/opt/petalinux-v2018.3/components/yocto/source/arm/layers/core/bitbake/lib/bb/data_smart.py", line 111, in VariableParse.var_sub(match=<_sre.SRE_Match object; span=(80, 88), match='${SRCPV}'>):
                 else:
    >                var = self.d.getVarFlag(key, "_content")
                 self.references.add(key)
  File "/opt/petalinux-v2018.3/components/yocto/source/arm/layers/core/bitbake/lib/bb/data_smart.py", line 794, in DataSmart.getVarFlag(var='SRCPV', flag='_content', expand=True, noweakdefault=False, parsing=False):
                     cachename = var + "[" + flag + "]"
    >            value = self.expand(value, cachename)
     
  File "/opt/petalinux-v2018.3/components/yocto/source/arm/layers/core/bitbake/lib/bb/data_smart.py", line 436, in DataSmart.expand(s='${@bb.fetch2.get_srcrev(d)}', varname='SRCPV'):
         def expand(self, s, varname = None):
    >        return self.expandWithRefs(s, varname).value
     
  File "/opt/petalinux-v2018.3/components/yocto/source/arm/layers/core/bitbake/lib/bb/data_smart.py", line 426, in DataSmart.expandWithRefs(s='${@bb.fetch2.get_srcrev(d)}', varname='SRCPV'):
                 except Exception as exc:
    >                raise ExpansionError(varname, s, exc) from exc
     
bb.data_smart.ExpansionError: Failure expanding variable SRCPV, expression was ${@bb.fetch2.get_srcrev(d)} which triggered exception NetworkAccess: Network access disabled through BB_NO_NETWORK (or set indirectly due to use of BB_FETCH_PREMIRRORONLY) but access requested with command git -c core.fsyncobjectfiles=0 ls-remote https://github.com/analogdevicesinc/libad9361-iio.git  (for url https://github.com/analogdevicesinc/libad9361-iio.git)
```

nunojsa, 2019-09-30:
Hi,

This is expected as our recipes were not developed with this offline builds in mind. So things like `SRCREV = "${AUTOREV}"` or `PV_append = "+git${SRCPV}"`. As for now, we are not adding support for offline builds so I'm closing this issue.

This might change in the future, if we start to see that this is something that a lot of people want...

closed 2019-09-30 by nunojsa