analogdevicesinc/meta-adi #32: 'no serial driver found' after fresh build

bkzshabbaz, 2019-10-24:
Hi, I built a fresh petalinux image targeting the ZC706/FMComms3-EBZ and when I boot the SD card, I get the following at the terminal:

`'No serial driver found
resetting ...'`

alexarsenault, 2019-11-14:
I have also experienced this exact issue.  Hoping to hear of some resolution for this soon.

nunojsa, 2019-11-14:
Hi guys,

I will have to try this project to see if i get the some. Unfortunately I don't have the time to do it right now. I will give some feedback as soon as I can try this.

ehsaneam, 2019-11-23:
The problem is stdout-path = "serial0:115200ns"; in system-conf.dtsi. There is a typo in *ns*, it was written *n8*.

avionion, 2019-12-05:
I had this issue on ADRV9361 and I was able to resolve it by changing  file contents of system-user.dtsi with:
`/include/ "system-conf.dtsi"
/ {
 aliases {
          serial0 = &uart1;
          ethernet0 = &gem0;
          spi0 = &qspi;
      }; 
     chosen {
          bootargs = "console=ttyPS0,115200 earlyprintk";
          stdout-path = "serial0:115200ns";
     };
};`

nunojsa, 2020-01-20:
So, I could confirm this problem when bringing support for petalinux 2019_r1. This happened for me with zc706/fmcomms2. The problem seems to be with `stdout-path`. Setting it to `serial0:115200n8` fixed the problem for me. It seems the fsbl or u-boot are somehow relying on this property. It's an odd problem since building things manually, without petalinux, worked out of the box without any change to the device-tree...

However, on the 2019_r1 (petalinux 2018_r3) branch everything worked fine...

The fix for the master branch should come in the next days (hopefully) along with support for petalinux 2019_r1 (which will be the supported release on our master branch).

If someone still has this problem in the 2019_R1 or 2018_R2 branch please let me know...

nunojsa, 2020-01-27:
This should be fixed by #54 . Let me know if some other project still has this...

For now, I'm closing this issue...

closed 2020-01-27 by nunojsa