analogdevicesinc/meta-adi #33: Layer adi-xilinx is not compatible

lakshmanrajam, 2019-11-26:
HI 

i am newly installed petalinux 2019.1  i am trying to add yacto User layer in my project 
for that i did like this 

> a configuration menu will come up. Go to Yocto Settings->User layers and add the <path-to-meta-adi>/meta-adi-xilinx and <path-to-meta-adi>/meta-adi-core

previously we are using 2018.3 petalinux its working fine  with met-adi but 2019.1 we facing some problems

>  ERROR: Unable to start bitbake server
ERROR: Server log for this session (/home/fpga/workspace/adrv/firinter1/build/bitbake-cookerdaemon.log):
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with rocko)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with rocko)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with rocko)

what might be the issue ??

how can i resolve this issue?
appreciate your help.

Thanks.

commodo, 2019-11-26:
Yocto doesn't work with this layer yet.

Only Petalinux.
We will fix the Petalinux build.

lakshmanrajam, 2019-11-26:
Thanks for your reply,
can we get any alternative solution for this?

commodo, 2019-11-26:
For Yocto, not yet.
It's on our todo-list to add Yocto support, but we don't have any resources to allocate for it.

I'm not sure if this layer can be shared between Petalinux & Yocto; in theory it could work. But depending on the complexity [for supporting both], we may create a new layer just for Yocto.

nunojsa, 2019-11-26:
This is expected. Our layer was only tested against petalinux 2018_R3 which is based on yocto rocko version so that, AFAWCT, our layer is only compatible with rocko. [This](https://github.com/analogdevicesinc/meta-adi/blob/b7131fd2c3a9f17b51a43ff13185233ea9688e16/meta-adi-xilinx/conf/layer.conf#L13) is responsible to setup this restrictions. You can try to do `LAYERSERIES_COMPAT_adi-xilinx = "rocko thud"`. This might just work, but be aware that we never tested our layer with 2019.1, so you can find all kind of build errors. In the case of errors, you are on your own for now. As @commodo said, we don't have the resources now for this and most likely, we will address this only next year...

noelpedro, 2020-01-09:
Hi @lakshmanrajam, How did you get it to work with petalinux 2018.3??

Im currently getting checksum errors. 

I'm running ubuntu 18.04.

nunojsa, 2020-01-10:
Checksum errors? Can you elaborate on that? Also note that I'm currently doing some re-work on our layers. In the next days, I hope to have some updates. Also note, that petalinux 2019.r1 is still not supported. I expect to have it in the next weeks...

lakshmanrajam, 2020-01-10:
> Hi @lakshmanrajam, How did you get it to work with petalinux 2018.3??
> 
> Im currently getting checksum errors.
> 
> I'm running ubuntu 18.04.

hi @noelpedro 
if You want to run with petalinux 2018.3 like me . i dont know my solution is correct or not but it is working fine now.

 resolved this issue by using 2019_R1 branch

DESCRIPTION = "ADI kernel"
BRANCH = "2019_R1"
always use latest source revision
SRCREV = "${AUTOREV}"
SRC_URI += "git://github.com/analogdevicesinc/linux.git;protocol=https;branch=${BRANCH}"
KBRANCH = "${BRANCH}"
PV = "4.14"

but i got error

FetchError: Fetcher failure: Unable to resolve '2019_R1' in upstream git repository in git ls-remote output for github.com/Xilinx/linux-xlnx.git

for this i did one modification in petalinux installed directory

i changed xilinx git hub to analog git in linux-xlnx.inc file

#KERNELURI ??= "git://github.com/Xilinx/linux-xlnx.git;protocol=https"
KERNELURI ??= "git://github.com/analogdevicesinc/linux.git;protocol=https"

note: dont try this branch 2018_R2 we are unable to up the adrv9009 in zcu102 board.

closed 2020-01-10 by lakshmanrajam