analogdevicesinc/meta-adi #42: system.dtb ERROR

noelpedro, 2020-01-10:
Hi all,
Im getting build errors. 
My target machine is zynqmp-zcu102-rev10-ad9361-fmcomms2-3"
petalinux verison: 2018.3
ubuntu 18.04
 Before this error system-user.dtsi and analog-devices.dtsi both had a strange "-e" appended at the top of the file that caused a syntax error. I deleted the "-e" and got this new error:

| DEBUG: Executing shell function do_compile
| system-top.dtb: ERROR (phandle_references): Reference to non-existent node or label "zynqmp_clk"
|
| system-top.dtb: ERROR (phandle_references): Reference to non-existent node or label "zynqmp_clk"
|
| ERROR: Input tree has errors, aborting (use -f to force output)
| WARNING: exit code 2 from a shell command.
| ERROR: Function failed: do_compile

nunojsa, 2020-01-13:
Hi,

This seems odd as we do not touch that node in our recipe. So, it should be there. The first error already looks strange? Is bash your default shell? We are currently doing some refactor to meta-adi. I might try this project on my side to check if I get the same error...

commodo, 2020-01-13:
The `zynqmp_clk` clock was renamed in 4.19 by Xilinx.
The clock reference used to be named `clk` in version 4.14 and they renamed everything.

This breaks some newer device-trees of ours that have not been adapted to 4.19.

nunojsa, 2020-01-13:
Hmm, I see. Then Im closing this issue since it's a problem in our linux tree rather than in meta-adi.

commodo, 2020-01-13:
> Hmm, I see. Then Im closing this issue since it's a problem in our linux tree rather than in meta-adi.

Well, in this layer/repo, we need to make sure things are aligned properly.
i.e. DTs here should be matched with proper linux & vivado versions;

if that happens, all should be fine;

nunojsa, 2020-01-13:
And they are (i hope). But in the master branch in this repo, what we do is just get our master branches and build things from there. Hence, if our linux master branch is somehow broken for this device-tree, then we need to fix it there. On the other hand, for the 2019_r1 and 2018_r2 branches this project should actually build...

noelpedro, 2020-01-13:
yes bash is my shell.

Should I checkout 2018_r2 to solve this?

nunojsa, 2020-01-14:
> However I still needed to remove the appended "-e" to the files specified in my initial post.

This is odd and it does not happen on my side. I just get the `zynqmp_clk` errors. It seems something is odd with your build environment.

> But I believe its kernel v4.14 :(

Yes, that's true. Unfortunately, there's no other way right now. This needs to be fixed on our kernel master branch which is 4.19...

nunojsa, 2020-01-16:
Well, just some info here. I was actually misunderstanding the problem. The device-tree on our kernel tree is actually fine. The problem is with our petalinux branch that needs to be updated to petalinux 2019_R1 in order to use 4.19 kernel. Hopefully, this will come in the next few days...

nunojsa, 2020-01-27:
This should be fixed by #54

closed 2020-01-27 by nunojsa