analogdevicesinc/meta-adi #46: Using custom device treee

noelpedro, 2020-01-13:
Hello guys I'm using meta-adi to build zyqnqMP zcu102 fmcomms2 Ad 9361 ADI image. However I want to make a simple change to the device tree .i.e I want to change the rate of the external clock from 40MHz to 10MHz of AD9361. How can I ensure that petalinux uses this updated device tree?

nunojsa, 2020-01-14:
Maybe take a look at [Extending the devicetree](https://github.com/analogdevicesinc/meta-adi/tree/master/meta-adi-xilinx#extending-the-devicetree)...

noelpedro, 2020-01-14:
So I can copy the ad-fmcomms2.dtsi onto petalinux's system-user.dtsi then make the changes?

nunojsa, 2020-01-15:
No, you can just overwrite the node you want in `system-user.dtsi`. No need to copy the full file...

nunojsa, 2020-01-22:
Since it seems there's not more questions about this, I'm closing this issue...

closed 2020-01-22 by nunojsa

noelpedro, 2020-01-23:
Big thanks