analogdevicesinc/meta-adi #51: Getting error while building from file pl-delete-node-.

aeonjose, 2020-01-16:
I am using adrv9361-z7035 in petalinux 2018.3 .While building the project error comes from the file pl-delete-nodes-zynq-adrv9361-z7035.dtsi .

Exactly why is this file used for? 

[image]

nunojsa, 2020-01-16:
Which branch are you using? For petalinux 2018.3 you should use 2019_R1. Im currently working in bringing support to petalinux 2019.1 in our master branch.

We still don't have a proper CI in this repo, so some projects can break as it seems to be your case...

> Exactly why is this file used for?
Petalinux will automatically generate a pl.dtsi file which contains device-tree nodes of the hdl project. In the past, we faced a lot of issues with this auto-generation thing. Moreover, we want our linux drivers to handle this devices. Hence, we want to include the device-trees defines in our kernel tree in the petalinux build. For that, we need to delete the nodes that are generated in pl.dtsi otherwise we would get a compile failure on the device-tree...

For your case, if are not using 2019_R1 branch, please use it for now. If you are already on this branch, you can just go to the pl-delete-nodes-zynq-adrv9361-z7035.dtsi and remove the lines corresponding to the nodes that are failing.

Hopefully, in the next few weeks I will setup some CI that can catch this kind of failure early on...

Hope this helps!

aeonjose, 2020-01-17:
Thank you for your response. Will try it and let you know the result.

Thank you
Aeon Jose

nunojsa, 2020-01-28:
Should be fixed by #57

PR #57 (2020-01-29): Fix 2019_R1 projects
Some projects were removed while others need changes in the
pl-delete-nodes*.

This supersedes #52. It should then fix #51
closed 2020-01-28 by nunojsa