analogdevicesinc/meta-adi #55: Cant use rootFS from SD card

noelpedro, 2020-01-24:
Hi guys  I successfully buillt peta-linux 2018.3 project using meta-adi/oring/2018_R2 yocto layers for zynqMP-zcu102-ad9361-fmcomms2-3.

I have two issues. I initiailly tried to use IntraramFS for root filesystem. However i couldnt login as root even though I entered the correct password. Per $;petalinux-config -c rootfs -> image settings Password.

Because of this I configured the project to use SD card for rootFS i.e. /dev/mmcblk0p2.
That resulted in a kenel panic. 

"VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
Please append a correct "root=" boot option; here are the available partitions:..."

Anyone have success with ZCU102??

nunojsa, 2020-01-25:
> Hi guys I successfully buillt peta-linux 2018.3 project using meta-adi/oring/2018_R2 yocto layers for zynqMP-zcu102-ad9361-fmcomms2-3.
 
You should not be using petalinux 2018.3 with our 2018_R2 branch. If you want to stick with tag petalinux version, you should use the 2019_R1 branch. Alternatively, you can use petalinux 2018.2 with the branch you are currently using.

After that, if you still have issues, I will have to try and reproduce them on my side.

> I have two issues. I initiailly tried to use IntraramFS for root filesystem. However i couldnt login as root even though I entered the correct password. Per $;petalinux-config -c rootfs -> image settings Password.

The password you should use is __analog__. That should work. We are probably overwriting the password to enforce our own pass.

Hope this helps!

closed 2020-02-13 by commodo

noelpedro, 2020-02-15:
I’m also having issues with rootfs o SD on zcu102 with petalinux 2018.3 using just the Xilinx kernel not the meta-adi yocto layer. 

So this issue must be with Xilinx and not with your layer.

💯

nunojsa, 2020-02-17:
Hi,

Not sure this is actually an issue. Just double check your bootargs. Most likely, you are missing the `root=/dev/mmcblk0p2` argument.

noelpedro, 2020-02-19:
Thanks for reply,

I re-imaged my SD card and..... BOOM it mounts the Rootfs on mmcblk0p2.

PR #59 (2020-02-13): device tree recipe refactor
This series add a big refactor to our device-tree recipe. Mainly, in how we handle things in petalinux to inlcude our own devicetrees. The commit description of  4ebaaa2 (meta-adi-xilinx: device-tree: Refactor recipe) details the changes made. There are also some more minor changes. 64470fb (meta-adi-xilinx: device-tree: Include system-conf.dtsi) should fix #55.