analogdevicesinc/meta-adi #56: Built error

esraltun, 2020-01-27:
I worked with petalinux 2018.3.

Firstly I built successfully project when I added AD9361 drivers. (This project contain ad9361 hdl reference design from 2019R1 branch and meta-adi (2018_R2 branch).)

Then I enabled FPGA Manager option in petalinux configuration  but didn't built my project

I'm facing below issue after enabling FPGA Manager. How can I fix it?

ERROR: device-tree-xilinx+gitAUTOINC+b7466bbeee-r0 do_compile: Function failed: do_compile (log file is located at /home/esra/petalinux-project/zcu102-ad9361-2019R1-hdf/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+b7466bbeee-r0/temp/log.do_compile.26466)
ERROR: Logfile of failure stored in: /home/esra/petalinux-project/zcu102-ad9361-2019R1-hdf/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+b7466bbeee-r0/temp/log.do_compile.26466
Log data follows:
| DEBUG: Executing shell function do_compile
| pl.dtbo: Warning (reg_format): "reg" property in /fragment@2/__overlay__/axi_ad9361@99020000 has invalid length (16 bytes) (#address-cells == 2, #size-cells == 1)
| pl.dtbo: Warning (reg_format): "reg" property in /fragment@2/__overlay__/axi_dmac@9c400000 has invalid length (16 bytes) (#address-cells == 2, #size-cells == 1)
| pl.dtbo: Warning (reg_format): "reg" property in /fragment@2/__overlay__/axi_dmac@9c420000 has invalid length (16 bytes) (#address-cells == 2, #size-cells == 1)
| pl.dtbo: Warning (reg_format): "reg" property in /fragment@2/__overlay__/axi_sysid@85000000 has invalid length (16 bytes) (#address-cells == 2, #size-cells == 1)
| pl.dtbo: Warning (reg_format): "reg" property in /fragment@2/__overlay__/PERIPHERAL@ff380000 has invalid length (16 bytes) (#address-cells == 2, #size-cells == 1)
| pl.dtbo: Warning (reg_format): "reg" property in /fragment@2/__overlay__/PERIPHERAL@ff990000 has invalid length (16 bytes) (#address-cells == 2, #size-cells == 1)
| pl.dtbo: Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
| pl.dtbo: Warning (unit_address_vs_reg): Node /fragment@1 has a unit name, but no reg property
| pl.dtbo: Warning (unit_address_vs_reg): Node /fragment@2 has a unit name, but no reg property
| pl.dtbo: Warning (avoid_default_addr_size): Relying on default #address-cells value for /fragment@2/__overlay__/axi_ad9361@99020000
| pl.dtbo: Warning (avoid_default_addr_size): Relying on default #size-cells value for /fragment@2/__overlay__/axi_ad9361@99020000
| pl.dtbo: Warning (avoid_default_addr_size): Relying on default #address-cells value for /fragment@2/__overlay__/axi_dmac@9c400000
| pl.dtbo: Warning (avoid_default_addr_size): Relying on default #size-cells value for /fragment@2/__overlay__/axi_dmac@9c400000
| pl.dtbo: Warning (avoid_default_addr_size): Relying on default #address-cells value for /fragment@2/__overlay__/axi_dmac@9c420000
| pl.dtbo: Warning (avoid_default_addr_size): Relying on default #size-cells value for /fragment@2/__overlay__/axi_dmac@9c420000
| pl.dtbo: Warning (avoid_default_addr_size): Relying on default #address-cells value for /fragment@2/__overlay__/axi_sysid@85000000
| pl.dtbo: Warning (avoid_default_addr_size): Relying on default #size-cells value for /fragment@2/__overlay__/axi_sysid@85000000
| pl.dtbo: Warning (avoid_default_addr_size): Relying on default #address-cells value for /fragment@2/__overlay__/PERIPHERAL@ff380000
| pl.dtbo: Warning (avoid_default_addr_size): Relying on default #size-cells value for /fragment@2/__overlay__/PERIPHERAL@ff380000
| pl.dtbo: Warning (avoid_default_addr_size): Relying on default #address-cells value for /fragment@2/__overlay__/PERIPHERAL@ff990000
| pl.dtbo: Warning (avoid_default_addr_size): Relying on default #size-cells value for /fragment@2/__overlay__/PERIPHERAL@ff990000
| Error: /home/esra/petalinux-project/zcu102-ad9361-2019R1-hdf/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+b7466bbeee-r0/pl-delete-nodes-zynqmp-zcu102-rev10-ad9361-fmcomms2-3.dtsi:6.15-26 Label or path axi_ad9361 not found
| Error: /home/esra/petalinux-project/zcu102-ad9361-2019R1-hdf/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+b7466bbeee-r0/pl-delete-nodes-zynqmp-zcu102-rev10-ad9361-fmcomms2-3.dtsi:7.15-26 Label or path misc_clk_0 not found
| Error: /home/esra/petalinux-project/zcu102-ad9361-2019R1-hdf/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+b7466bbeee-r0/pl-delete-nodes-zynqmp-zcu102-rev10-ad9361-fmcomms2-3.dtsi:8.15-34 Label or path axi_ad9361_adc_dma not found
| Error: /home/esra/petalinux-project/zcu102-ad9361-2019R1-hdf/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+b7466bbeee-r0/pl-delete-nodes-zynqmp-zcu102-rev10-ad9361-fmcomms2-3.dtsi:9.15-34 Label or path axi_ad9361_dac_dma not found
| Error: /home/esra/petalinux-project/zcu102-ad9361-2019R1-hdf/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+b7466bbeee-r0/pl-delete-nodes-zynqmp-zcu102-rev10-ad9361-fmcomms2-3.dtsi:10.15-28 Label or path psu_ctrl_ipi not found
| Error: /home/esra/petalinux-project/zcu102-ad9361-2019R1-hdf/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+b7466bbeee-r0/pl-delete-nodes-zynqmp-zcu102-rev10-ad9361-fmcomms2-3.dtsi:11.15-35 Label or path psu_message_buffers not found
| Error: /home/esra/petalinux-project/zcu102-ad9361-2019R1-hdf/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+b7466bbeee-r0/pl-delete-nodes-zynqmp-zcu102-rev10-ad9361-fmcomms2-3.dtsi:12.15-27 Label or path axi_sysid_0 not found
| FATAL ERROR: Syntax error parsing input tree

nunojsa, 2020-01-28:
Hi,

So I could confirm your issue. When enabling the FPGA Manager option the `pl.dtsi` will be compiled as a dtboverlay. This complicates things a lot since this is handled also by another recipe (rather than devicetree.bb). So, at this point, I'm not sure when we can properly support this since it is not trivial. So I would say for now that enabling FPGA Manager is not the best option to use with our layer.

If you really want to have this feature I can try to help you with some pointers on how you can start "hot" fixing this. So:

1. You need to go [devicetree](https://github.com/analogdevicesinc/meta-adi/blob/2019_R1/meta-adi-xilinx/recipes-bsp/device-tree/device-tree.bbappend) and comment the line `echo -e "/include/ \"${KERNEL_PL_DTB_FILE}\"" | cat - ${WORKDIR}/analog-devices.dtsi > temp && mv temp ${WORKDIR}/analog-devices.dtsi`. This will prevent us to delete nodes that do not exist in this configuration. This would be the first part.

2. From what I could see, `fpga-manager-util_1.0.bb` recipe is doing the dtboverlay compilation. So, this recipe would be the one to append. You need to delete the nodes that correspond to ADI cores so that our drivers are probed. You could just copy `pl-delete-nodes-zynqmp-zcu102-rev10-ad9361-fmcomms2-3.dtsi` and include it at the end of pl.dtsi. I would prepend the `do_compile()` task o add these hacks. Give a look at `${your-petalinux-install-path}/components/yocto/source/aarch64/layers/meta-xilinx-tools/recipes-bsp/fpga-manager-util/fpga-manager-util_1.0.bb`.

Note these are just some ideas that I'm giving you from top of my head!

Hope this helps!

esraltun, 2020-01-29:
Hi, Thank you very much for your helping.
 I built successfully my project but didn't boot on zcu102. I didnt see anything minicom serial terminal.
Esra

nunojsa, 2020-02-04:
Some updates on this. Unfortunately, this is something that we won't official support. The problem is that you need to only build a minimal devicetree with no PL nodes on it (otherwise the kernel will stuck since it will try to read/write some piece of nonexistent HW). On top of this "minimal" dts, you need to create an overlay for the reference design you are using on top of [in this case] zynqMP. So, since in our kernel tree we are not supporting these overlys, we won't be maintaining them here... 

However, I will try to at least have a working example (on this platform) so that people can follow... But for others platforms, it will be the user responsibility to prepare the dtoverlay

rgetz, 2020-02-04:
Maybe the thing to do, is complain to Xilinx - just turning on FPGA manager shouldn't mean that you want to separate things the way Xilinx does - it just means that you want a portion (for Partial Reconfig) of the FPGA managed...

I think this is a side effect of the tools over-reaching it's intended purpose.

closed 2020-02-11 by commodo

hamzah96343, 2024-08-27:
@nunojsa Hey I saw this post because I am having similar issues. I followed your commit to the best of my ability but fpga-manager-util no longer exists in 2023.2. Could you guide me with a flow for using DT overlays for the PL and using the meta-adi layer and fpga manager with petalinux 2023.2?

PR #58 (2020-02-11): Partial Support fpga manager
This PR adds some hooks/guidelines on how one can use FPGA Manager. It fixes #56