analogdevicesinc/meta-adi #64: Fpga manager enable on zcu102 with fmcomms5

esraltun, 2020-04-07:
Hi,
I work with ad9361 on zcu102.(hdl master branch for fmcomms5)  I build the project with using petalinux tool(v2019.2)
The project fail, when I enabled Fpga manager or devicetree overlay.

**Error: pl.dtsi:8.1-9.10  Header flags don't match earlier ones**

pl.dtsi
/*
 * CAUTION: This file is automatically generated by Xilinx.
 * Version:  
 * Today is: Tue Apr  7 13:17:57 2020
 */

/dts-v1/;
/plugin/;
/ {
	fragment@0 {
		target = <&fpga_full>;
		overlay0: __overlay__ {
			#address-cells = <2>;
			#size-cells = <2>;
			firmware-name = "system_top.bit.bin";
			resets = <&zynqmp_reset 116>;
		};
	};
	fragment@1 {
		target = <&amba>;
		overlay1: __overlay__ {
			afi0: afi0 {
				compatible = "xlnx,afi-fpga";
				config-afi = < 0 0>, <1 0>, <2 0>, <3 0>, <4 0>, <5 0>, <6 0>, <7 0>, <8 0>, <9 0>, <10 0>, <11 0>, <12 0>, <13 0>, <14 0xa00>, <15 0x000>;
			};
			clocking0: clocking0 {
				#clock-cells = <0>;
				assigned-clock-rates = <99990005>;
				assigned-clocks = <&zynqmp_clk 71>;
				clock-output-names = "fabric_clk";
				clocks = <&zynqmp_clk 71>;
				compatible = "xlnx,fclk";
			};
			clocking1: clocking1 {
				#clock-cells = <0>;
				assigned-clock-rates = <249975021>;
				assigned-clocks = <&zynqmp_clk 72>;
				clock-output-names = "fabric_clk";
				clocks = <&zynqmp_clk 72>;
				compatible = "xlnx,fclk";
			};
			clocking2: clocking2 {
				#clock-cells = <0>;
				assigned-clock-rates = <499950043>;
				assigned-clocks = <&zynqmp_clk 73>;
				clock-output-names = "fabric_clk";
				clocks = <&zynqmp_clk 73>;
				compatible = "xlnx,fclk";
			};
		};
	};
	fragment@2 {
		target = <&amba>;
		overlay2: __overlay__ {
			#address-cells = <2>;
			#size-cells = <2>;
			axi_ad9361_0: axi_ad9361@99020000 {
				clock-names = "delay_clk", "clk", "s_axi_aclk";
				clocks = <&zynqmp_clk 73>, <&misc_clk_0>, <&zynqmp_clk 71>;
				compatible = "xlnx,axi-ad9361-1.0";
				reg = <0x0 0x99020000 0x0 0x10000>;
			};
			misc_clk_0: misc_clk_0 {
				#clock-cells = <0>;
				clock-frequency = <100000000>;
				compatible = "fixed-clock";
			};
			axi_ad9361_1: axi_ad9361@99040000 {
				clock-names = "delay_clk", "clk", "s_axi_aclk";
				clocks = <&zynqmp_clk 73>, <&misc_clk_0>, <&zynqmp_clk 71>;
				compatible = "xlnx,axi-ad9361-1.0";
				reg = <0x0 0x99040000 0x0 0x10000>;
			};
			axi_ad9361_adc_dma: axi_dmac@9c400000 {
				clock-names = "s_axi_aclk", "m_dest_axi_aclk", "fifo_wr_clk";
				clocks = <&zynqmp_clk 71>, <&zynqmp_clk 72>, <&misc_clk_0>;
				compatible = "xlnx,axi-dmac-1.0";
				interrupt-names = "irq";
				interrupt-parent = <&gic>;
				interrupts = <0 109 4>;
				reg = <0x0 0x9c400000 0x0 0x1000>;
			};
			axi_ad9361_dac_dma: axi_dmac@9c420000 {
				clock-names = "s_axi_aclk", "m_src_axi_aclk", "m_axis_aclk";
				clocks = <&zynqmp_clk 71>, <&zynqmp_clk 72>, <&misc_clk_0>;
				compatible = "xlnx,axi-dmac-1.0";
				interrupt-names = "irq";
				interrupt-parent = <&gic>;
				interrupts = <0 108 4>;
				reg = <0x0 0x9c420000 0x0 0x1000>;
			};
			axi_sysid_0: axi_sysid@85000000 {
				clock-names = "s_axi_aclk";
				clocks = <&zynqmp_clk 71>;
				compatible = "xlnx,axi-sysid-1.0";
				reg = <0x0 0x85000000 0x0 0x10000>;
			};
		};
	};
};

How can I fix it?
Thanks

nunojsa, 2020-04-07:
Are you actually using meta-adi? Also note that 2019.2 version is not supported...

I would try a clean build (if not the case)...

esraltun, 2020-04-07:
Thanks for replying.
Yes I use meta-adi master branch and I build successfully the project (using 2019.2 petalinux) 
I clean project and again build so it builds.

esraltun, 2020-04-07:
Addition How can I create .dtsi as pl-fmcomms5-zcu102-overlay.dtsi for fmcomms2-3 ?

nunojsa, 2020-04-08:
Hi,

Lots of things might go wrong here. Pretty much you are using a petalinux version that we don't support yet. So you should use 2019.1. If you really need 2019.2 you are kind on your own for now...

Regarding FPGA Manager, this is also not officially supported as stated in [fpga manager](https://github.com/analogdevicesinc/meta-adi/tree/master/meta-adi-xilinx#fpga-manager)... We just give minimal support for this. Doing the overlays is up to the user. You can go to our [kernel](https://github.com/analogdevicesinc/linux) and look at [ynqmp-zcu102-rev10-ad9361-fmcomms2-3.dts](https://github.com/analogdevicesinc/linux/blob/master/arch/arm64/boot/dts/xilinx/zynqmp-zcu102-rev10-ad9361-fmcomms2-3.dts) if this is the one you are using.

Hope this helps,

esraltun, 2020-04-09:
Hi,
 I'm building  on my project with using petalinux 2019.1 but anything not changed. 

Thanks for advising.

nunojsa, 2020-04-09:
Hi,

Again, fpga manager is not officially supported and what we have in meta-adi should serve as an example to get going... Using it is not straight. First you need to guarantee that the devicetree to boot the system is minimal. Meaning, no PL nodes can be there because in this mode your bitstream is not loaded which will hang the kernel if some driver tries to access PL mapped memory. On top of this, you need to take care about the overlay for your project... 

Are you booting with initramfs or with sdcard? If sdcard is being used make sure to set `root=/dev/mmcblk0p2` on uEnv.txt...

Hope this helps!

nunojsa, 2020-07-14:
No updates for some time. Closing this issue

closed 2020-07-14 by nunojsa