analogdevicesinc/meta-adi #68: ZC706 ad9361Unable to parse input tree

thecodeforzj, 2020-07-20:
https://github.com/analogdevicesinc/meta-adi/issues/14.
I'm experiencing the exact same issue. I use it '$ sudo dpkg-reconfigure dash' , but it wasn't sliven it. I am using Petalinux-v2018.3

nunojsa, 2020-08-12:
> Petalinux-v2018.3 is supported on the 2019_R1 branch. Is that the one you are using? If so, make sure you have a proper environment to build petalinux. Alternatively, you can use the petalinux dockerfile to build your project...

This is my reply in #14. Typically, this kind of issue is related with your environment... Can you actually build without including meta-adi layers? If you still have this problem using our docker image for petalinux then I will need more details about your issue. Otherwise, It seems that this is not really a problem related with meta-adi

nunojsa, 2020-09-25:
As this issue has been quite for a long time now, I'm closing it...

closed 2020-09-25 by nunojsa