analogdevicesinc/meta-adi #69: getting error when building

noelpedro, 2020-07-28:
Hi all Im getting an error when attempting to build my linux image for device z7035ad9361-bob. Just last week I was able to build image with 2019.1 using meta-adi mater branch. Now Im getting this error.

Seems to be related to drivers for adrv9002 ??

adleman@zirconium:~/work/z7035_linux$ petalinux-build 
[INFO] building project
[INFO] sourcing bitbake
[INFO] generating user layers
INFO: bitbake petalinux-user-image
Loading cache: 100% |############################################| Time: 0:00:01
Loaded 3813 entries from dependency cache.
Parsing recipes: 100% |##########################################| Time: 0:00:04
Parsing of 2783 .bb files complete (2772 cached, 11 parsed). 3818 targets, 163 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
Initialising tasks: 100% |#######################################| Time: 0:00:03
Checking sstate mirror object availability: 100% |###############| Time: 0:00:02
Sstate summary: Wanted 154 Found 26 Missed 256 Current 797 (16% match, 86% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: linux-xlnx-4.19-adi_master+gitAUTOINC+a1641c6568-r0 do_compile: oe_runmake failed
ERROR: linux-xlnx-4.19-adi_master+gitAUTOINC+a1641c6568-r0 do_compile: Function failed: do_compile (log file is located at /home/adleman/work/z7035_linux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/linux-xlnx/4.19-adi_master+gitAUTOINC+a1641c6568-r0/temp/log.do_compile.20914)
ERROR: Logfile of failure stored in: /home/adleman/work/z7035_linux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/linux-xlnx/4.19-adi_master+gitAUTOINC+a1641c6568-r0/temp/log.do_compile.20914
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8 HOSTCC=gcc  -isystem/home/adleman/work/z7035_linux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/linux-xlnx/4.19-adi_master+gitAUTOINC+a1641c6568-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/home/adleman/work/z7035_linux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/linux-xlnx/4.19-adi_master+gitAUTOINC+a1641c6568-r0/recipe-sysroot-native/usr/lib -L/home/adleman/work/z7035_linux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/linux-xlnx/4.19-adi_master+gitAUTOINC+a1641c6568-r0/recipe-sysroot-native/lib -Wl,-rpath-link,/home/adleman/work/z7035_linux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/linux-xlnx/4.19-adi_master+gitAUTOINC+a1641c6568-r0/recipe-sysroot-native/usr/lib -Wl,-rpath-link,/home/adleman/work/z7035_linux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/linux-xlnx/4.19-adi_master+gitAUTOINC+a1641c6568-r0/recipe-sysroot-native/lib -Wl,-rpath,/home/adleman/work/z7035_linux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/linux-xlnx/4.19-adi_master+gitAUTOINC+a1641c6568-r0/recipe-sysroot-native/usr/lib -Wl,-rpath,/home/adleman/work/z7035_linux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/linux-xlnx/4.19-adi_master+gitAUTOINC+a1641c6568-r0/recipe-sysroot-native/lib -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/adleman/work/z7035_linux/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 HOSTCPP=gcc  -E zImage CC=arm-xilinx-linux-gnueabi-gcc  -mno-thumb-interwork -marm -fuse-ld=bfd -fdebug-prefix-map=/home/adleman/work/z7035_linux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/linux-xlnx/4.19-adi_master+gitAUTOINC+a1641c6568-r0=/usr/src/debug/linux-xlnx/4.19-adi_master+gitAUTOINC+a1641c6568-r0 -fdebug-prefix-map=/home/adleman/work/z7035_linux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/linux-xlnx/4.19-adi_master+gitAUTOINC+a1641c6568-r0/recipe-sysroot= -fdebug-prefix-map=/home/adleman/work/z7035_linux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/linux-xlnx/4.19-adi_master+gitAUTOINC+a1641c6568-r0/recipe-sysroot-native=  -fdebug-prefix-map=/home/adleman/work/z7035_linux/build/tmp/work-shared/plnx-zynq7/kernel-source=/usr/src/kernel   LD=arm-xilinx-linux-gnueabi-ld.bfd   UIMAGE_LOADADDR=0x8000
|   GEN     ./Makefile
|   Using /home/adleman/work/z7035_linux/build/tmp/work-shared/plnx-zynq7/kernel-source as source for kernel
|   CALL    /home/adleman/work/z7035_linux/build/tmp/work-shared/plnx-zynq7/kernel-source/scripts/checksyscalls.sh
|   CHK     include/generated/compile.h
|   GEN     .version
|   CHK     include/generated/compile.h
|   AR      built-in.a
|   LD      vmlinux.o
|   MODPOST vmlinux.o
| arm-xilinx-linux-gnueabi-ld.bfd: drivers/iio/adc/navassa/adrv9002.o: in function `adrv9002_intf_change_delay':
| /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:2194: undefined reference to `adi_adrv9001_Ssi_Delay_Configure'
| arm-xilinx-linux-gnueabi-ld.bfd: drivers/iio/adc/navassa/adrv9002.o: in function `adrv9002_check_tx_test_pattern':
| /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:2215: undefined reference to `adi_adrv9001_Ssi_Tx_TestMode_Status_Inspect'
| arm-xilinx-linux-gnueabi-ld.bfd: /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:2236: undefined reference to `adi_adrv9001_Ssi_Tx_TestMode_Status_Inspect'
| arm-xilinx-linux-gnueabi-ld.bfd: drivers/iio/adc/navassa/adrv9002.o: in function `adrv9002_intf_test_cfg':
| /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:2269: undefined reference to `adi_adrv9001_Ssi_Tx_TestMode_Configure'
| arm-xilinx-linux-gnueabi-ld.bfd: /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:2293: undefined reference to `adi_adrv9001_Ssi_Rx_TestMode_Configure'
| arm-xilinx-linux-gnueabi-ld.bfd: /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:2307: undefined reference to `adi_adrv9001_Ssi_Rx_TestMode_Configure'
| arm-xilinx-linux-gnueabi-ld.bfd: /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:2282: undefined reference to `adi_adrv9001_Ssi_Tx_TestMode_Configure'
| arm-xilinx-linux-gnueabi-ld.bfd: drivers/iio/adc/navassa/adrv9002.o: in function `adrv9002_ssi_delays_write':
| /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:3111: undefined reference to `adi_adrv9001_Ssi_Delay_Configure'
| arm-xilinx-linux-gnueabi-ld.bfd: drivers/iio/adc/navassa/adrv9002.o: in function `adrv9002_ssi_rx_test_mode_set':
| /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:2929: undefined reference to `adi_adrv9001_Ssi_Rx_TestMode_Configure'
| arm-xilinx-linux-gnueabi-ld.bfd: drivers/iio/adc/navassa/adrv9002.o: in function `adrv9002_ssi_tx_test_mode_set':
| /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:3029: undefined reference to `adi_adrv9001_Ssi_Tx_TestMode_Configure'
| arm-xilinx-linux-gnueabi-ld.bfd: drivers/iio/adc/navassa/adrv9002.o: in function `adrv9002_intf_tuning_unlocked':
| /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:2392: undefined reference to `adi_adrv9001_Ssi_Delay_Configure'
| arm-xilinx-linux-gnueabi-ld.bfd: drivers/iio/adc/navassa/adrv9002.o: in function `adrv9002_irq_handler':
| /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:1817: undefined reference to `adi_adrv9001_gpio_GpIntHandler'
| arm-xilinx-linux-gnueabi-ld.bfd: /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:1835: undefined reference to `adi_adrv9001_cals_InitCals_Run'
| arm-xilinx-linux-gnueabi-ld.bfd: drivers/iio/adc/navassa/adrv9002.o: in function `adrv9002_get_port_en_mode':
| /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:909: undefined reference to `adi_adrv9001_Radio_ChannelEnableMode_Get'
| arm-xilinx-linux-gnueabi-ld.bfd: drivers/iio/adc/navassa/adrv9002.o: in function `adrv9002_set_port_en_mode':
| /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:932: undefined reference to `adi_adrv9001_Radio_ChannelEnableMode_Set'
| arm-xilinx-linux-gnueabi-ld.bfd: drivers/iio/adc/navassa/adrv9002.o: in function `adrv9002_get_ensm_mode':
| /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:718: undefined reference to `adi_adrv9001_Radio_Channel_State_Get'
| arm-xilinx-linux-gnueabi-ld.bfd: drivers/iio/adc/navassa/adrv9002.o: in function `adrv9002_set_ensm_mode':
| /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:694: undefined reference to `adi_adrv9001_Radio_Channel_ToState'
| arm-xilinx-linux-gnueabi-ld.bfd: /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:682: undefined reference to `adi_adrv9001_Radio_Channel_State_Get'
| arm-xilinx-linux-gnueabi-ld.bfd: drivers/iio/adc/navassa/adrv9002.o: in function `adrv9002_phy_tx_read':
| /usr/src/kernel/drivers/iio/adc/navassa/adrv9002.c:1335: undefined reference to `adi_adrv9001_cals_Tracking_Get'
| arm-xilinx-linux-gnueabi-ld.bfd: drivers/iio/adc/navassa/adrv9002.o: in function `adrv9002_phy_rx_read':
[... cut ...]
| ERROR: Function failed: do_compile (log file is located at /home/adleman/work/z7035_linux/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/linux-xlnx/4.19-adi_master+gitAUTOINC+a1641c6568-r0/temp/log.do_compile.20914)
ERROR: Task (/home/adleman/src/meta-adi/meta-adi-xilinx/recipes-kernel/linux/linux-xlnx_adi-dev.bb:do_compile) failed with exit code '1'

Second Keyboard Interrupt, stopping...

Summary: 1 task failed:
  /home/adleman/src/meta-adi/meta-adi-xilinx/recipes-kernel/linux/linux-xlnx_adi-dev.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
ERROR: Failed to build project

mhennerich, 2020-07-29:
Thanks for reporting. We are aware of the issue. It's only triggered when the kernel is build from Petalinux. It doesn't happen if you invoke make from the kernel source directory. For now please disable kernel CONFIG_ADRV9001. Alternatively comment
linux/drivers/iio/adc/Makefile line:

# obj-$(CONFIG_ADRV9001) += navassa/ 

someone will start looking into this soon.

commodo, 2020-07-29:
Should be fixed now.
Merged https://github.com/analogdevicesinc/linux/pull/1085

closed 2020-07-29 by commodo