analogdevicesinc/meta-adi #71: Cannot find pl.dtsi during petalinux build

NeutrinoXY, 2020-08-06:
When building petalinux using meta-adi-xilinx and meta-adi-core layers, I got an error during do_compile of the linux-xlnx recipe.

I tried to include using FILESEXTRAPATHS_prepend in a linux-xlnx_%.bbappend in my meta-user layer, but it doesn't work.

I quickly fixed it by adding the two following lines in meta-adi-xilinx/recipes-bsp/device-tree/device-tree.bbappend, but it seems messy to me:

cp ${DT_FILES_PATH}/pl.dtsi ${DTS_INCLUDE_PATH}/pl.dtsi
cp ${WORKDIR}/${DTB_PL_DELETE}.dtsi ${DTS_INCLUDE_PATH}/${DTB_PL_DELETE}.dtsi

Any idea? Thanks.

nunojsa, 2020-08-12:
> I got an error during do_compile of the linux-xlnx recipe.

This is really odd. Building the kernel should not have nothing to do with pl.dtsi. Are you using the master branch? What project are you trying to build? Are you trying to build any devicetree as part of the `do_compile` on the linux-xlnx recipe?

Those `cp` commands should not be needed.  Anyways, I will definitely need more details like log files, project you are trying to build, etc to check this as this is not happening in our builds...

On a side note, it might take sometime to check this since I'm only returning to work in September...

nunojsa, 2020-09-25:
>  I also used the zcu102 bsp to create my project. Not sure if that would affect this

I guess it should not matter but I never tried that so I'm not really sure...

Did you had to delete those lines in the master branch? If so, it might indicate that there are some differences after all by importing the bsp...

@SpontaneousDuck  Did you also had issues on your linux compilation as @NeutrinoXY ? It would be nice to get some more info about this to see if I can reproduce this...

SpontaneousDuck, 2020-09-25:
@nunojsa So I am on commit dd9f8e5ae1b814b2cc87db411173c7a220314d54 and using Petalinux 2019.2. I had to make both the change that @NeutrinoXY did as well as the one I did above in order to compile properly. I think I only had these issues and had to make these changes when importing the BSP. What else do you need to know? Thanks!

nunojsa, 2020-09-28:
Thanks for your reply. So, Petalinux 2019.2 is still not supported which might also be an indication of the problems you had. It should be supported by the end of this year or beginning of the next one (no promises though). I might give it a try with the BSP importing (when I find the time ) even though I'm not sure that's a workflow we want to support...
Anyways, I will wait a couple of more days to see if @NeutrinoXY was using the same environment as you and if so, I'm closing this issue for now as we do not support both petalinux 2019.2 and importing BSP files. If doable, I might add importing BSP files as a new feature...

Thanks for your feedback!

SpontaneousDuck, 2020-09-28:
Understood! Marking the issue here for anyone who needs it in the future.

nunojsa, 2020-10-05:
Closing the issue for the stated reasons above...

closed 2020-10-05 by nunojsa