analogdevicesinc/meta-adi #73: Issues adding DMAC driver to dtsi. adrv9361z7035 bob master meta-adi

noelpedro, 2020-10-06:
``dma: dma@7c420000 {
	compatible = "adi,axi-dmac-1.00.a";
	reg = <0x7c420000 0x10000>;
	interrupts = <0 57 0>;
	clocks = <&clkc 16>;
	#dma-cells = <1>;

	adi,channels {
		#size-cells = <0>;
		#address-cells = <1>;

		dma-channel@0 {
			reg = <0>;
			adi,source-bus-width = <32>;
			adi,source-bus-type = <ADI_AXI_DMAC_TYPE_MM_AXI>;
			adi,destination-bus-width = <64>;
			adi,destination-bus-type = <ADI_AXI_DMAC_TYPE_FIFO>;
		};
	};
};
``

commodo, 2020-10-07:
so what's the error?
this is a pretty vague report;

maybe don't use `ADI_AXI_DMAC_TYPE_MM_AXI` ; use direct numerical values;
otherwise you need to include a header file that defines the numerical value for ADI_AXI_DMAC_TYPE_MM_AXI

nunojsa, 2020-10-07:
How are you adding this to your petalinux build? Looking at this, there's no obvious errors with the dts node. I will also need more detail about the build error? And which hdl + petalinux version + meta-adi branch you are using?

noelpedro, 2020-10-07:
> so what's the error?
> this is a pretty vague report;
> 
> maybe don't use `ADI_AXI_DMAC_TYPE_MM_AXI` ; use direct numerical values;
> otherwise you need to include a header file that defines the numerical value for ADI_AXI_DMAC_TYPE_MM_AXI

Thanks
T

T

noelpedro, 2020-10-07:
Where can I find the device tree source for the adrv9361 7035 bob that meta-adi master branch uses?

commodo, 2020-10-08:
In the Linux kernel repo.
https://github.com/analogdevicesinc/linux/blob/master/arch/arm/boot/dts/zynq-adrv9361-z7035-bob.dts

I feel that you are having troubles understanding the internals of Yocto or Petalinux.
If that is the case, please try to better understand Yocto or Petalinux.

It's also not nice to do the method of "try-something-it-doesn-t-work-open-support-issue" vs "try-something-it-doesn-t-work-investigate" method.

noelpedro, 2020-10-08:
It’s also not nice to be unhappy to answer my question.

I did look in that directory before posting the Q.

Nevertheless, thanks

closed 2020-10-08 by noelpedro