analogdevicesinc/meta-adi #90: Error: Driver 'adm1177' is already registered, aborting...

NeutrinoXY, 2021-01-08:
**Environment (please complete the following information):**
 - Project to build: custom (fmcomms2-like)
 - Petalinux version: 2019.2
 - HDL branch: custom
 - META-ADI branch: master, commit ca00f171982d2a4b0b8f8b70f4bdb3ffe8141efa
 - Using meta-adi petalinux docker image? No

**Describe the bug**

I built a petalinux project with the meta-adi layers. I first built it for a zcu102 and it worked. Then I tried to build it for a custom board, after of course some modifications to the device-tree. The main difference with the evaluation board is that there is no i2c. However, when I boot the kernel, I get the following error in the boot log:

[    4.480451] Error: Driver 'adm1177' is already registered, aborting...

So I tried to blacklist this module by adding a .conf file in /etc/modprobe.d, but the error remains. What could cause this module to be loaded twice?

Thanks.

commodo, 2021-01-08:
We have 2 adm1177 drivers in our tree.
This one [1] https://github.com/analogdevicesinc/linux/blob/master/drivers/iio/adc/adm1177.c
And this one  https://github.com/analogdevicesinc/linux/blob/master/drivers/hwmon/adm1177.c

The latter i also present in mainline kernel.
You should make sure that the first driver [1] is patched with this commit  https://github.com/analogdevicesinc/linux/commit/3a901a778f83f4c12106bef6670db85c49f9d5be

nunojsa, 2021-01-08:
Also note that petalinux 2019.2 is not yet supported... Nevertheless, what @commodo said is valid. However, I would say that the commit should be present as we just get the current tip of master (unless a offline build is done). So make sure to do a clean petalinux build so that the latest sources are fetched...

Since this is not a supported version, I will leave this around some more time to help you with some minor things and then I will close it...

NeutrinoXY, 2021-01-08:
I modified the revision of the linux source to the commit you cited and it now works (I use an offline build).

Thanks for your support.

closed 2021-01-08 by NeutrinoXY