analogdevicesinc/meta-adi #91: Not able to boot the build images for adrv9361z7045/ccbob_lvds project

pnkjslr, 2021-02-02:
- Project to build: adrv9361z7035/ccbob_lvds with zynq device changed from z7035 to z7045 induatrial grade version for custom board but our board has ad9364 instead of ad9361.
 - Petalinux version: 2019.2
 - Vivado version: 2019.2
 - HDL branch: hdl-2019-r2
 - META-ADI branch: 2019-R2
 - Using meta-adi petalinux docker image?

I have build the Linux image and boot image using the meta-adi flow successfully. But when I boot the BOOT.BIN and image.ub files using sd card there are no boot logs. If I create a simple board design using z7045 device and zynq-7000 IP in vivado and use its .xsa file to build simple linux then in that case I am able to boot successfully.  

I am using a custom board which have z7045 and ad9364. I have also included TTC timer in board my design which is required as per ug1144 document and KERNEL_DTB="zynq-adrv9361-z7035-bob".

pnkjslr, 2021-02-03:
context.bbappend:
For u-boot I am using default zynq u-boot config that is "zynq_zc702_config". do I need to change it? If yes, then why it is not mentioned in building with petalinux guide and which one is more suitable for my hardware?

nunojsa, 2021-02-03:
Hi,

Well, 2019_r2 is to be used with petalinux 2019.1. The hdl branch is also using Vivado 2019.1. Hence, since this is not supported or was never tested for these versions, there are lots of things that might go wrong...

Moreover, we do not have any design for z7045 with ad9364 which means that just using `zynq-adrv9361-z7035-bob"` might not really work... I would also look at an example like [this](https://github.com/analogdevicesinc/linux/blob/master/arch/arm/boot/dts/zynq-adrv9364-z7020.dtsi) to see how to define the devicetree for ad9364.  You should also add your own devicetree (as there's none in the ADI linux tree for this setup) following [adding a new devicetree](https://github.com/analogdevicesinc/meta-adi/tree/2019_R2/meta-adi-xilinx#adding-a-new-devicetree).

Please, bear in mind that this is the most I can do since this is not a supported version for meta-adi. We will be jumping one release and the next petalinux version that we will support is 2020.1.

Hope this helps!

pnkjslr, 2021-02-04:
For the device tree problem: I think KERNEL_DTB is used for deleting PL nodes. Further, I have modified vivado project adrv9361z7035 for z7045 device and then I generated xsa file. Hence using KERNEL_DTB as "zynq-adrv9361-z7035-bob" should work as PL logic is similar?

pnkjslr, 2021-02-04:
One more doubt, how to replace "model" property in device tree for a custom hardware?

nunojsa, 2021-02-04:
> For the device tree problem: I think KERNEL_DTB is used for deleting PL nodes

Nope, `KERNEL_DTB` is the devicetree you want to use (and using a devicetree for ad9361 might not work for ad9364). To delete the PL nodes, a `pl-delete-nodes-${KERNEL_DTB}` is typically created and is included at the end of the pl.dtsi. The goal is to delete all the custom nodes that you create in Vivado since petalinux will most likely create invalid dts nodes for it...

> One more doubt, how to replace "model" property in device tree for a custom hardware?

You need to change the dts property. In meta-adi, this is how one typically [extends](https://github.com/analogdevicesinc/meta-adi/tree/2019_R2/meta-adi-xilinx/#extending-the-devicetree) the devicetree...

pnkjslr, 2021-02-08:
Hey, thanks for the help! Now, I have a clear picture of what is required and I'm working on custom-devicetree.dts file. As the device tree of my board would require "[zynq-adrv9364-z7020.dtsi](https://github.com/analogdevicesinc/linux/blob/master/arch/arm/boot/dts/zynq-adrv9364-z7020.dtsi)" include file, do I need to add my custom-devicetree.dts in [adi-Linux](https://github.com/analogdevicesinc/linux/tree/2019_R2)'s git repository for Petalinux build?

nunojsa, 2021-02-08:
I don't think so. The devicetree recipe points dtc to the kernel tree to look for .dtsi files. Hence, if that devicetree is present in adi-linux, dtc shoud be able to find it...

Closing now this issue as it's not really an issue at this point and as I explained before, 2019.2 is not a supported version so we can't really offer too much support for it...

good luck!

closed 2021-02-08 by nunojsa