analogdevicesinc/meta-adi #92: Recommended branch

mnsgs, 2021-02-04:
Hi,

I have previously experienced that there is a tight coupling between the kernel HDL release, and it is not clear to me which branch of this meta layer is the recommended? From https://github.com/analogdevicesinc/hdl/releases 2019_r1 appears to be the most recent release, however https://github.com/analogdevicesinc/meta-adi/pull/85 indicates 2019_r2 ? Which version do you currently recommend? 

Thanks,
Martin

nunojsa, 2021-02-05:
Hi,

As this is not really an issue I would say to use the new github [Discussions](https://github.com/analogdevicesinc/meta-adi/discussions) tab in the futrue. So as for your question the [readme](https://github.com/analogdevicesinc/meta-adi/blob/2019_R2/meta-adi-xilinx/README.md) file in each supported branch should have that info...

- Nuno Sá

closed 2021-02-08 by nunojsa

mnsgs, 2021-02-08:
With the introduction of https://github.com/analogdevicesinc/hdl/releases/tag/hdl_2019_r2 the answer is given - thx....