analogdevicesinc/meta-adi #93: Custom devicetree compilation error: custom-devicetree addition guide of meta-adi not working

pnkjslr, 2021-02-17:
**Environment (please complete the following information):**
 - Project to build: adrv9361-z7035
 - Petalinux version:2019.1
 - HDL branch: hdl_2019_r2
 - META-ADI branch: 2019_R2
 

**Bug**
My custom hardware board is based on AES-PZSDR-FMC-G eval-board with some functionality removed from it. For that I had to write one custom-devicetree.dts and add it to /project-spec/meta-adi-2019_R2/meta-adi-xilinx/recipes-bsp/device-tree/files directory and correspondingly modified device-tree.bbappend as provided in meta-adi guide for custom devicetree addition. But, Petalinux is giving compilation error. Any help will be highly appreciated.

**device-tree.bbappend changes**
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"
SRC_URI_append_zynq = " \
		file://pl-delete-nodes-custom-devicetree.dtsi \
		file://custom-devicetree.dts"
KERNEL_DTB_PATH ?= "${WORKDIR}"

**custom-devicetree.dts**
#incude "zynq.dtsi"
#include <dt-bindings/interrupt-controller/irq.h>
/{
	model="Hello RemoteT";
	memory {
		device_type = "memory";
		reg = <0x00000000 0x40000000>;
	};

	chosen {
		stdout-path = "/amba@0/uart@E0001000";
	};

	clocks {

		ad9364_clkin: clock@0 {
			#clock-cells = <0>;
			compatible = "adjustable-clock";
			clock-frequency = <40000000>;
			clock-accuracy = <200000>; /* 200 ppm (ppb) */
			clock-output-names = "ad9364_ext_refclk";
		};

		xo_40mhz_fixed_clk: clock@0 {
			#clock-cells = <0>;
			compatible = "adjustable-clock";
			clock-frequency = <40000000>;
			clock-accuracy = <200000>; /* 200 ppm (ppb) */
			clock-output-names = "XO_40MHz";
		};

		usb_ulpi_fixed_clk: clock@2 {
			#clock-cells = <0>;
			compatible = "fixed-clock";
			clock-frequency = <24000000>;
			clock-output-names = "24MHz";
		};
	};

	ad9361_clkin: ad9361-refclk-gpio-gate@0 {
		#clock-cells = <0>;
		compatible = "gpio-gate-clock";
		clocks = <&xo_40mhz_fixed_clk>;
		enable-gpios = <&gpio0 105 0>; /* Set to 1 for extern AD9361_CLK (J1 on PZSDRCC-FMC) */
		clock-output-names = "ad9361_ext_refclk";
	};

	usb_ulpi_clk: usb-ulpe-gpio-gate@0 {
		#clock-cells = <0>;
		compatible = "gpio-gate-clock";
		clocks = <&usb_ulpi_fixed_clk>;
		enable-gpios = <&gpio0 9 1>; /* Active  Low */
	};

	leds {
		compatible = "gpio-leds";
		led0 {
			label = "led0:red";
			gpios = <&gpio0 61 0>;
		};

		led1 {
			label = "led1:red";
			gpios = <&gpio0 59 0>;
		};

		led2 {
			label = "led2:red";
			gpios = <&gpio0 58 0>;
		};

		led3 {
			label = "led3:red";
			gpios = <&gpio0 60 0>;
		};
	};

	gpio_keys {
		compatible = "gpio-keys";
		#address-cells = <1>;
		#size-cells = <0>;
		autorepeat;

		bt0 {
			label = "BT0";
			linux,code = <KEY_LEFT>;
			gpios = <&gpio0 54 0>;
		};

		bt1 {
			label = "BT1";
			linux,code = <KEY_RIGHT>;
			gpios = <&gpio0 55 0>;
		};

		bt2 {
			label = "BT2";
			linux,code = <KEY_ENTER>;
			gpios = <&gpio0 56 0>;
		};

		bt3 {
			label = "BT3";
			linux,code = <KEY_ESC>;
			gpios = <&gpio0 57 0>;
		};

		sw0 {
			label = "SW0";
			linux,input-type = <5>;
			linux,code = <0>;
			gpios = <&gpio0 65 0>;
		};

		sw1 {
			label = "SW1";
			linux,input-type = <5>;
			linux,code = <1>;
			gpios = <&gpio0 62 0>;
		};

		sw2 {
			label = "SW2";
			linux,input-type = <5>;
			linux,code = <2>;
			gpios = <&gpio0 64 0>;
		};

		sw3 {
			label = "SW3";
			linux,input-type = <5>;
			linux,code = <3>;
			gpios = <&gpio0 63 0>;
		};
};

&gem0 {
	phy-handle = <&phy0>;
	phy-mode = "rgmii-id";

	phy0: phy@0 {
		device_type = "ethernet-phy";
		reg = <0x0>;
		marvell,reg-init = <3 16 0xff00 0x1e 3 17 0xfff0 0x00>;
	};
};

&usb0 {
	xlnx,phy-reset-gpio = <&gpio0 7 0>;
};

&qspi {
	status = "okay";
	is-dual = <0>;
	num-cs = <1>;
	primary_flash: ps7-qspi@0 {
		#address-cells = <1>;
		#size-cells = <1>;
		spi-tx-bus-width = <1>;
		spi-rx-bus-width = <4>;
		compatible = "n25q256a", "jedec,spi-nor"; /* same as S25FL256 */
		reg = <0x0>;
		spi-max-frequency = <50000000>;
		partition@qspi-fsbl-uboot {
			label = "qspi-fsbl-uboot";
			reg = <0x0 0xE0000>; /* 896k */
		};
		partition@qspi-uboot-env {
			label = "qspi-uboot-env";
			reg = <0xE0000 0x20000>; /* 128k */
		};
		partition@qspi-linux {
			label = "qspi-linux";
			reg = <0x100000 0x500000>; /* 5M */
		};
		partition@qspi-device-tree {
			label = "qspi-device-tree";
			reg = <0x600000 0x20000>; /* 128k */
		};
		partition@qspi-rootfs {
			label = "qspi-rootfs";
			reg = <0x620000 0xCE0000>; /* 12875k */
		};
		partition@qspi-bitstream {
			label = "qspi-bitstream";
			reg = <0x1300000 0xD00000>; /* 13 M */
		};
	};
};

&sdhci0 {
	status = "okay";
	disable-wp;
};

**Error log**
ERROR: device-tree-xilinx+gitAUTOINC+73e546e312-r0 do_compile: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:devicetree_do_compile(d)
     0003:
File: '/opt/petalinux/2019.1/components/yocto/source/arm/layers/core/meta/classes/devicetree.bbclass', lineno: 131, function: devicetree_do_compile
     0127:            if not(os.path.isfile(dtspath)) or not(dts.endswith(".dts") or devicetree_source_is_overlay(dtspath)):
     0128:                continue # skip non-.dts files and non-overlay files
     0129:        except:
     0130:            continue # skip if can't determine if overlay
 *** 0131:        devicetree_compile(dtspath, includes, d)
     0132:}
     0133:
     0134:devicetree_do_install() {
     0135:    for DTB_FILE in `ls *.dtb *.dtbo`; do
File: '/opt/petalinux/2019.1/components/yocto/source/arm/layers/core/meta/classes/devicetree.bbclass', lineno: 119, function: devicetree_compile
     0115:        dtcargs += ["-i", i]
     0116:    dtcargs += ["-o", "{0}.{1}".format(dtname, "dtbo" if isoverlay else "dtb")]
     0117:    dtcargs += ["-I", "dts", "-O", "dtb", "{0}.pp".format(dts)]
     0118:    bb.note("Running {0}".format(" ".join(dtcargs)))
 *** 0119:    subprocess.run(dtcargs, check = True)
     0120:
     0121:python devicetree_do_compile() {
     0122:    includes = expand_includes("DT_INCLUDE", d)
     0123:    listpath = d.getVar("DT_FILES_PATH")
File: '/opt/petalinux/2019.1/components/yocto/source/arm/buildtools/sysroots/x86_64-petalinux-linux/usr/lib/python3.5/subprocess.py', lineno: 398, function: run
     0394:            raise
     0395:        retcode = process.poll()
     0396:        if check and retcode:
     0397:            raise CalledProcessError(retcode, process.args,
 *** 0398:                                     output=stdout, stderr=stderr)
     0399:    return CompletedProcess(process.args, retcode, stdout, stderr)
     0400:
     0401:
     0402:def list2cmdline(seq):
Exception: subprocess.CalledProcessError: Command '['dtc', '-R', '8', '-b', '0', '-p', '0x1000', '-i', '/home/ashish/1_pankaj/project/petalinux/2019.1/Hello_7R-Terminal/build/tmp/work-shared/plnx-zynq7/kernel-source/scripts/dtc/include-prefixes', '-i', '/home/ashish/1_pankaj/project/petalinux/2019.1/Hello_7R-Terminal/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/device-tree/xilinx+gitAUTOINC+73e546e312-r0', '-i', '/home/ashish/1_pankaj/project/petalinux/2019.1/Hello_7R-Terminal/build/../components/plnx_workspace/device-tree/device-tree', '-i', '/home/ashish/1_pankaj/project/petalinux/2019.1/Hello_7R-Terminal/build/tmp/work-shared/plnx-zynq7/kernel-source/arch/arm/boot/dts', '-i', '/home/ashish/1_pankaj/project/petalinux/2019.1/Hello_7R-Terminal/build/tmp/work-shared/plnx-zynq7/kernel-source/include', '-o', 'system-top.dtb', '-I', 'dts', '-O', 'dtb', 'system-top.dts.pp']' returned non-zero exit status 1

ERROR: device-tree-xilinx+gitAUTOINC+73e546e312-r0 do_compile: Function failed: devicetree_do_compile
ERROR: Logfile of failure stored in: /home/ashish/1_pankaj/project/petalinux/2019.1/Hello_7R-Terminal/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/device-tree/xilinx+gitAUTOINC+73e546e312-r0/temp/log.do_compile.1002
ERROR: Task (/opt/petalinux/2019.1/components/yocto/source/arm/layers/meta-xilinx/meta-xilinx-bsp/recipes-bsp/device-tree/device-tree.bb:do_compile) failed with exit code '1'

nunojsa, 2021-02-17:
Hi, this can be an issue with your device tree. Check the actual errors at `/home/ashish/1_pankaj/project/petalinux/2019.1/Hello_7R-Terminal/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/device-tree/xilinx+gitAUTOINC+73e546e312-r0/temp/log.do_compile.100`.

pnkjslr, 2021-02-18:
**Errors**
Error: /../components/plnx_workspace/device_tree/device_tree/system_top.dts:1.1-2 syntax errror
FATAL ERROR: Unable to parse input tree
ERROR: Error executing a python function in exec_func()

I have used zynq-adrv9364-z7020.dtsi to build my device-tree. Now as per binding rule I have also changed my model property same as zynq-adrv9364-z7020.dtsi. I do not understand why I'm getting this syntax error. I have also shared my custom-devicetree.dts with you.

nunojsa, 2021-02-18:
It looks like your devicetree has some problems... Where are you defining `/dts-v1/;`? I assume you are also defining `KERNEL_DTB` with your `custom-devicetree`?

pnkjslr, 2021-02-23:
Even after defining **/dts-v1/;** and KERNEL_DTB as **custom-devicetree**, still getting the same compilation error.  Can I use a **model** property which is not defined in device-tree bindings?

nunojsa, 2021-02-23:
So the `/dts-v1/;` needs to be the first thing you do in your top devicetree. Can you compile your devicetree without being in petalinux? I would suggest to first make sure you have a compilable devicetree before trying to add it in petalinux... You can, for example, clone our 2019_R1 linux tree and try to compile the dts in our kernel sources...

nunojsa, 2021-04-06:
will close this since there's no activity here...

closed 2021-04-06 by nunojsa