analogdevicesinc/meta-adi #94: petalinux build fails with meta-adi layers

chupalt, 2021-02-22:
**Environment (please complete the following information):**
 - Project to build: Xilinx 2019.1
 - Petalinux version: Desired 4.19 (got 5.4)
 - HDL branch: master
 - META-ADI branch: master
 - Using meta-adi petalinux docker image? No

**Describe the bug**
Build currently fails for Petalinux 2019.1 using meta-adi master.  It appears that sometime between December and now the analog devices linux repo has transitioned master from kernel version 4.19 to 5.4.  linux-xlnx_%..bbappend specifies the kernel version as 4.19, then pulls from the master branch of analogdevicesinc/linux.git and retrieves 5.4.  petalinux-build fails due to the version mismatch.  Changing KBRANCH = "adi-4.19.0" fixes the build

nunojsa, 2021-02-23:
Hi,

So, first thing. Do not use meta-adi master for now. It is not stable at this point (I' working in supporting 2020.1 petalinux). I'm also not sure what you mean in here:

> Project to build: Xilinx 2019.1

For instance, on meta-adi 2019_r2 branch there's a list of [projects](https://github.com/analogdevicesinc/meta-adi/tree/2019_R2/meta-adi-xilinx#supported-projects) supported by meta-adi. This is what I'm expecting to see in here...

> Petalinux version: Desired 4.19 (got 5.4)

This is not what I expect in here. This should be the petalinux version you want to use (e.g: 2019.1). It looks like that this is the kernel version...

From your description, you want to use meta-adi [2019_R2](https://github.com/analogdevicesinc/meta-adi/tree/2019_R2/meta-adi-xilinx). That branch is supposed to work with petalinux 2019.1 and it will pull the 4.19 kernel... Look also [here](https://github.com/analogdevicesinc/meta-adi/tree/2019_R2/meta-adi-xilinx#building-with-petalinux) to see what you shoud use with this branch...

chupalt, 2021-02-23:
Thanks,  I appreciate the update.  That helps.  I will offer that trying to find what version of petalinux, meta-adi, and hdl is compatible is nearly impossible.  The analog devices Wikis and readmes are full of tons of outdated information.  Take for example the primary petalinux wiki (references that the layer has been tested against petalinux 2018.3 - https://wiki.analog.com/resources/tools-software/linux-build/generic/petalinux).  Even the current readme from the master branch of meta-adi-xilinx says the master branch supports petalinux 2019.1 and the master branch of the hdl.  

It would make is so much easier if the readme at the top layer of meta-adi said what versions of petalinux are supported by  each meta-adi branch, and what version  of the hdl corresponded to that particular meta-adi branch.  The way it stands now, you wait for your build to break and then start trying to look through branches and commits to see what happened.  All along the readme's aren't updated.

nunojsa, 2021-02-26:
Well, the master branch is in active development and still not ready to say it supports petalinux 2020.1  (but yes, I could have temporarily change the README to reflect that should not be used at the moment)...

> It would make is so much easier if the readme at the top layer of meta-adi said what versions of petalinux are supported by each meta-adi branch

Maybe something to add. Nevertheless, if you go to the release branch you will directly see what version of petalinux and hdl is supported there (It's just one click)...

As for the wiki, here I have to do mea culpa. Yes, you are totally right and the information there is completely outdated (it's on my todo list to fix it). Github is the place to go and I reinforce that the REAME files are normally up to date. You just got the master branch in a release cycle and in transition to move on to a different version...

nunojsa, 2021-04-06:
will close this since there's no activity here...

closed 2021-04-06 by nunojsa