analogdevicesinc/meta-adi #102: Device tree error on PetaLinux 2020.1

SpontaneousDuck, 2021-07-16:
**Environment (please complete the following information):**
 - Project to build: fmcomms2.zcu102
 - Petalinux version: 202.1
 - HDL branch: master
 - META-ADI branch: master
 - Using meta-adi petalinux docker image? no

**Describe the bug**
Following the README exactly using master branches and a fresh PetaLinux 2020.1 install yields an error in building the device tree. The error is shown below. It looks like it is an error in the `do_configure` script for the device tree that `meta-adi-xilinx` is adding. 

**Error Log**
```
DEBUG: Executing shell function do_configure
MISC_ARG is  -hdf_type xsa -yamlconf /home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2020.1+gitAUTOINC+bc84458333-r0/device-tree.yaml
APP_ARG is  -app "device-tree"
Using xsct from: /tools/Xilinx/PetaLinux/2020.1/tools/xsct//bin/xsct
cmd is: xsct -sdx -nodisp /home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2020.1+gitAUTOINC+bc84458333-r0/dtgen.tcl -ws /home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/project-spec/configs/../../components/plnx_workspace/device-tree -pname device-tree -rp /home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2020.1+gitAUTOINC+bc84458333-r0/git -processor psu_cortexa53_0 -hdf /home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/deploy/images/zynqmp-generic/Xilinx-zynqmp-generic.xsa -arch 64  -app "device-tree"  -hdf_type xsa -yamlconf /home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2020.1+gitAUTOINC+bc84458333-r0/device-tree.yaml
INFO: [Hsi 55-2053] elapsed time for repository (/tools/Xilinx/PetaLinux/2020.1/tools/xsct/data/embeddedsw) loading 0 seconds
hsi::open_hw_design: Time (s): cpu = 00:00:12 ; elapsed = 00:00:11 . Memory (MB): peak = 1508.469 ; gain = 466.703 ; free physical = 914 ; free virtual = 25652
WARNING: Interrupt pin "gps_pps_irq" of IP block: "axi_ad9361" is not connected to any interrupt controller

WARNING: label 'amba_apu' found in existing tree
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: Frequency 33.330 used instead of 33.333
WARNING: psu_usb_xhci_0: No reset found
WARNING: label 'usb0' found in existing tree
WARNING: label 'amba_apu' found in existing tree
hsi::generate_target: Time (s): cpu = 00:00:15 ; elapsed = 00:00:21 . Memory (MB): peak = 1508.469 ; gain = 0.000 ; free physical = 1576 ; free virtual = 25673
INFO: [Hsi 55-2053] elapsed time for repository (/tools/Xilinx/PetaLinux/2020.1/tools/xsct/data/embeddedsw) loading 0 seconds
hsi::open_hw_design: Time (s): cpu = 00:00:12 ; elapsed = 00:00:13 . Memory (MB): peak = 1572.938 ; gain = 145.203 ; free physical = 1752 ; free virtual = 25903
/home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2020.1+gitAUTOINC+bc84458333-r0/temp/run.do_configure.1376353: 152: [: y: unexpected operator
sed: can't read /home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2020.1+gitAUTOINC+bc84458333-r0/zynqmp-zcu102-revA.dts: No such file or directory
WARNING: exit code 2 from a shell command.
ERROR: Execution of '/home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2020.1+gitAUTOINC+bc84458333-r0/temp/run.do_configure.1376353' failed with exit code 2:
MISC_ARG is  -hdf_type xsa -yamlconf /home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2020.1+gitAUTOINC+bc84458333-r0/device-tree.yaml
APP_ARG is  -app "device-tree"
Using xsct from: /tools/Xilinx/PetaLinux/2020.1/tools/xsct//bin/xsct
cmd is: xsct -sdx -nodisp /home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2020.1+gitAUTOINC+bc84458333-r0/dtgen.tcl -ws /home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/project-spec/configs/../../components/plnx_workspace/device-tree -pname device-tree -rp /home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2020.1+gitAUTOINC+bc84458333-r0/git -processor psu_cortexa53_0 -hdf /home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/deploy/images/zynqmp-generic/Xilinx-zynqmp-generic.xsa -arch 64  -app "device-tree"  -hdf_type xsa -yamlconf /home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2020.1+gitAUTOINC+bc84458333-r0/device-tree.yaml
INFO: [Hsi 55-2053] elapsed time for repository (/tools/Xilinx/PetaLinux/2020.1/tools/xsct/data/embeddedsw) loading 0 seconds
hsi::open_hw_design: Time (s): cpu = 00:00:12 ; elapsed = 00:00:11 . Memory (MB): peak = 1508.469 ; gain = 466.703 ; free physical = 914 ; free virtual = 25652
WARNING: Interrupt pin "gps_pps_irq" of IP block: "axi_ad9361" is not connected to any interrupt controller

WARNING: label 'amba_apu' found in existing tree
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: Frequency 33.330 used instead of 33.333
WARNING: psu_usb_xhci_0: No reset found
WARNING: label 'usb0' found in existing tree
WARNING: label 'amba_apu' found in existing tree
hsi::generate_target: Time (s): cpu = 00:00:15 ; elapsed = 00:00:21 . Memory (MB): peak = 1508.469 ; gain = 0.000 ; free physical = 1576 ; free virtual = 25673
INFO: [Hsi 55-2053] elapsed time for repository (/tools/Xilinx/PetaLinux/2020.1/tools/xsct/data/embeddedsw) loading 0 seconds
hsi::open_hw_design: Time (s): cpu = 00:00:12 ; elapsed = 00:00:13 . Memory (MB): peak = 1572.938 ; gain = 145.203 ; free physical = 1752 ; free virtual = 25903
/home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2020.1+gitAUTOINC+bc84458333-r0/temp/run.do_configure.1376353: 152: [: y: unexpected operator
sed: can't read /home/k.witham/source/zcu102_gnss/linux/zcu102_gnss/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2020.1+gitAUTOINC+bc84458333-r0/zynqmp-zcu102-revA.dts: No such file or directory
WARNING: exit code 2 from a shell command.
```

SpontaneousDuck, 2021-07-16:
I was able to resolve the issue by modifying the below line in `meta-adi-xilinx`.
https://github.com/analogdevicesinc/meta-adi/blob/db6a3a83d6b06fa85c8453ea634b161fdd542b96/meta-adi-xilinx/recipes-bsp/device-tree/device-tree.bbappend#L133

Since I am using kernel sources, removing the conditional and just always running the cp command fixed the problem and let me build successfully. Looks like this is a syntax thing but not really sure what is going on

nunojsa, 2021-07-19:
Hi,

This is odd as It's not reproducible in our CI. Doing a quick test the error only pop's up in case you're using `dash` as your shell. With bash, things work properly. As bash is the only supported shell when building petalinux, I guess this is not really an issue. Can you confirm if you're shell is `bash`? If not, I recommend to set as the default shell

pcercuei, 2021-07-19:
Should maybe be `[ "${USE_KERNEL_SOURCES}" == "y" ]`, otherwise if the variable is not defined, bash will read `[ == "y" ]`.

nunojsa, 2021-07-19:
Unless I'm missing something, it should always be defined because of [this](https://github.com/analogdevicesinc/meta-adi/blob/master/meta-adi-xilinx/recipes-bsp/device-tree/device-tree.bbappend#L91)... This has pretty much the same meaning to this assignment in bash `A={B:-foo}`. 

Anyways, having that or even `[[ ]]` might also not be a bad idea to be on the safe side...

SpontaneousDuck, 2021-07-19:
I did check and I am running `/usr/bin/bash`. On further investigation, I did use a different machine than normal and it ended up being an `Ubuntu 20.04.2 LTS` machine. That could also explain it since 20.04 is not supported either. Go ahead and close this if nothing needs to be changed!

nunojsa, 2021-07-19:
Yeah, I will close this for now. I may add some changes in the future to be on the safe side...

closed 2021-07-19 by nunojsa

nunojsa, 2023-01-09:
Hi,

This is still odd to me since AFAIK, `=` and `==` (this one is a bash extension) should behave the same as long as bash is being used. I mean, using `==` does pattern matching but in this case I don't really think that matters. From bash man page:

```
string1 == string2
string1 = string2
    True if the strings are equal.  = should be used with the test command for POSIX conformance.  When used with the [[ command, this performs pattern matching as described above (Compound Commands).
```
 
So to me, we would only care about this  if we want to be POSIX compliant (and to make sure our script would work in every shell).

Anyways, for sure there is something very subtle in here otherwise I would expect this to always fail...

That said, I'll try to remember to just change all tests to use `[[ ]]` as bash is a yocto requirement anyways... If you have the time, could you make sure that using "[[ ]]" instead of changing `==` to `=` also fixes the issue for you?

pcercuei, 2023-01-09:
@nunojsa I would think the problem is more that ${USE_KERNEL_SOURCES} is note in double-quotes. If not set, it resolves to `[ = "y" ]` which is a syntax error.

nunojsa, 2023-01-09:
Look at my comment in here https://github.com/analogdevicesinc/meta-adi/issues/102#issuecomment-882389864

But maybe, in the end, you are right and there's some issue where it's no guaranteed that ${USE_KERNEL_SOURCES}  is always set

pcercuei, 2023-01-09:
Well it won't be defined if you set `USE_KERNEL_SOURCES=`

nunojsa, 2023-01-09:
That is true in theory but I would also argue that's a user dumb mistake... That said, it's also true that users (me included :D) are stupid and can very well do that so the code should be prepared for that possibility. Anyways, using `[[ ]]` would also work even if the variable is not set.

Also note that this specific problem should not be because `USE_KERNEL_SOURCES` is not set otherwise this change:

```
-       [ ${USE_KERNEL_SOURCES} == "y" ] && cp -rf "${DTS_INCLUDE_PATH}/"* "${WORKDIR}/"
+       [ ${USE_KERNEL_SOURCES} = "y" ] && cp -rf "${DTS_INCLUDE_PATH}/"* "${WORKDIR}/"
```

would also fail...

mestofer, 2023-01-11:
Changing the line to this does not work for me.
```
[[ ${USE_KERNEL_SOURCES} == "y" ]] && cp -rf "${DTS_INCLUDE_PATH}/"* "${WORKDIR}/"
```
I get...
```
<path>/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/temp/run.do_configure.63956: 206: [[: not found
```
I think the issue is that I am using the bash shell but I must be starting it in posix mode. I'm not doing that intentionally or because I necessarily want to. I figure perhaps that was changed in the default behavior for ubuntu 20.
So ```[ x = y ]``` is bash and posix compliant which would work for this particular comparison but if you wanted to actually use ```[[ x == y]]``` to do pattern matching that won't work if anyone is using a posix compliant bash shell.
Not sure what all of that means for this code and whether or not it should change. I think perhaps the solution might be to tell the user to use the bash shell and not in posix mode. I'm trying to figure out how to do that (I can't figure out where that option is being set or passed).
Thanks for your time and help.

nunojsa, 2023-01-11:
Hi,

Thanks for testing it out. Well, I'm fairly sure yocto used to have bash (petalinux still has) as a depency but apparently that changed:

```
When you create these types of functions in your recipe or class files, you need to follow the shell programming rules.
The scripts are executed by /bin/sh, which may not be a bash shell but might be something such as dash. You should not
use Bash-specific script (bashisms).
``` 

So yes, this is a no go for using `[[ ]]` and using a single `=` because using `==` might not workout as expected for other shells... Alright, I'll have to go through all the do_configure function to remove all the bashisms

commit 041d8f64 (2023-01-31): mata-adi-xilinx: Replace comparison operator with POSIX compatible

It seems that scripts in Yocto are executed by a non-bash
compatible shell.

I suggest to replace '==' and '=', because they are compatible
expressions a executable also by other shells if they are POSIX
matching.

closes: #102