analogdevicesinc/meta-adi #104: Kernel build failed for daq2_zcu102 on petalinux 2021.1 and meta-adi staging/2021_1-support.

gudokim, 2021-11-05:
**Environment (please complete the following information):**
 - Project to build: daq2_zcu102
 - Petalinux version: 2021.1
 - HDL branch: master
 - META-ADI branch: staging/2021_1-support
 - Using meta-adi petalinux docker image? no
 - OS : ubuntu 20.04.1

**Describe the bug**
Kernel build failed for daq2_zcu102 on petalinux 2021.1 and meta-adi staging/2021_1-support.

I got the following errors.
 - ERROR: device-tree-xilinx-v2021.1+gitAUTOINC+f2642103e9-r0 do_compile: Error executing a python function in exec_python_func() autogenerated:
 - ERROR: Input tree has errors, aborting (use -f to force output)
 - ERROR: Logfile of failure stored in: 
 - ERROR: Task (/home/aa/petalinux-project/daq2_zcu102/components/yocto/layers/meta-xilinx/meta-xilinx-bsp/recipes-bsp/device-tree/device-tree.bb:do_compile) failed with exit code '1'

** Kernel build procedure **
```
petalinux-create -t project --template zynqMP --name daq2_zcu102

petalinux-config --get-hw-description=/home/aa/win-hdl/hdl/projects/daq2/zcu102/daq2_zcu102.sdk
# meta-adi layer0
/home/aa/petalinux-project/meta-adi/meta-adi-core
# meta-adi layer1
/home/aa/petalinux-project/meta-adi/meta-adi-xilinx

echo "KERNEL_DTB=\"zynqmp-zcu102-rev10-fmcdaq2\"" >> project-spec/meta-user/conf/petalinuxbsp.conf

petalinux-build
```

**Screenshots**
[image]
[image]

**Additional context**
Add any other context about the problem here.

nunojsa, 2021-11-05:
Hi,

Thanks for reporting!

This looks like a valid error... I'm still missing the CI setup for all the supported projects in 2021 so this is kind of expected for now. Most likely there's a mismatch between the pl.dtsi and the pl-delete-nodes. Anyways, since you're reporting this, I will run this project in the next days and see if the fix is as easy as I think it is.

gudokim, 2021-11-10:
Hi,

Thanks for replying.
Did you run and check it?

oQuQu, 2021-11-10:
ditto

nunojsa, 2021-11-10:
Yes,

And I can reproduce the issue... It should not be hard to fix. I will push a fix to the staging branch today if nothing unexpected pops up.

nunojsa, 2021-11-12:
This should be fixed already... The staging branch is ready to merge with master, it will happen soon.

gudokim, 2021-11-15:
Wow, I've just checked that built project successfully.
Thanks.

gudokim, 2022-06-05:
Hi~

I tried built the petalinux project same as before but the only difference was the project name.
The error message similar to the previous one.

Regards

Gudo

- Project to build: rlpm_zcu102
- Petalinux version: 2021.1
- HDL branch: master
- META-ADI branch: staging/2021_1-support
- Using meta-adi petalinux docker image? no
- OS : ubuntu 20.04.1

```
petalinux-create -t project --template zynqMP --name rlpm_zcu102

petalinux-config --get-hw-description=/home/aa/win-hdl/hdl/projects/daq2/zcu102/rlpm_zcu102.sdk
# meta-adi layer0
/home/aa/petalinux-project/meta-adi/meta-adi-core
# meta-adi layer1
/home/aa/petalinux-project/meta-adi/meta-adi-xilinx

echo "KERNEL_DTB=\"zynqmp-zcu102-rev10-fmcdaq2\"" >> project-spec/meta-user/conf/petalinuxbsp.conf

petalinux-buil
```

```
ERROR: device-tree-xilinx-v2021.1+gitAUTOINC+f2642103e9-r0 do_compile: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:devicetree_do_compile(d)
     0003:
File: '/home/aa/petalinux-project/rlpm_zcu102/components/yocto/layers/core/meta/classes/devicetree.bbclass', lineno: 131, function: devicetree_do_compile
     0127:            if not(os.path.isfile(dtspath)) or not(dts.endswith(".dts") or devicetree_source_is_overlay(dtspath)):
     0128:                continue # skip non-.dts files and non-overlay files
     0129:        except:
     0130:            continue # skip if can't determine if overlay
 *** 0131:        devicetree_compile(dtspath, includes, d)
     0132:}
     0133:
     0134:devicetree_do_install() {
     0135:    for DTB_FILE in `ls *.dtb *.dtbo`; do
File: '/home/aa/petalinux-project/rlpm_zcu102/components/yocto/layers/core/meta/classes/devicetree.bbclass', lineno: 119, function: devicetree_compile
     0115:        dtcargs += ["-i", i]
     0116:    dtcargs += ["-o", "{0}.{1}".format(dtname, "dtbo" if isoverlay else "dtb")]
     0117:    dtcargs += ["-I", "dts", "-O", "dtb", "{0}.pp".format(dts)]
     0118:    bb.note("Running {0}".format(" ".join(dtcargs)))
 *** 0119:    subprocess.run(dtcargs, check = True, stdout=subprocess.PIPE, stderr=subprocess.STDOUT)
     0120:
     0121:python devicetree_do_compile() {
     0122:    includes = expand_includes("DT_INCLUDE", d)
     0123:    listpath = d.getVar("DT_FILES_PATH")
File: '/home/aa/petalinux/2021.1/components/yocto/buildtools/sysroots/x86_64-petalinux-linux/usr/lib/python3.8/subprocess.py', lineno: 512, function: run
     0508:            # We don't call process.wait() as .__exit__ does that for us.
     0509:            raise
     0510:        retcode = process.poll()
     0511:        if check and retcode:
 *** 0512:            raise CalledProcessError(retcode, process.args,
     0513:                                     output=stdout, stderr=stderr)
     0514:    return CompletedProcess(process.args, retcode, stdout, stderr)
     0515:
     0516:
Exception: subprocess.CalledProcessError: Command '['dtc', '-@', '-p', '0x1000', '-@', '-i', '/home/aa/petalinux-project/rlpm_zcu102/build/tmp/work-shared/zynqmp-generic/kernel-source/scripts/dtc/include-prefixes', '-i', '/home/aa/petalinux-project/rlpm_zcu102/project-spec/configs/../../components/plnx_workspace/device-tree/device-tree', '-i', '/home/aa/petalinux-project/rlpm_zcu102/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+f2642103e9-r0', '-o', 'system-top.dtb', '-I', 'dts', '-O', 'dtb', 'system-top.dts.pp']' returned non-zero exit status 1.

Subprocess output:
Error: /home/aa/petalinux-project/rlpm_zcu102/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+f2642103e9-r0/pl-delete-nodes-zynqmp-zcu102-rev10-fmcdaq2.dtsi:3.15-30 Label or path axi_ad9144_dma not found
Error: /home/aa/petalinux-project/rlpm_zcu102/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+f2642103e9-r0/pl-delete-nodes-zynqmp-zcu102-rev10-fmcdaq2.dtsi:5.15-49 Label or path axi_ad9144_offload_i_data_offload not found
Error: /home/aa/petalinux-project/rlpm_zcu102/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+f2642103e9-r0/pl-delete-nodes-zynqmp-zcu102-rev10-fmcdaq2.dtsi:8.15-30 Label or path axi_ad9680_dma not found
Error: /home/aa/petalinux-project/rlpm_zcu102/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+f2642103e9-r0/pl-delete-nodes-zynqmp-zcu102-rev10-fmcdaq2.dtsi:10.15-49 Label or path axi_ad9680_offload_i_data_offload not found
FATAL ERROR: Syntax error parsing input tree

ERROR: Logfile of failure stored in: /home/aa/petalinux-project/rlpm_zcu102/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+f2642103e9-r0/temp/log.do_compile.1889877
ERROR: Task (/home/aa/petalinux-project/rlpm_zcu102/components/yocto/layers/meta-xilinx/meta-xilinx-bsp/recipes-bsp/device-tree/device-tree.bb:do_compile) failed with exit code '1'
```

closed 2022-06-15 by gudokim