analogdevicesinc/meta-adi #105: Kernel build failed for zynq-zed-adv7511 on petalinux 2021.1 and meta-adi staging/2021_1-support.

oQuQu, 2021-11-12:
**Environment (please complete the following information):**
 - Project to build: zynq-zed-adv7511
 - Petalinux version: 2021.1
 - HDL branch: master
 - META-ADI branch: staging/2021_1-support
 - Using meta-adi petalinux docker image? No

**Describe the bug**
Kernel build failed for zynq-zed-adv7511 on petalinux 2021.1 and meta-adi staging/2021_1-support. 
( can not find <zynq-zed.dtsi> )

**Screenshots**
[image]
[image]

nunojsa, 2021-11-12:
That is odd. I recently ran a successful build on this project. Just started a new one now to see if something changed... Moreover, your error looks very weird (that file should exist). Did you already tried a clean build?

nunojsa, 2021-11-12:
Well, the build went just fine for me... So I would really suggest for you to make a clean build of the project.

nunojsa, 2021-11-16:
Any updates here? I will wait a few days more and close this issue if nothing new comes...

closed 2021-11-30 by nunojsa

chegbeli, 2021-12-03:
This issue might appear if /bin/sh is not set to bash. 
In order to change set it properly please try this:
` sudo ln -sf bash /bin/sh`