analogdevicesinc/meta-adi #108: ERROR: device-tree-xilinx-v2021.1

sarosse, 2022-02-11:
**Environment (please complete the following information):**
 - Project to build: ZCU102
 - Petalinux version: 2021.1
 - HDL branch:2019_r2
 - META-ADI branch:master
 - Using meta-adi petalinux docker image?No

**Describe the bug**
Buid device tree fails , 

**Screenshots**
ERROR: device-tree-xilinx-v2021.1+gitAUTOINC+252758eb1f-r0 do_configure: Execution of '/home/d59006/Petalinux_projects/Project_0_0_0_68/build/tmp/work/zcu102_zynqmp-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/temp/run.do_configure.134877' failed with exit code 2:
MISC_ARG is  -hdf_type xsa -yamlconf /home/d59006/Petalinux_projects/Project_0_0_0_68/build/tmp/work/zcu102_zynqmp-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/device-tree.yaml
APP_ARG is  -app "device-tree"
Using xsct from: /home/d59006/petalinux/tools/xsct//bin/xsct
cmd is: xsct -sdx -nodisp /home/d59006/Petalinux_projects/Project_0_0_0_68/build/tmp/work/zcu102_zynqmp-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/dtgen.tcl -ws /home/d59006/Petalinux_projects/Project_0_0_0_68/project-spec/configs/../../components/plnx_workspace/device-tree -pname device-tree -rp /home/d59006/Petalinux_projects/Project_0_0_0_68/build/tmp/work/zcu102_zynqmp-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/git -processor_ip psu_cortexa53 -hdf /home/d59006/Petalinux_projects/Project_0_0_0_68/build/tmp/deploy/images/zcu102-zynqmp/Xilinx-zcu102-zynqmp.xsa -arch 64   -app "device-tree"  -hdf_type xsa -yamlconf /home/d59006/Petalinux_projects/Project_0_0_0_68/build/tmp/work/zcu102_zynqmp-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/device-tree.yaml
INFO: [Hsi 55-2053] elapsed time for repository (/home/d59006/petalinux/tools/xsct/data/embeddedsw) loading 0 seconds
hsi::open_hw_design: Time (s): cpu = 00:00:10 ; elapsed = 00:00:08 . Memory (MB): peak = 2168.305 ; gain = 687.211 ; free physical = 13444 ; free virtual = 19143
WARNING: Clock pin "clk" of IP block "axi_adrv9009_rx_clkgen" is not connected to any of the pl_clk"

WARNING: Clock pin "fifo_wr_clk" of IP block "axi_adrv9009_rx_dma" is not connected to any of the pl_clk"

WARNING: Clock pin "core_clk" of IP block "axi_adrv9009_rx_jesd_rx_axi" is not connected to any of the pl_clk"

WARNING: Clock pin "clk" of IP block "axi_adrv9009_rx_os_clkgen" is not connected to any of the pl_clk"

WARNING: Clock pin "fifo_wr_clk" of IP block "axi_adrv9009_rx_os_dma" is not connected to any of the pl_clk"

WARNING: Clock pin "core_clk" of IP block "axi_adrv9009_rx_os_jesd_rx_axi" is not connected to any of the pl_clk"

WARNING: Clock pin "clk" of IP block "axi_adrv9009_tx_clkgen" is not connected to any of the pl_clk"

WARNING: Clock pin "core_clk" of IP block "axi_adrv9009_tx_jesd_tx_axi" is not connected to any of the pl_clk"

WARNING: the label "rx_adrv9009_tpl_core_adc_tpl_core" length is 33 characters which is greater than default 31 characters as per DT SPEC...user need to fix the label

WARNING: Clock pin "link_clk" of IP block "rx_adrv9009_tpl_core_adc_tpl_core" is not connected to any of the pl_clk"

WARNING: the label "rx_os_adrv9009_tpl_core_adc_tpl_core" length is 36 characters which is greater than default 31 characters as per DT SPEC...user need to fix the label

WARNING: Clock pin "link_clk" of IP block "rx_os_adrv9009_tpl_core_adc_tpl_core" is not connected to any of the pl_clk"

WARNING: the label "tx_adrv9009_tpl_core_dac_tpl_core" length is 33 characters which is greater than default 31 characters as per DT SPEC...user need to fix the label

WARNING: Clock pin "link_clk" of IP block "tx_adrv9009_tpl_core_dac_tpl_core" is not connected to any of the pl_clk"

WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: quotes to be removed or use 2018.1 version for &sata params param0..param7
WARNING: Frequency 33.333333333 used instead of 33.333
WARNING: Frequency 33.333 used instead of 27.00
zocl:false
ext_platform:
WARNING: psu_usb_xhci_0: No reset found
WARNING: label 'usb0' found in existing tree
WARNING: the label "rx_adrv9009_tpl_core_adc_tpl_core" length is 33 characters which is greater than default 31 characters as per DT SPEC...user need to fix the label

WARNING: the label "rx_adrv9009_tpl_core_adc_tpl_core" length is 33 characters which is greater than default 31 characters as per DT SPEC...user need to fix the label

WARNING: the label "rx_os_adrv9009_tpl_core_adc_tpl_core" length is 36 characters which is greater than default 31 characters as per DT SPEC...user need to fix the label

WARNING: the label "rx_os_adrv9009_tpl_core_adc_tpl_core" length is 36 characters which is greater than default 31 characters as per DT SPEC...user need to fix the label

WARNING: the label "tx_adrv9009_tpl_core_dac_tpl_core" length is 33 characters which is greater than default 31 characters as per DT SPEC...user need to fix the label

WARNING: the label "tx_adrv9009_tpl_core_dac_tpl_core" length is 33 characters which is greater than default 31 characters as per DT SPEC...user need to fix the label

hsi::generate_target: Time (s): cpu = 00:00:29 ; elapsed = 00:00:31 . Memory (MB): peak = 2168.305 ; gain = 0.000 ; free physical = 13440 ; free virtual = 19140
/home/d59006/Petalinux_projects/Project_0_0_0_68/build/tmp/work/zcu102_zynqmp-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/temp/run.do_configure.134877: 194: [: y: unexpected operator
sed: can't read /home/d59006/Petalinux_projects/Project_0_0_0_68/build/tmp/work/zcu102_zynqmp-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/zynqmp-zcu102-revA.dts: No such file or directory
WARNING: exit code 2 from a shell command.

ERROR: Logfile of failure stored in: /home/d59006/Petalinux_projects/Project_0_0_0_68/build/tmp/work/zcu102_zynqmp-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/temp/log.do_configure.134877

nunojsa, 2022-03-07:
Hi,

Sorry for the late reply. The master branch is supposed to be used with hdl master and not 2019_r2...

sarosse, 2022-03-08:
Hi Nunojsa , that is my mustake ... i had used hdl master :  

    Project to build: ZCU102
    Petalinux version: 2021.1
    HDL branch:master
    META-ADI branch:master
    Ubuntu 20.04.4
    Using meta-adi petalinux docker image?No

I follow all the procedures descibed in meta-adi but it gives me alwas this error  

Hi Nunojsa , that was my mistake ... i had used hdl master :  

    Project to build: ZCU102
    Petalinux version: 2021.1
    HDL branch:master
    META-ADI branch:master
    Ubuntu 20.04.4
    Using meta-adi petalinux docker image?No

I followed all the procedures described in meta-adi ... I have cleaned the project, I made a new project , anyway I tryed a lots of ways but it always gives me  this error  ...

sarosse, 2022-03-08:
Hi Nunojsa , that is my mistake ... i had used hdl master :  

    Project to build: ZCU102
    Petalinux version: 2021.1
    HDL branch:master
    META-ADI branch:master
    Ubuntu 20.04.4
    Using meta-adi petalinux docker image?No

I followed all the procedures descibed in meta-adi ... I have cleaned the project I made a new project , anyway I tryed a lots of ways but it always gives me  this error  ...

sarosse, 2022-03-08:
Hi Nunojsa , that was my mistake ... i had used hdl master :  

    Project to build: ZCU102
    Petalinux version: 2021.1
    HDL branch:master
    META-ADI branch:master
    Ubuntu 20.04.4
    Using meta-adi petalinux docker image?No

I followed all the procedures described in meta-adi ... I have cleaned the project, I made a new project , anyway I tryed a lots of ways but it always gives me  this error  ... :(

nunojsa, 2022-03-08:
Ok, I will need a bit more detail on which project are you trying to build? ZCU102 is very generic. Can you point me to the project you are trying to build from this [table](https://github.com/analogdevicesinc/meta-adi/tree/master/meta-adi-xilinx#supported-projects) and I will try it on my end?

sarosse, 2022-03-08:
sorry for forgetting important details and thank you very much for your quick response
the project is: hdl / projects / adrv9009 / zcu102 /,
and I am using the ADRV9008-2W / PCBZ analog device board on the ZCU102 board
I followed this page to build hdl:
https://wiki.analog.com/resources/fpga/docs/build
then I followed this page to build the project:
https://github.com/analogdevicesinc/meta-adi/tree/master/meta-adi-xilinx
I used: zcu102-rev10-adrv9008-2-jesd204-fsm in the DTB kernel.

adrv9009_zcu102 | zynqmp-zcu102-rev10-adrv9008-2-jesd204-fsm

BR

nunojsa, 2022-03-08:
So, things work for me... What do you see when you do `ls -la /bin/sh`? I'm suspecting you shell is dash...

sarosse, 2022-03-08:
I thought I had already changed it ... but you have right !! it gives me dash 
ls -al ssh
lrwxrwxrwx 1 root root 4 ott 29 16:50 sh -> dash
... Thank you very much !!!

nunojsa, 2022-03-09:
Ok, closing this issue then

closed 2022-03-09 by nunojsa

pollai2, 2023-11-15:
i have got also same error how to change dash to bash.

pollai2, 2023-11-15:
after changee to bash also showig same error.

pollai2, 2023-11-15:
After changes aslo getting error  regarding device-treee.how to reslove this issue

ERROR: device-tree-xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0 do_configure: Error: Could not find "pl-delete-nodes-delete-nodes-zynq-zc706-adv7511-adrv9371.dtsi.dtsi" in "/home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/build/tmp/work/zc706-xilinx-linux-gnueabi/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0"
ERROR: device-tree-xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0 do_configure: Execution of '/home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/build/tmp/work/zc706-xilinx-linux-gnueabi/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/temp/run.do_configure.213422' failed with exit code 1:
MISC_ARG is  -hdf_type xsa -yamlconf /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/build/tmp/work/zc706-xilinx-linux-gnueabi/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/device-tree.yaml
APP_ARG is  -app "device-tree"
Using xsct from: /home/icomm/petalinux_new/tools/xsct//bin/xsct
cmd is: xsct -sdx -nodisp /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/build/tmp/work/zc706-xilinx-linux-gnueabi/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/dtgen.tcl -ws /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/project-spec/configs/../../components/plnx_workspace/device-tree -pname device-tree -rp /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/build/tmp/work/zc706-xilinx-linux-gnueabi/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/git -processor_ip ps7_cortexa9 -hdf /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/build/tmp/deploy/images/zynq-generic/Xilinx-zynq-generic.xsa -arch 32   -app "device-tree"  -hdf_type xsa -yamlconf /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/build/tmp/work/zc706-xilinx-linux-gnueabi/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/device-tree.yaml
INFO: [Hsi 55-2053] elapsed time for repository (/home/icomm/petalinux_new/tools/xsct/data/embeddedsw) loading 0 seconds
zocl:false
ext_platform:
hsi::generate_target: Time (s): cpu = 00:00:04 ; elapsed = 00:00:07 . Memory (MB): peak = 2152.941 ; gain = 3.625 ; free physical = 1240 ; free virtual = 2354
INFO: [Hsi 55-2053] elapsed time for repository (/home/icomm/petalinux_new/tools/xsct/data/embeddedsw) loading 0 seconds
WARNING: /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/build/tmp/work/zc706-xilinx-linux-gnueabi/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/temp/run.do_configure.213422:293 exit 1 from 'exit 1'
WARNING: Backtrace (BB generated script): 
	#1: bbfatal, /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/build/tmp/work/zc706-xilinx-linux-gnueabi/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/temp/run.do_configure.213422, line 293
	#2: do_configure, /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/build/tmp/work/zc706-xilinx-linux-gnueabi/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/temp/run.do_configure.213422, line 200
	#3: main, /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/build/tmp/work/zc706-xilinx-linux-gnueabi/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/temp/run.do_configure.213422, line 297

Backtrace (metadata-relative locations):
	#1: bbfatal, /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/components/yocto/layers/core/meta/classes/logging.bbclass, line 60
	#2: do_configure, /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/components/yocto/layers/meta-xilinx-tools/classes/xsctbase.bbclass, line 78
ERROR: Logfile of failure stored in: /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/build/tmp/work/zc706-xilinx-linux-gnueabi/device-tree/xilinx-v2021.2+gitAUTOINC+c0acd8f064-r0/temp/log.do_configure.213422
ERROR: Task (/home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/components/yocto/layers/meta-xilinx/meta-xilinx-bsp/recipes-bsp/device-tree/device-tree.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2864 tasks of which 2757 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/components/yocto/layers/meta-xilinx/meta-xilinx-bsp/recipes-bsp/device-tree/device-tree.bb:do_configure
Summary: There was 1 WARNING message shown.
Summary: There were 4 ERROR messages shown, returning a non-zero exit code.
ERROR: Failed to build project. Check the /home/icomm/petalinux_new/ad9371/xilinx-zc706-2021.2/build/build.log file for more details...