analogdevicesinc/meta-adi #109: Zu11EG + fmcomms 8

noelpedro, 2022-03-05:
Hi guys will a yocto layer for zu11EG + fmcomms 8 be Realeased ? Or can I use the existing zu11EG adrv2crr_fmc device tree and add the two extra adrv9009 devices that come with the fmcomms8? 

thanks

nunojsa, 2022-03-07:
Hi,

I can add this project to meta-adi (only in master branch) as it makes sense to be there...

nunojsa, 2022-03-07:
Branch pushed [here](https://github.com/analogdevicesinc/meta-adi/tree/dev/zu11EG-fmcomms8-support). @noelpedro can you give this a try and make sure all the IIO devices come up?

nunojsa, 2022-03-14:
@noelpedro, any chance to give this a try?

noelpedro, 2022-03-17:
@nunojsa not yet.

I’ll try next week.

Unless there’s a later sable release.

The HDL latest stable branch is 2019_r2 which is good for vivado 2019.1.
I’ve been using also petalinux 2019.1
For meta adi’s 2019_R2.

I had issues with master meta-ADI with petalinux 2019.1.

The above is in regards to adrv9009ZU11EG with adrv2fmc

noelpedro, 2022-03-22:
@nunojsa Attempting to do so now with petalinux 2019.1

```
ERROR: Server log for this session (/home/user/work/adrv9009zu11EG_FMCOMMS8/adv9009zu11EG-fmcomms8/build/bitbake-cookerdaemon.log):
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Unable to start bitbake server
ERROR: Server log for this session (/home/user/work/adrv9009zu11EG_FMCOMMS8/adv9009zu11EG-fmcomms8/build/bitbake-cookerdaemon.log):
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Unable to start bitbake server
ERROR: Server log for this session (/home/user/work/adrv9009zu11EG_FMCOMMS8/adv9009zu11EG-fmcomms8/build/bitbake-cookerdaemon.log):
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Unable to start bitbake server
ERROR: Server log for this session (/home/user/work/adrv9009zu11EG_FMCOMMS8/adv9009zu11EG-fmcomms8/build/bitbake-cookerdaemon.log):
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Unable to start bitbake server
ERROR: Server log for this session (/home/user/work/adrv9009zu11EG_FMCOMMS8/adv9009zu11EG-fmcomms8/build/bitbake-cookerdaemon.log):
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Unable to start bitbake server
ERROR: Server log for this session (/home/user/work/adrv9009zu11EG_FMCOMMS8/adv9009zu11EG-fmcomms8/build/bitbake-cookerdaemon.log):
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Unable to start bitbake server
ERROR: Server log for this session (/home/user/work/adrv9009zu11EG_FMCOMMS8/adv9009zu11EG-fmcomms8/build/bitbake-cookerdaemon.log):
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Unable to start bitbake server
ERROR: Server log for this session (/home/user/work/adrv9009zu11EG_FMCOMMS8/adv9009zu11EG-fmcomms8/build/bitbake-cookerdaemon.log):
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: thud (layer is compatible with gatesgarth)
[INFO] generating u-boot configuration files
[INFO] generating kernel configuration files
[INFO] generating kconfig for Rootfs
[INFO] silentconfig rootfs
[INFO] generating petalinux-user-image.bb

```

nunojsa, 2022-03-22:
That is expected. Note that I said that this would only be supported in the master branch. Typically, only Bugfixes are backported to release branches

noelpedro, 2022-03-22:
SO proceed with petalinux 2021.1? with vivado 2019.1 for HDL?

noelpedro, 2022-03-22:
Using Master branch with petalinux 2021.1
Built master hdl with vivado 2021.1

Getting Errors

```
NOTE: Executing Tasks
ERROR: device-tree-xilinx-v2021.1+gitAUTOINC+252758eb1f-r0 do_configure: Execution of '/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/temp/run.do_configure.4747' failed with exit code 2:
MISC_ARG is  -hdf_type xsa -yamlconf /home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/device-tree.yaml
APP_ARG is  -app "device-tree"
Using xsct from: /home/user/Downloads/petalinux_2021_1/tools/xsct//bin/xsct
cmd is: xsct -sdx -nodisp /home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/dtgen.tcl -ws /home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/project-spec/configs/../../components/plnx_workspace/device-tree -pname device-tree -rp /home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/git -processor_ip psu_cortexa53 -hdf /home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/build/tmp/deploy/images/zynqmp-generic/Xilinx-zynqmp-generic.xsa -arch 64   -app "device-tree"  -hdf_type xsa -yamlconf /home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/device-tree.yaml
INFO: [Hsi 55-2053] elapsed time for repository (/home/user/Downloads/petalinux_2021_1/tools/xsct/data/embeddedsw) loading 0 seconds
hsi::open_hw_design: Time (s): cpu = 00:00:09 ; elapsed = 00:00:07 . Memory (MB): peak = 2198.906 ; gain = 708.484 ; free physical = 7805 ; free virtual = 28995
WARNING: no s_axi_aclk for clockwizard IP block: " dma_clk_wiz"

```

nunojsa, 2022-03-23:
> Using meta-adi fmcomms8 branch with petalinux 2021.1
Built master hdl with vivado 2021.1

Yah, this should be the combo to use. Error in do_configure are unexpected as It worked on my side (building at least). I will give it another try on my end to make sure nothing changed.

Can you paste the log of your do_configure task?

nunojsa, 2022-03-23:
Well things are building for me...

```
petalinux-build 
[INFO] Sourcing buildtools
[INFO] Building project
[INFO] Sourcing build environment
[INFO] Generating workspace directory
INFO: bitbake petalinux-image-minimal
NOTE: Started PRServer with DBfile: /home/nsa/work/petalinux/2021_R1/adrv9009zu11eg_adrv2crr_fmcomms8/build/cache/prserv.sqlite3, IP: 127.0.0.1, PORT: 33097, PID: 182013
Loading cache: 100% |                                                                                                                                                                                                                                                                                         | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |########################################################################################################################################################################################################################################################################################| Time: 0:00:44
Parsing of 3475 .bb files complete (0 cached, 3475 parsed). 5110 targets, 223 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
NOTE: Fetching uninative binary shim file:///home/nsa/work/petalinux/2021_R1/adrv9009zu11eg_adrv2crr_fmcomms8/components/yocto/downloads/uninative/5ec5a9276046e7eceeac749a18b175667384e1f445cd4526300a41404d985a5b/x86_64-nativesdk-libc.tar.xz;sha256sum=5ec5a9276046e7eceeac749a18b175667384e1f445cd4526300a41404d985a5b (will check PREMIRRORS first)
Initialising tasks: 100% |#####################################################################################################################################################################################################################################################################################| Time: 0:00:08
Checking sstate mirror object availability: 100% |#############################################################################################################################################################################################################################################################| Time: 0:00:20
Sstate summary: Wanted 1844 Found 1514 Missed 330 Current 0 (82% match, 0% complete)
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 5599 tasks of which 4137 didn't need to be rerun and all succeeded.
INFO: Failed to copy built images to tftp dir: /tftpboot
[INFO] Successfully built project
petalinux@38cf0e8f6180:/home/nsa/work/petalinux/2021_R1/adrv9009zu11eg_adrv2crr_fmcomms8$ 
```

This is also my user conf file

```
petalinux@38cf0e8f6180:/home/nsa/work/petalinux/2021_R1/adrv9009zu11eg_adrv2crr_fmcomms8$ cat project-spec/meta-user/conf/petalinuxbsp.conf 
#User Configuration

#OE_TERMINAL = "tmux"
KERNEL_DTB = "zynqmp-adrv9009-zu11eg-revb-adrv2crr-fmc-revb-sync-fmcomms8-jesd204-fsm"
```

noelpedro, 2022-03-23:
interesting I only had `KERNEL_DTB` defined in `meta-adi-xilinx/recipes-bsp/device-tree/device-tree.bbappend`

noelpedro, 2022-03-23:
```
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['virtual:native:/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/core/meta/recipes-bsp/u-boot/u-boot-tools_2020.07.bb:do_populate_sysroot', '/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/core/meta/recipes-devtools/binutils/binutils-cross_2.35.1.bb:do_populate_sysroot', 'virtual:native:/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/core/meta/recipes-kernel/dtc/dtc_1.6.0.bb:do_populate_sysroot', 'virtual:native:/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/core/meta/recipes-devtools/bison/bison_3.7.2.bb:do_populate_sysroot', 'virtual:native:/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/core/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/core/meta/recipes-extended/xz/xz_5.2.5.bb:do_populate_sysroot', 'virtual:native:/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/meta-xilinx/meta-xilinx-bsp/recipes-kernel/dtc/python3-dtc_1.5.1.bb:do_populate_sysroot', '/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/core/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot', '/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/core/meta/recipes-kernel/kmod/kmod-native_git.bb:do_populate_sysroot', 'virtual:native:/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/core/meta/recipes-extended/bc/bc_1.07.1.bb:do_populate_sysroot', 'virtual:native:/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/core/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/core/meta/recipes-devtools/python/python3_3.8.5.bb:do_populate_sysroot', '/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/core/meta/recipes-devtools/gcc/gcc-cross_10.2.bb:do_populate_sysroot', '/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/core/meta/recipes-devtools/gcc/libgcc_10.2.bb:do_populate_sysroot', 'virtual:native:/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/meta-openembedded/meta-python/recipes-devtools/python/python3-pyyaml_5.3.1.bb:do_populate_sysroot', '/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/components/yocto/layers/core/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot']
NOTE: Installed into sysroot: []
NOTE: Skipping as already exists in sysroot: ['u-boot-tools-native', 'binutils-cross-aarch64', 'dtc-native', 'bison-native', 'pseudo-native', 'xz-native', 'python3-dtc-native', 'quilt-native', 'kmod-native', 'bc-native', 'patch-native', 'python3-native', 'gcc-cross-aarch64', 'libgcc', 'python3-pyyaml-native', 'kern-tools-native', 'openssl-native', 'flex-native', 'gnu-config-native', 'texinfo-dummy-native', 'autoconf-native', 'zlib-native', 'gtk-doc-native', 'automake-native', 'pkgconfig-native', 'libtool-native', 'readline-native', 'libffi-native', 'gdbm-native', 'util-linux-native', 'ncurses-native', 'bzip2-native', 'libtirpc-native', 'libnsl2-native', 'sqlite3-native', 'gettext-minimal-native', 'libyaml-native', 'swig-native', 'attr-native', 'mpfr-native', 'libmpc-native', 'gmp-native', 'linux-libc-headers', 'glibc', 'python3-setuptools-native', 'python3-cython-native', 'm4-native', 'libpcre2-native', 'libcap-ng-native', 'libpcre-native', 'unzip-native']
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing shell function do_configure
MISC_ARG is  -hdf_type xsa -yamlconf /home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/device-tree.yaml
APP_ARG is  -app "device-tree"
Using xsct from: /home/user/Downloads/petalinux_2021_1/tools/xsct//bin/xsct
cmd is: xsct -sdx -nodisp /home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/dtgen.tcl -ws /home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/project-spec/configs/../../components/plnx_workspace/device-tree -pname device-tree -rp /home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/git -processor_ip psu_cortexa53 -hdf /home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/build/tmp/deploy/images/zynqmp-generic/Xilinx-zynqmp-generic.xsa -arch 64   -app "device-tree"  -hdf_type xsa -yamlconf /home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/device-tree.yaml
INFO: [Hsi 55-2053] elapsed time for repository (/home/user/Downloads/petalinux_2021_1/tools/xsct/data/embeddedsw) loading 0 seconds
[... cut ...]
/home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/temp/run.do_configure.861: 200: [: y: unexpected operator
sed: can't read /home/user/work/adrv9009zu11EG_FMCOMMS8/adrv9009zu11eg_fmcomms8_linux/build/tmp/work/zynqmp_generic-xilinx-linux/device-tree/xilinx-v2021.1+gitAUTOINC+252758eb1f-r0/zynqmp-adrv9009-zu11eg-reva.dtsi: No such file or directory
WARNING: exit code 2 from a shell command.
                                                                                           
```

noelpedro, 2022-03-23:
are you using petalinux 2021.1 or 2021.1 updated version? 

When I source petalinux I get 

```
WARNING: /bin/sh is not bash! 
bash is PetaLinux recommended shell. Please set your default shell to bash.

```

noelpedro, 2022-03-24:
Looks like its missing some files from linux/arch/arm64/boot/dts/xilinx

namely the adrv9009 ones

noelpedro, 2022-03-24:
Manually copying them now. Hope it dosent lead to other errors when I finish.

noelpedro, 2022-03-24:
Okay copying those files over lead to successful build.

noelpedro, 2022-03-24:
Starting to boot now. Stuck in this loop 

```

evice 0: unknown device
starting USB...
No working controllers found
ethernet@ff0b0000 Waiting for PHY auto negotiation to complete......................................... TIMEOUT !
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5
BOOTP broadcast 6
BOOTP broadcast 7
BOOTP broadcast 8
BOOTP broadcast 9
BOOTP broadcast 10
BOOTP broadcast 11
BOOTP broadcast 12
BOOTP broadcast 13
BOOTP broadcast 14
BOOTP broadcast 15
BOOTP broadcast 16
BOOTP broadcast 17

Retry time exceeded; starting again
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-zynqmp-zynqmp
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-zynqmp
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default
*** ERROR: `serverip' not set
Config file not found
starting USB...
No working controllers found
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5
BOOTP broadcast 6
BOOTP broadcast 7
BOOTP broadcast 8
```

noelpedro, 2022-03-24:
oops forgot boot.scr lets see now

noelpedro, 2022-03-24:
```
iio:device2      ptyc3  ptyr2  ptyw1  random  ttya2   ttyp1  ttyu0  ttyyf
iio:device3      ptyc4  ptyr3  ptyw2  rfkill  ttya3   ttyp2  ttyu1  ttyz0
iio:device4      ptyc5  ptyr4  ptyw3  shm     ttya4   ttyp3  ttyu2  ttyz1
iio:device5      ptyc6  ptyr5  ptyw4  snd     ttya5   ttyp4  ttyu3  ttyz2
iio:device6      ptyc7  ptyr6  ptyw5  stderr  ttya6   ttyp5  ttyu4  ttyz3
iio:device7      ptyc8  ptyr7  ptyw6  stdin   ttya7   ttyp6  ttyu5  ttyz4
iio:device8      ptyc9  ptyr8  ptyw7  stdout  ttya8   ttyp7  ttyu6  ttyz5
iio:device9      ptyca  ptyr9  ptyw8  tty     ttya9   ttyp8  ttyu7  ttyz6
initctl          ptycb  ptyra  ptyw9  tty0    ttyaa   ttyp9  ttyu8  ttyz7
input            ptycc  ptyrb  ptywa  tty1    ttyab   ttypa  ttyu9  ttyz8
ion              ptycd  ptyrc  ptywb  tty10   ttyac   ttypb  ttyua  ttyz9
kmsg             ptyce  ptyrd  ptywc  tty11   ttyad   ttypc  ttyub  ttyza
log              ptycf  ptyre  ptywd  tty12   ttyae   ttypd  ttyuc  ttyzb
loop-control     ptyd0  ptyrf  ptywe  tty13   ttyaf   ttype  ttyud  ttyzc
loop0            ptyd1  ptys0  ptywf  tty14   ttyb0   ttypf  ttyue  ttyzd
loop1            ptyd2  ptys1  ptyx0  tty15   ttyb1   ttyq0  ttyuf  ttyze
loop2            ptyd3  ptys2  ptyx1  tty16   ttyb2   ttyq1  ttyv0  ttyzf
loop3            ptyd4  ptys3  ptyx2  tty17   ttyb3   ttyq2  ttyv1  uio0
loop4            ptyd5  ptys4  ptyx3  tty18   ttyb4   ttyq3  ttyv2  uio1
loop5            ptyd6  ptys5  ptyx4  tty19   ttyb5   ttyq4  ttyv3  uio2
loop6            ptyd7  ptys6  ptyx5  tty2    ttyb6   ttyq5  ttyv4  uio3
loop7            ptyd8  ptys7  ptyx6  tty20   ttyb7   ttyq6  ttyv5  urandom
mem              ptyd9  ptys8  ptyx7  tty21   ttyb8   ttyq7  ttyv6  vcs
mmcblk0          ptyda  ptys9  ptyx8  tty22   ttyb9   ttyq8  ttyv7  vcs1
mmcblk0p1        ptydb  ptysa  ptyx9  tty23   ttyba   ttyq9  ttyv8  vcs2
mmcblk0p2        ptydc  ptysb  ptyxa  tty24   ttybb   ttyqa  ttyv9  vcs3
mqueue           ptydd  ptysc  ptyxb  tty25   ttybc   ttyqb  ttyva  vcs4
mtd0             ptyde  ptysd  ptyxc  tty26   ttybd   ttyqc  ttyvb  vcs5
mtd0ro           ptydf  ptyse  ptyxd  tty27   ttybe   ttyqd  ttyvc  vcs6
mtd1             ptye0  ptysf  ptyxe  tty28   ttybf   ttyqe  ttyvd  vcsa
mtd1ro           ptye1  ptyt0  ptyxf  tty29   ttyc0   ttyqf  ttyve  vcsa1
mtd2             ptye2  ptyt1  ptyy0  tty3    ttyc1   ttyr0  ttyvf  vcsa2
mtd2ro           ptye3  ptyt2  ptyy1  tty30   ttyc2   ttyr1  ttyw0  vcsa3
mtd3             ptye4  ptyt3  ptyy2  tty31   ttyc3   ttyr2  ttyw1  vcsa4
mtd3ro           ptye5  ptyt4  ptyy3  tty32   ttyc4   ttyr3  ttyw2  vcsa5
mtdblock0        ptye6  ptyt5  ptyy4  tty33   ttyc5   ttyr4  ttyw3  vcsa6
mtdblock1        ptye7  ptyt6  ptyy5  tty34   ttyc6   ttyr5  ttyw4  vcsu
mtdblock2        ptye8  ptyt7  ptyy6  tty35   ttyc7   ttyr6  ttyw5  vcsu1
mtdblock3        ptye9  ptyt8  ptyy7  tty36   ttyc8   ttyr7  ttyw6  vcsu2
net              ptyea  ptyt9  ptyy8  tty37   ttyc9   ttyr8  ttyw7  vcsu3
null             ptyeb  ptyta  ptyy9  tty38   ttyca   ttyr9  ttyw8  vcsu4
port             ptyec  ptytb  ptyya  tty39   ttycb   ttyra  ttyw9  vcsu5
ptmx             ptyed  ptytc  ptyyb  tty4    ttycc   ttyrb  ttywa  vcsu6
pts              ptyee  ptytd  ptyyc  tty40   ttycd   ttyrc  ttywb  vhci
ptya0            ptyef  ptyte  ptyyd  tty41   ttyce   ttyrd  ttywc  watchdog
ptya1            ptyp0  ptytf  ptyye  tty42   ttycf   ttyre  ttywd  watchdog0
ptya2            ptyp1  ptyu0  ptyyf  tty43   ttyd0   ttyrf  ttywe  zero
ptya3            ptyp2  ptyu1  ptyz0  tty44   ttyd1   ttys0  ttywf
ptya4            ptyp3  ptyu2  ptyz1  tty45   ttyd2   ttys1  ttyx0
ubuntu@localhost:~$ cd /sys/bus/iio/
devices/           drivers_autoprobe  uevent             
drivers/           drivers_probe      
ubuntu@localhost:~$ cd /sys/bus/iio/
devices/           drivers_autoprobe  uevent             
drivers/           drivers_probe      
ubuntu@localhost:~$ cd /sys/bus/iio/devices/
ubuntu@localhost:/sys/bus/iio/devices$ ls
iio:device0   iio:device11  iio:device4  iio:device7  iio_sysfs_trigger
iio:device1   iio:device2   iio:device5  iio:device8
iio:device10  iio:device3   iio:device6  iio:device9
ubuntu@localhost:/sys/bus/iio/devices$ cat iio:device3/name
hmc7044-fmc
ubuntu@localhost:/sys/bus/iio/devices$ cat iio:device4/name
hmc7044
ubuntu@localhost:/sys/bus/iio/devices$ cat iio:device5/name
adrv9009-phy-c
ubuntu@localhost:/sys/bus/iio/devices$ cat iio:device6/name
adrv9009-phy-d
ubuntu@localhost:/sys/bus/iio/devices$ cat iio:device7/name
adrv9009-phy
ubuntu@localhost:/sys/bus/iio/devices$ cat iio:device8/name
adrv9009-phy-b
```

noelpedro, 2022-03-24:
Payday! devices are showing thanks man!  Can you help me fix issue with the dts file? OI had to copy them again before building.  Other then that MERGE that branch ! haha

nunojsa, 2022-03-24:
> Can you help me fix issue with the dts file? 

That should not be needed... I would suggest for you to do a clean build. Maybe between all of the tries (and having do_configure failing by using dash), you left your build in some weird state.

As I said, things work for me out of the box (building)... I just needed the assurance that things were ok and devices were showing up before merging...

I would also suggest for you to get a bit familiarized with docker and use [Dockerfile.petalinux](https://github.com/analogdevicesinc/meta-adi/blob/master/Dockerfile.petalinux). In theory, this will give you an environment ready to build petalinux and hence, the bash issue you had, would not happen in the first place.

noelpedro, 2022-03-24:
Same issue with Docker container

noelpedro, 2022-03-25:
Docker build worked

closed 2022-03-25 by noelpedro