analogdevicesinc/meta-adi #110: is there a way to use the META-ADI and iio for the fmcdaq2 on k26c custom board?

gudokim, 2022-03-09:
**Environment (please complete the following information):**
 - Project to build: custom k26c som
 - Petalinux version: 2021.1
 - HDL branch: master
 - META-ADI branch: master
 - Using meta-adi petalinux docker image? no

**Describe the bug**
Based on META-ADI and HDL, the fmcdaq2 and an iio applicaton were tested on zcu102 DK board.

After that, I made a custom board that supports the fmc interface and the kria k26c som to test the fmcdaq2.
Now, I wonder if there is a way to use the META-ADI and iio for the fmcdaq2 on k26c custom board.

the analogdevicesinc/linux master is based on xilinx 2021.1 and it seems not support the kria k26, yet.
So, Can I make a petalinux project like the below?
Can I still use meta-adi environment of analog device while making a project based on k26c bsp? 

1. create a project with k26 bsp
2. config a hardware desc and add the meta-adi layer0/1
3. add custom devicetree
4. build
```
petalinux-create -t project -s xilinx-k26-som-v2021.1-updated-final.bsp

petalinux-config --get-hw-description=k26c_som_custom.sdk
# add meta-adi layer0
~/meta-adi/meta-adi-core
# add meta-adi layer1
~/meta-adi/meta-adi-xilinx

# modify and add custom devicetree 
...

petalinux-build
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

nunojsa, 2022-03-09:
Hi,

> petalinux-create -t project -s xilinx-k26-som-v2021.1-updated-final.bsp

So, meta-adi was never tested with xilinx bsps... The flow for meta-adi is to use the hdl .xsa output file and import it in petalinux with `petalinux-config --get-hw-description=$PATH_TO_XSA_FILE`

> Can I still use meta-adi environment of analog device while making a project based on k26c bsp?

You can give it a try, but as I said it was never tested with bsp files...

>  kria k26c som to test the fmcdaq2

You also need to make sure your custom devicetree for this board works. Then you can try to follow the [adding-a-new-devicetree](https://github.com/analogdevicesinc/meta-adi/tree/master/meta-adi-xilinx#adding-a-new-devicetree) to see where it leads you...

On top of this, I don't think we ever tested our kernel on a kria platform... So that is another unknown outcome.

There are lots of "if's" here, so it's very unlikely things just work :). That means, you'll need to do some debugging on your own...

- Nuno Sá

gudokim, 2022-03-10:
Thank you.
It was helpful to think what to do.

gudokim, 2022-03-11:
Hi, Nuno Sá

Can you see our crazy(?) idea that it is possible or not?

We already tested the fmcdaq2 HDL on zcu102 with our own iio application and would like to use current HDL and meta-adi environment for testing fmcdaq2 and kria k26c som.

It seems that the xilinx kernel supports the kria k26c som since v5.17. so we will try to migration the fmcdaq2 and iio related drivers to the xilinx kernel v5.17 and modify the [linux-xlnx_%.bbappend](https://github.com/analogdevicesinc/meta-adi/blob/master/meta-adi-xilinx/recipes-kernel/linux/linux-xlnx_%25.bbappend) for linking to the modified kernel.
 - https://github.com/analogdevicesinc/linux#description : adi linux kernel v5.10
 - https://github.com/Xilinx/linux-xlnx/tree/zynqmp-soc-for-v5.17/arch/arm64/boot/dts/xilinx : xilinx linux kernel v5.17

Gudo

nunojsa, 2022-04-07:
closing this now

closed 2022-04-07 by nunojsa