analogdevicesinc/meta-adi #117: Xilinx tool version

yhanalog, 2022-10-25:
Quick question. 
Since we are using latest FPGA from Xilinx, we were recommended to use the 2022.x tool. 
However, it is stated that the 2021.2 is supported. 
So is it possible to use the 2022.x tool, and if not then do you have a time frame to move to the 2022.x version?
-YH

nunojsa, 2022-10-25:
Hi,

> So is it possible to use the 2022.x tool....

Most likely no...

> and if not then do you have a time frame to move to the 2022.x version?

Not really, we do have plans to update our tools soon but I cannot really say how soon it will be... Sorry!

- Nuno Sá

yhanalog, 2022-10-25:
Thanks a lot!
-YH

closed 2022-10-25 by nunojsa

tfcollins, 2022-10-28:
What Xilinx FPGA are you using?

yhanalog, 2022-10-31:
([https://www.xilinx.com/products/som/kria/k26i-industrial.html](https://urldefense.com/v3/__https:/www.xilinx.com/products/som/kria/k26i-industrial.html__;!!A3Ni8CS0y2Y!5AtUSq2-yhcg3JF6EQgRAqlakmOukAkm6h2qj7vxD5c4z4FKGJAaU7s6NbdRoeorhd-pJW5OTjyYXrwtDnmcT5DQdA$)