analogdevicesinc/meta-adi #123: [Adding a new Devicetree]Getting error while building from file pl-delete-node-

XuChunling, 2023-05-05:
**Environment (please complete the following information):**
 - Project to build: openwifi
 - Petalinux version: petalinux 2019.1
 - HDL branch:
 - META-ADI branch: 2019_R2
 - Using meta-adi petalinux docker image? no

**Describe the bug**
I want to "Adding a new Devicetree" as the readme of Meta-adi branch 2019_R2. For I want to build openwifi on a board which is not Xlinx but Alinx with xczu9e
 
[https://github.com/analogdevicesinc/meta-adi/blob/2019_R2/meta-adi-xilinx/README.md#adding-a-new-devicetree](url)

While building the project error comes from the file pl-delete-nodes-sariwifi-alinx-nopl.dtsi.

**Screenshots**
If applicable, add screenshots to help explain your problem.
/home/ring/图片/2023-05-05 17-32-42屏幕截图.png

errror log:
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++===
[INFO] building project
[INFO] sourcing bitbake
[INFO] generating user layers
INFO: bitbake petalinux-user-image
Loading cache: 100% |###################################################################################################################| Time: 0:00:00
Loaded 2354 entries from dependency cache.
ERROR: /home/ring/app/pz/petalinux-v2019.1/components/yocto/source/aarch64/layers/meta-xilinx/meta-xilinx-bsp/recipes-bsp/device-tree/device-tree.bb: Error executing a python function in <code>:

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 3, function: <module>
     0001:__anon_22__home_ring_app_pz_petalinux_v2019_1_components_yocto_source_aarch64_layers_core_meta_conf_machine_include_arm_feature_arm_thumb_inc(d)
     0002:__anon_25__home_ring_app_pz_petalinux_v2019_1_components_yocto_source_aarch64_layers_core_meta_classes_patch_bbclass(d)
 *** 0003:__anon_688__home_ring_app_pz_petalinux_v2019_1_components_yocto_source_aarch64_layers_core_meta_classes_base_bbclass(d)
     0004:__anon_20__home_ring_app_pz_petalinux_v2019_1_components_yocto_source_aarch64_layers_core_meta_classes_blacklist_bbclass(d)
     0005:__anon_1186__home_ring_app_pz_petalinux_v2019_1_components_yocto_source_aarch64_layers_core_meta_classes_insane_bbclass(d)
     0006:__anon_251__home_ring_app_pz_petalinux_v2019_1_components_yocto_source_aarch64_layers_core_meta_classes_package_bbclass(d)
     0007:__anon_718__home_ring_app_pz_petalinux_v2019_1_components_yocto_source_aarch64_layers_core_meta_classes_package_rpm_bbclass(d)
File: '/home/ring/app/pz/petalinux-v2019.1/components/yocto/source/aarch64/layers/core/meta/classes/base.bbclass', lineno: 585, function: __anon_688__home_ring_app_pz_petalinux_v2019_1_components_yocto_source_aarch64_layers_core_meta_classes_base_bbclass
     0581:    needsrcrev = False
     0582:    srcuri = d.getVar('SRC_URI')
     0583:    for uri in srcuri.split():
     0584:        (scheme, _ , path) = bb.fetch.decodeurl(uri)[:3]
 *** 0585:
     0586:        # HTTP/FTP use the wget fetcher
     0587:        if scheme in ("http", "https", "ftp"):
     0588:            d.appendVarFlag('do_fetch', 'depends', ' wget-native:do_populate_sysroot')
     0589:
File: '/home/ring/app/pz/petalinux-v2019.1/components/yocto/source/aarch64/layers/core/bitbake/lib/bb/fetch2/__init__.py', lineno: 368, function: decodeurl
     0364:    """
     0365:
     0366:    m = re.compile('(?P<type>[^:]*)://((?P<user>[^/;]+)@)?(?P<location>[^;]+)(;(?P<parm>.*))?').match(url)
     0367:    if not m:
 *** 0368:        raise MalformedUrl(url)
     0369:
     0370:    type = m.group('type')
     0371:    location = m.group('location')
     0372:    if not location:
Exception: bb.fetch2.MalformedUrl: The URL: 'pl-delete-nodes-sariwifi-alinx-nopl.dtsi' is invalid and cannot be interpreted

ERROR: Failed to parse recipe: /home/ring/app/pz/petalinux-v2019.1/components/yocto/source/aarch64/layers/meta-xilinx/meta-xilinx-bsp/recipes-bsp/device-tree/device-tree.bb
WARNING: /home/ring/app/kernel/AnalogDevice/meta-adi-2019_R2/meta-adi-core/recipes-core/jesd-status/jesd-status_dev.bb: Exception during build_dependencies for AUTOREV
WARNING: /home/ring/app/kernel/AnalogDevice/meta-adi-2019_R2/meta-adi-core/recipes-core/jesd-status/jesd-status_dev.bb: Error during finalise of /home/ring/app/kernel/AnalogDevice/meta-adi-2019_R2/meta-adi-core/recipes-core/jesd-status/jesd-status_dev.bb
WARNING: /home/ring/app/pz/petalinux-v2019.1/components/yocto/source/aarch64/layers/meta-xilinx/meta-xilinx-bsp/recipes-kernel/linux/linux-xlnx_2019.1.bb: Exception during build_dependencies for CXXFLAGS
WARNING: /home/ring/app/pz/petalinux-v2019.1/components/yocto/source/aarch64/layers/meta-xilinx/meta-xilinx-bsp/recipes-kernel/linux/linux-xlnx_2019.1.bb: Error during finalise of /home/ring/app/pz/petalinux-v2019.1/components/yocto/source/aarch64/layers/meta-xilinx/meta-xilinx-bsp/recipes-kernel/linux/linux-xlnx_2019.1.bb

Summary: There were 4 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
ERROR: Failed to build project

werneazc, 2023-05-16:
Hello,

it looks like you habe a typo in your `device-tree.bbappend` file. Have you written something like this in your `bbappend`-recipe:

```sh
FILEEXTRAPATHS_prepend := "${THIS_DIR}/${PL}:"

SRC_URI += " \
    file://pl-delete-nodes-sariwifi-alinx-nopl.dtsi \
"
```

Without the prefix `file://` bitbake does not know which fetcher it needs to use.

XuChunling, 2023-05-20:
> Hello,
> 
> it looks like you habe a typo in your `device-tree.bbappend` file. Have you written something like this in your `bbappend`-recipe:
> 
> ```shell
> FILEEXTRAPATHS_prepend := "${THIS_DIR}/${PL}:"
> 
> SRC_URI += " \
>     file://pl-delete-nodes-sariwifi-alinx-nopl.dtsi \
> "
> ```
> 
> Without the prefix `file://` bitbake does not know which fetcher it needs to use.

thks.
you are right.

now, I modify my device-tree.bbappend like that:
########################################################
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"

SRC_URI += "file://system-user.dtsi"

SRC_URI_append = " \
	file://pl-delete-nodes-sariwifi-alinx-ad9361.dtsi \
	file://sariwifi-alinx-ad9361.dts \
"

KERNEL_DTB_PATH = "${WORKDIR}"
##############################################

but ,there is another err:

***********************************************************************************************************************
ring@ring-MS-7B98:~/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build$ petalinux-build
[INFO] building project
[INFO] sourcing bitbake
[INFO] generating user layers
INFO: bitbake petalinux-user-image
Parsing recipes: 100% |#################################################################################################################| Time: 0:00:29
Parsing of 2781 .bb files complete (0 cached, 2781 parsed). 3816 targets, 150 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
Initialising tasks: 100% |##############################################################################################################| Time: 0:00:04
Checking sstate mirror object availability: 100% |######################################################################################| Time: 0:00:14
Sstate summary: Wanted 983 Found 804 Missed 358 Current 0 (81% match, 0% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: device-tree-xilinx+gitAUTOINC+73e546e312-r0 do_configure: Function failed: do_configure (log file is located at /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/temp/log.do_configure.5633)
ERROR: Logfile of failure stored in: /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/temp/log.do_configure.5633
Log data follows:
| DEBUG: Executing shell function do_configure
| MISC_ARG is  -hdf_type hdf -yamlconf /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/device-tree.yaml
| APP_ARG is  -app "device-tree"
| Using xsct from: /home/ring/app/pz/petalinux-v2019.1/tools/xsct/bin/xsct
| cmd is: xsct -sdx -nodisp /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/dtgen.tcl -ws /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/build -pname device-tree -rp /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/git -processor psu_cortexa53_0 -hdf /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/deploy/images/plnx-zynqmp/Xilinx-plnx-zynqmp.hdf -arch 64  -app "device-tree"  -hdf_type hdf -yamlconf /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/device-tree.yaml
| INFO: [Hsi 55-1698] elapsed time for repository loading 0 seconds
| hsi::open_hw_design: Time (s): cpu = 00:00:29 ; elapsed = 00:00:29 . Memory (MB): peak = 788.836 ; gain = 219.156 ; free physical = 291 ; free virtual = 13599
| WARNING: Interrupt pin "gps_pps_irq" of IP block: "axi_ad9361" is not connected to any interrupt controller
| 
| WARNING: not supported pl_clk:
| WARNING: Interrupt pin "s2mm_intr" of IP block: "openwifi_ip_rx_intf_0" is not connected to any interrupt controller
| 
| WARNING: not supported pl_clk:
| WARNING: not supported pl_clk:
| WARNING: Frequency 33.330 used instead of 33.333
| WARNING: label 'usb0' found in existing tree
| WARNING: Only boolean type can have empty value. Fail to add driver(psu_ctrl_ipi) property(clock-names) type(stringlist) value()
| WARNING: Please add the property manually
| WARNING: Only boolean type can have empty value. Fail to add driver(psu_message_buffers) property(clock-names) type(stringlist) value()
| WARNING: Please add the property manually
| hsi::generate_target: Time (s): cpu = 00:00:21 ; elapsed = 00:00:22 . Memory (MB): peak = 788.836 ; gain = 0.000 ; free physical = 270 ; free virtual = 13289
| sed: can't read /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/build/device-tree/system-conf.dtsi: No such file or directory
| WARNING: /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/temp/run.do_configure.5633:1 exit 2 from 'sed -i '/&qspi/,/^$/d; /&axi_ethernet/,/^$/d' "/home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/build/device-tre [... cut ...]
NOTE: Tasks Summary: Attempted 3310 tasks of which 2473 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/ring/app/pz/petalinux-v2019.1/components/yocto/source/aarch64/layers/meta-xilinx/meta-xilinx-bsp/recipes-bsp/device-tree/device-tree.bb:do_configure
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
ERROR: Failed to build project
*******************************************************************************************************************************************

nunojsa, 2023-05-20:
Hi, 

can you paste the actual error in the configure task? From the logs you should be able to find the location  of the log file and that should tell us what’s the error…

werneazc, 2023-05-21:
> Hi,
> 
> can you paste the actual error in the configure task? From the logs you should be able to find the location of the log file and that should tell us what’s the error…

I think that's the issue:

```
| sed: can't read /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/build/device-tree/system-conf.dtsi: No such file or directory
| WARNING: /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/temp/run.do_configure.5633:1 exit 2 from 'sed -i '/&qspi/,/^$/d; /&axi_ethernet/,/^$/d' "/home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/build/device-tree/system-conf.dtsi"'
```

It seems there is a file missing that needs to be available by the tool flow. This file is autogenerated and located at components/plnx_workspace in your PetaLinux project. Please delete the line `KERNEL_DTB_PATH = "${WORKDIR}"
` from your `bbappend`. It is not necessary in my opinion.

XuChunling, 2023-05-22:
> Hi,
> 
> can you paste the actual error in the configure task? From the logs you should be able to find the location of the log file and that should tell us what’s the error…

err log file "build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/temp/log.do_configure.5633" :

################################################################################
DEBUG: Executing shell function do_configure
MISC_ARG is  -hdf_type hdf -yamlconf /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/device-tree.yaml
APP_ARG is  -app "device-tree"
Using xsct from: /home/ring/app/pz/petalinux-v2019.1/tools/xsct/bin/xsct
cmd is: xsct -sdx -nodisp /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/dtgen.tcl -ws /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/build -pname device-tree -rp /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/git -processor psu_cortexa53_0 -hdf /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/deploy/images/plnx-zynqmp/Xilinx-plnx-zynqmp.hdf -arch 64  -app "device-tree"  -hdf_type hdf -yamlconf /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/device-tree.yaml
INFO: [Hsi 55-1698] elapsed time for repository loading 0 seconds
hsi::open_hw_design: Time (s): cpu = 00:00:29 ; elapsed = 00:00:29 . Memory (MB): peak = 788.836 ; gain = 219.156 ; free physical = 291 ; free virtual = 13599
WARNING: Interrupt pin "gps_pps_irq" of IP block: "axi_ad9361" is not connected to any interrupt controller

WARNING: not supported pl_clk:
WARNING: Interrupt pin "s2mm_intr" of IP block: "openwifi_ip_rx_intf_0" is not connected to any interrupt controller

WARNING: not supported pl_clk:
WARNING: not supported pl_clk:
WARNING: Frequency 33.330 used instead of 33.333
WARNING: label 'usb0' found in existing tree
WARNING: Only boolean type can have empty value. Fail to add driver(psu_ctrl_ipi) property(clock-names) type(stringlist) value()
WARNING: Please add the property manually
WARNING: Only boolean type can have empty value. Fail to add driver(psu_message_buffers) property(clock-names) type(stringlist) value()
WARNING: Please add the property manually
hsi::generate_target: Time (s): cpu = 00:00:21 ; elapsed = 00:00:22 . Memory (MB): peak = 788.836 ; gain = 0.000 ; free physical = 270 ; free virtual = 13289
sed: can't read /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/build/device-tree/system-conf.dtsi: No such file or directory
WARNING: /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/temp/run.do_configure.5633:1 exit 2 from 'sed -i '/&qspi/,/^$/d; /&axi_ethernet/,/^$/d' "/home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/build/device-tree/system-conf.dtsi"'
ERROR: Function failed: do_configure (log file is located at /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/temp/log.do_configure.5633)

werneazc, 2023-05-22:
Hi, as I said, it is a problem with your project configuration and a the missing file `system-conf.dtsi`. 

```
sed: can't read /home/ring/app/pz/project/alinx/sariwifi_alinx_9361_peta19.1_01/build/tmp/work/plnx_zynqmp-xilinx-linux/device-tree/xilinx+gitAUTOINC+73e546e312-r0/build/device-tree/system-conf.dtsi: No such file or directory
```

It seems that your workspace is broken. Maybe you need to run device-tree recipe again without adding meta-adi layer to your project. The missing file is generated by petalinux tool flow automatically reading the hardware description file you had provided using `petalinux-config —get-he-description` and executing `petalinux-build -c device-tree`.

Have you accidentally deleted the plnx_workspace folder or the missing file in that folder?

XuChunling, 2023-05-23:
> Hi, as I said, it is a problem with your project configuration and a the missing file `system-conf.dtsi`.
>
> ```
> sed: can't read [...]

"Maybe you need to run device-tree recipe again without adding meta-adi layer to your project." 
-----> But my project need to use RF ad9361 , so, I build follow the readme link, "Adding a new Devicetree" :
https://github.com/analogdevicesinc/meta-adi/blob/2019_R2/meta-adi-xilinx/README.md

werneazc, 2023-05-24:
> "Maybe you need to run device-tree recipe again without adding meta-adi layer to your project." -----> But my project need to use RF ad9361 , so, I build follow the readme link, "Adding a new [...]

Hi, I understood your statement. The thing is, the file is missing and it is generated from your hardware description file, exported from Vivado. When I integrated the ADI kernel into my project I did the following:

1. Create my block design in Vivado and export the hardware description including the bit stream.
2. Create a new PetaLinux project with `petalinux-create -t project --template zynqMP -n <projectname>`
3. Import hardware description file to PetaLinux project: `petalinux-config --get-hw-description <path to the exported file from vivado>`
4.  Add my device-tree in a separate file (`custom-board.dtsi`) and include it in `system-user.dtsi` (Do not forget to add the file to the to meta-user/recipes-bsp/device-tree/files, and the corresponding bbappend)

```
/include/ "system-conf.dtsi"
#include "custom-board.dtsi"
/{
};
```

5.  Build the project's device tree without the meta-adi layers: `petalinux-build -c device-tree`
6. Copy `<PetaLinuxRoot>/components/plnx_workspace/device-tree/system-top.dts` to `project-spec/meta-user/recipes-bsp/device-tree/files` and rename it, e.g. `custom-board-rev0.dts`
7. Create (empty) corresponding pl-delete file, e.g. `touch <PetalinuxRoot>/project-spec/meta-user/recipes-bsp/device-tree/files/pl-delete-nodes-custom-board-rev0.dtsi`
8. Add layers from meta-adi as submodule using git submodule feature: `cd project-spec && git submodule add https://github.com/analogdevicesinc/meta-adi.git`
9. Use petalinux-config to add layers to project: `petalinux-config`-> Yocto Settings->User Layers

```
CONFIG_USER_LAYER_0="${PROOT}/project-spec/meta-adi/meta-adi-core"
CONFIG_USER_LAYER_1="${PROOT}/project-spec/meta-adi/meta-adi-xilinx"
```

10. Run `petalinux-build` to build with added ADI kernel.

It seems that something was broken in your project, however, I cannot except that something additional is wrong with your recipes located in meta-user.

nunojsa, 2023-06-02:
Well, you could workaround the issue by adding this in the do_configure function:

```
[  -e "${DT_FILES_PATH}/system-conf.dtsi" ] && \
		{ sed -i '/&qspi/,/^$/d; /&axi_ethernet/,/^$/d' "${DT_FILES_PATH}/system-conf.dtsi" }
```

However, as @werneazc mentioned, that file should be automatically created by petalinux which means doing the above might get you going during compilation but you can find other issues (eg: not having a proper kernel command line).

I did found this issues once when moving to a newer petalinux version:

https://github.com/search?q=repo%3Aanalogdevicesinc%2Fmeta-adi+75207db10b95561005c7cce25012093f152d1237&type=commits

But this should not be your issue...

werneazc, 2023-06-09:
Today, I ran into the error of missing a `system-conf.dtsi` file if the layer priority of the meta-user is higher than meta-adi-xilinx which is 8. Have you changed this property for the meta-user layer to something higher? Then you need to decrease this number at least once and execute 'petalinux-build -c device-tree'.

nunojsa, 2023-06-12:
> Today, I ran into the error of missing a system-conf.dtsi file if the layer priority of the meta-user layer is higher than meta-adi-xilinx which is 8. Have you changed this property for the meta-user layer to something higher?

Check the commit above... I changed the priority in meta-adi-xilinx so I don't have to do it all the time... But that should not happen for the supported petalinux versions.

nunojsa, 2023-07-27:
Looks like things settled in this one. Closing...

closed 2023-07-27 by nunojsa