analogdevicesinc/meta-adi #126: Corner case handling in device-tree.bbappend broken

balister, 2023-06-05:
Background is here:

https://ez.analog.com/linux-software-drivers/f/q-a/570110/unable-to-build-petalinux-2021-2-for-zynq-adrv9361-z7035-bob-cmos/495320

Adjusting the bbappend like this:

dtb_tag_file="${KERNEL_DTB_PATH}/zynq-adrv9364-z7020-bob.dts"

resolved the issue, but the comments in the code suggest this is wrong. The logic that copies the device trees from the kernel into WORKDIR seems to now copy them. The bbappend works for the non corner cases because teh device tree is copied into "${DT_FILES_PATH}/system-top.dts"

Any suggestions for the correct fix to this issue?

nunojsa, 2023-06-06:
> Any suggestions for the correct fix to this issue?

Not really sure... As you stated:

> The logic that copies the device trees from the kernel into WORKDIR seems to now copy them. 

And this should be enough for the "corner cases" to work as the goal is that all devicetrees are copied into WORKDIR. This logic is already in 2021_R2 which is the branch that should be used for petalinux 2021.2 (the one mentioned in the thread).

So, I would say that what needs to be figured is why are the devicetrees not being copied into WORKDIR.... Anyways, I'll also reply in EZ.

Thanks for bringing this to github (easier for me to notice things in here :))

balister, 2023-06-06:
As you can see in EZ, we went through the process of checking what was in WORKDIR and basically no dts files there.

nunojsa, 2023-06-06:
Sure... Still wondering why the dts files are not there. For example:

https://github.com/analogdevicesinc/meta-adi/blob/master/meta-adi-xilinx/recipes-bsp/device-tree/device-tree.bbappend#LL96C1-L96C59

All zynqMP projects rely pretty much on the same logic and I also have an internal weekly CI that runs for all projects (for master and the latest release branch - 2021_R2) and the builds are working for both `adrv9361z7035-ccbob_cmos` and `adrv9364z7020-ccbob_cmos`...

nunojsa, 2023-06-06:
Closed as it seems to be fixed:

https://ez.analog.com/linux-software-drivers/f/q-a/570110/unable-to-build-petalinux-2021-2-for-zynq-adrv9361-z7035-bob-cmos/495320

closed 2023-06-06 by nunojsa