analogdevicesinc/meta-adi #127: ReadMe not match latest YOCTO

oharwot, 2023-06-23:
**Environment (please complete the following information):**
 - Project to build:
 - Petalinux version: 2022
 - HDL branch: 
 - META-ADI branch: master
 - Using meta-adi petalinux docker image? no

**Describe the bug**
The README not match latest YOCTO syntax. I propose apply this patch (cant commit directly to the repo)

 ase) ondrej@CZ-M-017-LINUX:~/BlockBox/ADgitLinux$ ls
AD9082_EK  hw_export  meta-adi
(base) ondrej@CZ-M-017-LINUX:~/BlockBox/ADgitLinux$ cd meta-adi/
(base) ondrej@CZ-M-017-LINUX:~/BlockBox/ADgitLinux/meta-adi$ git config user.name Ondrej Harwot
(base) ondrej@CZ-M-017-LINUX:~/BlockBox/ADgitLinux/meta-adi$ git config user.email ondrej.harwot@huld.io
(base) ondrej@CZ-M-017-LINUX:~/BlockBox/ADgitLinux/meta-adi$ git format-patch -1 HEAD
0001-Update-the-readme.patch
(base) ondrej@CZ-M-017-LINUX:~/BlockBox/ADgitLinux/meta-adi$ cat 0001-Update-the-readme.patch 
From 2c8a4b2aa7aaff24e81a83af417128980b999366 Mon Sep 17 00:00:00 2001
From: Ondrej Harwot <ondrej.harwot@huld.io>
Date: Fri, 23 Jun 2023 21:52:48 +0200
Subject: [PATCH] Update the readme

Update the readme to match YOCTO new syntax
---
 meta-adi-xilinx/README.md | 10 +++++-----
 1 file changed, 5 insertions(+), 5 deletions(-)

diff --git a/meta-adi-xilinx/README.md b/meta-adi-xilinx/README.md
index e5a3015..230ffda 100644
--- a/meta-adi-xilinx/README.md
+++ b/meta-adi-xilinx/README.md
@@ -240,14 +240,14 @@ The `pl-delete-nodes-${KERNEL_DTB}` should look:
 
 > NOTE: We typically just want to delete custom nodes which do not refer to xilinx IP cores. If your reference design uses some xilinx IPs, you should not delete those nodes since they should be just fine!
 
-Having this done, it is time to add both the devicetree and pl-delete-nodes-${KERNEL_DTB} to the build system. The recommended way of doing this is to further append the devicetree recipe in the following manner (assuming you devicetree is named "`custom-devicetree`"):
+Having this done, it is time to add both the devicetree and pl-delete-nodes-${KERNEL_DTB} to the build system. The recommended way of doing this is to further append the devicetree recipe (device-tree.bbappend) in the following manner (assuming you devicetree is named "`custom-devicetree`"):
 
 ``` bash
-FILESEXTRAPATHS_prepend := "${THISDIR}/files:"
+FILESEXTRAPATHS::prepend := "${THISDIR}/files:"
 
-SRC_URI_append = " \
-       pl-delete-nodes-custom-devicetree.dtsi \
-       custom-devicetree.dts \
+SRC_URI::append = " \
+       file://pl-delete-nodes-custom-devicetree.dtsi \
+       file://custom-devicetree.dts \
 "
 
 # If the project being added is supported by ADI (with no custom changes), the devicetree should be present in the kernel sources. Hence, there's no need to redefine KERNEL_DTB_PATH
-- 
2.25.1

nunojsa, 2023-06-26:
Hi,

Yes, that makes sense. Good catch! Please, feel free to open a Pull Request with this patch. Do not forget to Sign-off your patch and follow the `git log` style of the project :)

Thanks!

nunojsa, 2023-11-06:
fixed in main and 2022_R2