analogdevicesinc/meta-adi #128: libiio points to non-existant branch

mbush00, 2023-08-28:
As of 2023-08-22, https://github.com/analogdevicesinc/libiio no longer has a 'master' branch. It was moved into 'libiio-v0'. The default branch for that project is set to 'main'. 

In meta-adi-xilinx/recipes-support/libiio/libiio_%.bbappend the branch referenced is 'master'.

Not sure if ADI's plans are to move to the latest branch, however trying within myown project, moving to 'main' does not build. I had to use the 'libiio-v0' branch.

nunojsa, 2023-08-29:
Hi,

> Not sure if ADI's plans are to move to the latest branch, however trying within myown project, moving to 'main' does not build. I had to use the 'libiio-v0' branch.

Yes, the plan for meta-adi master is to just keep up with master/main on the supported packages... If it does not build, I'll have to look into that.

Thanks for reporting,
I'm just back from vacation so this one was missed...

nunojsa, 2023-09-04:
Hi,

So I just looked at this one. Apparently libiio v0.25 changed considerably (I thought it would only happen with libiio 1.0) and on top of changing the branch from master -> main, there is also API breakage. That all means that projects using libiio are now failing to compile. Hence:

> moving to 'main' does not build

It's not libiio failing to build but `libad9361-iio` and `adrv9009-zu11eg-fan-control`. So, if you want to use the great and latest on libiio (and don't need the above packages), you can do the following workaround until the above projects are updated with the new API:

```
diff --git a/meta-adi-xilinx/dynamic-layers/meta-petalinux/recipes-core/images/petalinux-image-minimal.bbappend b/meta-adi-xilinx/dynamic-layers/meta-petalinux/recipes-core/images/petalinux-image-minimal.bbappend
index a401f218a14b..b573d6bb2120 100644
--- a/meta-adi-xilinx/dynamic-layers/meta-petalinux/recipes-core/images/petalinux-image-minimal.bbappend
+++ b/meta-adi-xilinx/dynamic-layers/meta-petalinux/recipes-core/images/petalinux-image-minimal.bbappend
@@ -4,9 +4,7 @@ IMAGE_INSTALL:append = " libiio  \
                         libiio-python3 \
                         avahi-daemon \
                         fru-tools \
-                        libad9361-iio \
-                        jesd-status \
-                        adrv9009-zu11eg-fan-control"
+                        jesd-status"
```

I'll anyways have to update libiio recipe to point to main and update PV to 0.25 and the offline git hash

nunojsa, 2023-09-07:
Apparently moving those projects to libiio v1.0 will take longer than I expected. Hence, I'm pointing libiio to the 'libiio-v0' branch (as you did already) in #129.

As soon as possible, I'll point libiio back to 'main'

PR #129 (2023-09-08): meta-adi-xilinx: recipes-support: libiio: move to libbio-v0 branch
Libiio renamed the master branch to main. More importantly, it moved to libiio v1.0 which is not backward compatible (at least for now) with libiio v0. Hence, all the projects that 'DEPEND' on libiio will fail to compile in meta-adi. Since it might take some time for those projects to be converted, let's move to the v0 branch and update the package 'PV' accordingly.

Fixes #128
closed 2023-09-07 by nunojsa