analogdevicesinc/meta-adi #130: Mismatch in kernel version

Zosoworld, 2023-10-04:
**Environment (please complete the following information):**
 - Project to build: adrv9364-z7020
 - Petalinux version: 2022.2
 - HDL branch: master
 - META-ADI branch: master
 - Using meta-adi petalinux docker image? No

**Describe the bug**
After downloading petalinux 2022.2 and creating a new project for `zynq`, I clone the `meta-adi` repository following the instructions from [here](https://github.com/analogdevicesinc/meta-adi/tree/2021_R2/meta-adi-xilinx), and try to build the project. Even though in the `bbappend` files the kernel appears to be 5.15, I get the following error:
```
ERROR: linux-xlnx-5.15-adi_master+gitAUTOINC+machine-r0 do_kernel_version_sanity_check: Package Version (5.15-adi_master+gitAUTOINC+machine) does not match of kernel being built (6.1). Please update the PV variable to match the kernel source or set KERNEL_VERSION_SANITY_SKIP="1" in your recipe.
ERROR: linux-xlnx-5.15-adi_master+gitAUTOINC+machine-r0 do_kernel_version_sanity_check: ExecutionError('/home/user/petalinux/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/linux-xlnx/5.15-adi_master+gitAUTOINC+machine-r0/temp/run.do_kernel_version_sanity_check.935360', 1, None, None)
ERROR: Logfile of failure stored in: /home/user/petalinux/build/tmp/work/zynq_generic-xilinx-linux-gnueabi/linux-xlnx/5.15-adi_master+gitAUTOINC+machine-r0/temp/log.do_kernel_version_sanity_check.935360
ERROR: Task (/home/user/petalinux/components/yocto/layers/meta-xilinx/meta-xilinx-core/recipes-kernel/linux/linux-xlnx_2022.2.bb:do_kernel_version_sanity_check) failed with exit code '1'
```

amadeugonga, 2023-10-23:
I manage to solve the issue by adding `KERNEL_VERSION_SANITY_SKIP="1"` at the petalinuxbsp.conf file in <path-to-project>/project-spec/meta-user/conf.
However, after successfully packing my customized image, I am not able to program the QSPI in QSPI boot mode (not JTAG) as I was able in previous Petalinux versions (i.e. 2019.2). Setting it to JTAG is not possible since I do not have access to the switches. Is there any solution?

SushiTee, 2023-10-24:
In the meantime you can simply workaround it by using an older version of the kernel.

Therefore, in the menu after `petalinux-config --get-hw-description=<..>` set the following:

* Linux Components Selection  ---> linux-kernel (remote)  --->    
    (X) remote
* Linux Components Selection  ---> Remote linux-kernel settings  --->    
    Remote linux-kernel git URL: `git://github.com/analogdevicesinc/linux.git;protocol=https`
    Remote linux-kernel git TAG/Commit ID: `cd8c11add73d7096be8bffc6683862b1c883f24a`

This sets the kernel to the commit right before the 2023.1 merge.

PR #132 (2023-11-03): petalinux 2023.1
This adds support to petalinux 2023.1. The most notable changes are:

1) Removing meta-adi-core (refer to commit message for more details);
2) Removing fpga-manager example. Petalinux changed the way fpga-manager is to be used so the given example is not working anymore. I intend to check how feasible it is to add back an example with the existing mechanism but I don't want to block moving on 2023.1 because of it. Note that the devicetree overlays used in the previous example were not removed.

Should fix #130 and #127
closed 2023-11-03 by nunojsa