analogdevicesinc/meta-adi #136: Petalinux issue

DiogoBaptistaBosch, 2023-11-16:
Hi, 

I am using the petalinux (2022.2) flow to generate a linux image with support for the FMCOMMS5 for a zcu102. I have been using this flow for quite some time now, but the latest updates to libiio and meta-adi ruined my environment. I saw that a major update was done (remove meta-adi-core) to meta-adi, so I download it again a tried using it in my flow but the following problem arrived after running petalinux-config:

ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: honister (layer is compatible with langdale)
ERROR: Layer adi-xilinx is not compatible with the core layer which only supports these series: honister (layer is compatible with langdale)

I also tried to keep the flow I had, were I fixed the issue of the "master" not being found (https://github.com/analogdevicesinc/meta-adi/issues/128). But, although it worked for the past months, after the update  3 days ago, it stopped to work. This error appears after running petalunx-build:
Traceback (most recent call last):
  File "/media/team_storage/NextSense_P10/Diogo/image_petalinux/GPS_L1_peta/components/yocto/layers/core/bitbake/lib/bb/fetch2/git.py", line 245, in Git.urldata_init(ud=<bb.fetch2.FetchData object at 0x7efc4b4e2b80>, d=<bb.data_smart.DataSmart object at 0x7efc4b4a8820>):
     
    >        ud.setup_revisions(d)
     
  File "/media/team_storage/NextSense_P10/Diogo/image_petalinux/GPS_L1_peta/components/yocto/layers/core/bitbake/lib/bb/fetch2/__init__.py", line 1331, in FetchData.setup_revisions(d=<bb.data_smart.DataSmart object at 0x7efc4b4a8820>):
             for name in self.names:
    >            self.revisions[name] = srcrev_internal_helper(self, d, name)
     
  File "/media/team_storage/NextSense_P10/Diogo/image_petalinux/GPS_L1_peta/components/yocto/layers/core/bitbake/lib/bb/fetch2/__init__.py", line 1196, in srcrev_internal_helper(ud=<bb.fetch2.FetchData object at 0x7efc4b4e2b80>, d=<bb.data_smart.DataSmart object at 0x7efc4b4a8820>, name='default'):
         if srcrev == "AUTOINC":
    >        srcrev = ud.method.latest_revision(ud, d, name)
     
  File "/media/team_storage/NextSense_P10/Diogo/image_petalinux/GPS_L1_peta/components/yocto/layers/core/bitbake/lib/bb/fetch2/__init__.py", line 1605, in Git.latest_revision(ud=<bb.fetch2.FetchData object at 0x7efc4b4e2b80>, d=<bb.data_smart.DataSmart object at 0x7efc4b4a8820>, name='default'):
             except KeyError:
    >            revs[key] = rev = self._latest_revision(ud, d, name)
                 return rev
  File "/media/team_storage/NextSense_P10/Diogo/image_petalinux/GPS_L1_peta/components/yocto/layers/core/bitbake/lib/bb/fetch2/git.py", line 728, in Git._latest_revision(ud=<bb.fetch2.FetchData object at 0x7efc4b4e2b80>, d=<bb.data_smart.DataSmart object at 0x7efc4b4a8820>, name='default'):
                         return sha1
    >        raise bb.fetch2.FetchError("Unable to resolve '%s' in upstream git repository in git ls-remote output for %s" % \
                 (ud.unresolvedrev[name], ud.host+ud.path))
bb.data_smart.ExpansionError: Failure expanding variable SRCPV, expression was ${@bb.fetch2.get_srcrev(d)} which triggered exception FetchError: Fetcher failure: Unable to resolve 'master' in upstream git repository in git ls-remote output for github.com/analogdevicesinc/fru_tools.git
The variable dependency chain for the failure is: SRCPV -> PV -> BP -> FILESPATH

Thanks in advanced,
Diogo Baptista

DiogoBaptistaBosch, 2023-11-16:
Thank you for the reply. That branch is from meta-adi or libiio?

nunojsa, 2023-11-16:
meta-adi

DiogoBaptistaBosch, 2023-11-16:
Thank you for the reply. I cloned the meta-adi 2022_R2 and it worked! Do you have plans for doing more updates to that branch?

nunojsa, 2023-11-16:
> Thank you for the reply. I cloned the meta-adi 2022_R2 and it worked! Do you have plans for doing more updates to that branch?

Nope... In theory, now only bugfixes go there...

DiogoBaptistaBosch, 2023-11-16:
Ok! Thank you for the help!

closed 2023-11-16 by DiogoBaptistaBosch