analogdevicesinc/meta-adi #137: Device-Tree isuue

pollai2, 2023-11-18:
**Environment (please complete the following information):**
 - Project to build: zc706 with AD9371
 - Petalinux version:2021.2
 - HDL branch:https://github.com/analogdevicesinc/hdl/tree/hdl_2021_r2
 - META-ADI branch:https://github.com/analogdevicesinc/meta-adi/tree/2021_R2
 - Using meta-adi petalinux docker image?

**Describe the bug**
I am Creating BSP project in petalinux after successful build, i am download the met-adi layers from github and placed outside of the project,run this command 
"echo "KERNEL_DTB=\"${dts_to_use}\"" >> project-spec/meta-user/conf/petalinuxbsp.conf"  in project root after go to build directory 
cd buuild  run the petalinuux-build command after successfull build, run the petalinux-config -c kernel command in <project-root>  enable the AD9371 drivers save it and exit, then run the petalinux-build command in <proj-root> . i got an error regarding Device-Tree how to reslove this issue.

**Screenshots**
[image]

nunojsa, 2023-12-19:
Hi,

The error is telling you everything. Look at the name of your devicetree... It does not look good and the system cannot find a matching pl-delete-nodes- file.

pollai2, 2023-12-19:
Hi

Thanks for your reply. I had solved  this error and also  generate the
bootable images .

Regards
P.Prasanth

closed 2023-12-19 by nunojsa