analogdevicesinc/meta-adi #138: Petalinux 2022.2 adrv9025 build fails

oharwot, 2023-12-07:
Hello

I am facing an issue when building for Petalinux 2022_2. I get the error

|   AR      drivers/staging/iio/resolver/built-in.a
|   CC      drivers/iio/accel/adis16201.o
|   CC      drivers/iio/adc/madura/devices/adrv9025/private/src/adrv9025_bf_analog_orx_mem_map.o
|   AR      drivers/media/usb/uvc/built-in.a
|   AR      drivers/media/usb/built-in.a
|   AR      drivers/of/built-in.a
|   AR      drivers/staging/iio/meter/built-in.a
| In file included from /home/ondrej/BlockBox/HuldGit/firmware/Linux/AD9082_EK/build/tmp/work-shared/zynqmp-generic/kernel-source/drivers/iio/adc/madura/devices/adrv9025/private/src/adrv9025_bf_analog_orx_mem_map.c:17:
| /home/ondrej/BlockBox/HuldGit/firmware/Linux/AD9082_EK/build/tmp/work-shared/zynqmp-generic/kernel-source/drivers/iio/adc/madura/devices/adrv9025/private/src/./../../private/include/adrv9025_bf_analog_orx_mem_map.h:20:10: fatal error: adi_adrv9025.h: No such file or directory
|    20 | #include "adi_adrv9025.h"
|       |          ^~~~~~~~~~~~~~~~
| compilation terminated.
| make[4]: *** [/home/ondrej/BlockBox/HuldGit/firmware/Linux/AD9082_EK/build/tmp/work-shared/zynqmp-generic/kernel-source/scripts/Makefile.build:277: drivers/iio/adc/madura/devices/adrv9025/private/src/adrv9025_bf_analog_orx_mem_map.o] Error 1
| make[3]: *** [/home/ondrej/BlockBox/HuldGit/firmware/Linux/AD9082_EK/build/tmp/work-shared/zynqmp-generic/kernel-source/scripts/Makefile.build:540: drivers/iio/adc/madura] Error 2
| make[2]: *** [/home/ondrej/BlockBox/HuldGit/firmware/Linux/AD9082_EK/build/tmp/work-shared/zynqmp-generic/kernel-source/scripts/Makefile.build:540: drivers/iio/adc] Error 2
| make[2]: *** Waiting for unfinished jobs....
|   CC      drivers/media/platform/xilinx/xilinx-scaler.o
|   AR      drivers/staging/iio/built-in.a
|   CC      drivers/iio/addac/ad74115.o
|   CC      drivers/jesd204/jesd204-fsm.o
|   AR      drivers/staging/built-in.a
|   CC      drivers/jesd204/jesd204-sysfs.o

As a temporary fix I am using modified kernel config with

CONFIG_ADRV9025=n 

However this is not ideal solution.

Any idea why this is happening?

Thanks for support
Ondrej

Abraxas3d, 2023-12-10:
We are noticing the same error. 

2022.2 Petalinux, adrv9009+zc706 build is failing as of 3 days ago.

From a message in build.log, we knew to look at the file .../build/tmp/work/zynq_generic-xilinx-linux-gnueabi/linux-xlnx/5.15-adi_master+gitAUTOINC+machine-r0/temp/log.do_compile.14242

And we see the same missing file. 

/home/abraxas3d/haifuraiya/challenge_coin/petalinux/challenge_coin/build/tmp/work-shared/zynq-generic/kernel-source/drivers/iio/adc/madura/devices/adrv9025/private/src/./../../private/include/adrv9025_bf_analog_orx_mem_map.h:20:10: fatal error: adi_adrv9025.h: No such file or directory
   20 | #include "adi_adrv9025.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.

Also, $petalinux-config --get-hw-description fails to add user layer. Running it again right after the failure seemingly solved it, but we're stating to think it isn't working as intended. 

[INFO] Generating plnxtool conf
[INFO] Adding user layers
ERROR: Failed to add user layer: /home/abraxas3d/haifuraiya/challenge_coin/petalinux/challenge_coin/project-spec/meta-user
ERROR: Failed to config project.
ERROR: Get config component Failed!. Check the /home/abraxas3d/haifuraiya/challenge_coin/petalinux/challenge_coin/build/config.log file for more details...

nunojsa, 2023-12-11:
> Any idea why this is happening?

Yeah, we had some issues in the kernel when introducing that driver. We had some fixes for it last Friday... Can you do a fresh build and see if this is still an issue?

If it is, I'll have to look at it...

nunojsa, 2023-12-11:
alright, this should fix it:

https://github.com/analogdevicesinc/linux/pull/2379

nunojsa, 2023-12-11:
should be fixed with:

https://github.com/analogdevicesinc/linux/pull/2379

closed 2023-12-11 by nunojsa