analogdevicesinc/meta-adi #150: device-tree-xilinx-v2023.2

ducanh1699, 2024-05-13:
Hello everyone, I try to build zcu102 with adrv9025 but when running petalinux-build I have an error. I find many different ways but I still can not solve this bug.
Can you help me some idea or suggestion ? Below It's full of my build log. Tks for your support ^^

ERROR: device-tree-xilinx-v2023.2+gitAUTOINC+1a5881d004-r0 do_compile: Error executing a python function in exec_func_python() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_func_python() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:devicetree_do_compile(d)
     0003:
File: '/home/xltn/petalinux/2023.2/xilinx-zcu102-2023.2/components/yocto/layers/poky/meta/classes-recipe/devicetree.bbclass', lineno: 137, function: devicetree_do_compile
     0133:            if not(os.path.isfile(dtspath)) or not(dts.endswith(".dts") or devicetree_source_is_overlay(dtspath)):
     0134:                continue # skip non-.dts files and non-overlay files
     0135:        except:
     0136:            continue # skip if can't determine if overlay
 *** 0137:        devicetree_compile(dtspath, includes, d)
     0138:}
     0139:
     0140:devicetree_do_install() {
     0141:    for DTB_FILE in `ls *.dtb *.dtbo`; do
File: '/home/xltn/petalinux/2023.2/xilinx-zcu102-2023.2/components/yocto/layers/poky/meta/classes-recipe/devicetree.bbclass', lineno: 125, function: devicetree_compile
     0121:        dtcargs += ["-i", i]
     0122:    dtcargs += ["-o", "{0}.{1}".format(dtname, "dtbo" if isoverlay else "dtb")]
     0123:    dtcargs += ["-I", "dts", "-O", "dtb", "{0}.pp".format(dts)]
     0124:    bb.note("Running {0}".format(" ".join(dtcargs)))
 *** 0125:    subprocess.run(dtcargs, check = True, stdout=subprocess.PIPE, stderr=subprocess.STDOUT)
     0126:
     0127:python devicetree_do_compile() {
     0128:    includes = expand_includes("DT_INCLUDE", d)
     0129:    listpath = d.getVar("DT_FILES_PATH")
File: '/home/xltn/petalinux/2023.2/components/yocto/buildtools/sysroots/x86_64-petalinux-linux/usr/lib/python3.10/subprocess.py', lineno: 524, function: run
     0520:            # We don't call process.wait() as .__exit__ does that for us.
     0521:            raise
     0522:        retcode = process.poll()
     0523:        if check and retcode:
 *** 0524:            raise CalledProcessError(retcode, process.args,
     0525:                                     output=stdout, stderr=stderr)
     0526:    return CompletedProcess(process.args, retcode, stdout, stderr)
     0527:
     0528:
Exception: subprocess.CalledProcessError: Command '['dtc', '-@', '-p', '0x1000', '-@', '-i', '/home/xltn/petalinux/2023.2/xilinx-zcu102-2023.2/build/tmp/work/xilinx_zcu102-xilinx-linux/device-tree/xilinx-v2023.2+gitAUTOINC+1a5881d004-r0/git/device_tree/data/kernel_dtsi/2023.2/include/', '-i', '/home/xltn/petalinux/2023.2/xilinx-zcu102-2023.2/components/plnx_workspace/device-tree/device-tree', '-i', '/home/xltn/petalinux/2023.2/xilinx-zcu102-2023.2/build/tmp/work/xilinx_zcu102-xilinx-linux/device-tree/xilinx-v2023.2+gitAUTOINC+1a5881d004-r0', '-o', 'system-top.dtb', '-I', 'dts', '-O', 'dtb', 'system-top.dts.pp']' returned non-zero exit status 1.

Subprocess output:
Error: /home/xltn/petalinux/2023.2/xilinx-zcu102-2023.2/build/tmp/work/xilinx_zcu102-xilinx-linux/device-tree/xilinx-v2023.2+gitAUTOINC+1a5881d004-r0/pl-delete-nodes-zynqmp-zcu102-rev10-adrv9025.dtsi:1.15-35 Label or path axi_adrv9026_rx_dma not found
Error: /home/xltn/petalinux/2023.2/xilinx-zcu102-2023.2/build/tmp/work/xilinx_zcu102-xilinx-linux/device-tree/xilinx-v2023.2+gitAUTOINC+1a5881d004-r0/pl-delete-nodes-zynqmp-zcu102-rev10-adrv9025.dtsi:3.15-43 Label or path axi_adrv9026_rx_jesd_rx_axi not found
Error: /home/xltn/petalinux/2023.2/xilinx-zcu102-2023.2/build/tmp/work/xilinx_zcu102-xilinx-linux/device-tree/xilinx-v2023.2+gitAUTOINC+1a5881d004-r0/pl-delete-nodes-zynqmp-zcu102-rev10-adrv9025.dtsi:4.15-36 Label or path axi_adrv9026_rx_xcvr not found
Error: /home/xltn/petalinux/2023.2/xilinx-zcu102-2023.2/build/tmp/work/xilinx_zcu102-xilinx-linux/device-tree/xilinx-v2023.2+gitAUTOINC+1a5881d004-r0/pl-delete-nodes-zynqmp-zcu102-rev10-adrv9025.dtsi:5.15-35 Label or path axi_adrv9026_tx_dma not found
Error: /home/xltn/petalinux/2023.2/xilinx-zcu102-2023.2/build/tmp/work/xilinx_zcu102-xilinx-linux/device-tree/xilinx-v2023.2+gitAUTOINC+1a5881d004-r0/pl-delete-nodes-zynqmp-zcu102-rev10-adrv9025.dtsi:6.15-43 Label or path axi_adrv9026_tx_jesd_tx_axi not found
Error: /home/xltn/petalinux/2023.2/xilinx-zcu102-2023.2/build/tmp/work/xilinx_zcu102-xilinx-linux/device-tree/xilinx-v2023.2+gitAUTOINC+1a5881d004-r0/pl-delete-nodes-zynqmp-zcu102-rev10-adrv9025.dtsi:7.15-36 Label or path axi_adrv9026_tx_xcvr not found
Error: /home/xltn/petalinux/2023.2/xilinx-zcu102-2023.2/build/tmp/work/xilinx_zcu102-xilinx-linux/device-tree/xilinx-v2023.2+gitAUTOINC+1a5881d004-r0/pl-delete-nodes-zynqmp-zcu102-rev10-adrv9025.dtsi:8.15-27 Label or path axi_sysid_0 not found
Error: /home/xltn/petalinux/2023.2/xilinx-zcu102-2023.2/build/tmp/work/xilinx_zcu102-xilinx-linux/device-tree/xilinx-v2023.2+gitAUTOINC+1a5881d004-r0/pl-delete-nodes-zynqmp-zcu102-rev10-adrv9025.dtsi:9.15-49 Label or path rx_adrv9026_tpl_core_adc_tpl_core not found
Error: /home/xltn/petalinux/2023.2/xilinx-zcu102-2023.2/build/tmp/work/xilinx_zcu102-xilinx-linux/device-tree/xilinx-v2023.2+gitAUTOINC+1a5881d004-r0/pl-delete-nodes-zynqmp-zcu102-rev10-adrv9025.dtsi:10.15-49 Label or path tx_adrv9026_tpl_core_dac_tpl_core not found
FATAL ERROR: Syntax error parsing input tree

[log.txt](https://github.com/analogdevicesinc/meta-adi/files/15297178/log.txt)

nunojsa, 2024-05-21:
Hmm, this seems to be a valid issue... I'll try to look at it later today. This also reminded me that this project is not in my CI and hence why I've not seen this before...

Sorry for the delay (just got back from vacations)

ducanh1699, 2024-05-21:
Thank you for your reply. I look forward to the solution from you.

nunojsa, 2024-05-21:
So, I'm not really having any issue... Are you sure you're using the latest hdl from [here](https://github.com/analogdevicesinc/hdl/tree/main/projects/adrv9026/zcu102)?

ducanh1699, 2024-05-21:
I have this error when I build it with the bsp zcu102 from here https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-design-tools.html

ducanh1699, 2024-05-21:
My step build project is:
1 petalinux-create -t project -s xilinx-zcu102-v2023.2-10140544.bsp
2 petalinux-config ( add user layer)
3 echo "KERNEL_DTB=\"${zynqmp-zcu102-rev10-adrv9025}\"" >> project-spec/meta-user/conf/petalinuxbsp.conf
4 petalinux-build
Then I have the error above

nunojsa, 2024-05-22:
> I have this error when I build it with the bsp zcu102 from here

Note that using bsp was never tested... So what system-top.xsa are you using? The one that comes with the bsp? That won't work...

> echo "KERNEL_DTB="${zynqmp-zcu102-rev10-adrv9025}"" >> project-spec/meta-user/conf/petalinuxbsp.conf

Run it like this `echo "KERNEL_DTB=\"zynqmp-zcu102-rev10-adrv9025\""`

nunojsa, 2024-06-20:
closing this due to inactivity...

closed 2024-06-20 by nunojsa