analogdevicesinc/meta-adi #151: Make the analog password for root an opt-out instead of forced override.

sumguytho, 2024-08-18:
This is a proposal and not a bug report, posting it here since I couldn't find a more fitting place. The petalinux-image-minimal.bbappend overrides EXTRA_USERS_PARAMS set by petalinux.conf, as is stated by the comments but why does it have to override it? Even if this has to be this way to keep compatibility there are ways to make this an opt-out rather than a silent override:

One can substitute assignment with appending like so:
```
EXTRA_USERS_PARAMS =+ "	\
	useradd -p '\$6\$xx\$OCk/lHkXahf1zu7kG4wzEic75NlaPVNtK8uwW3Ytjas229MmjVA.x/WFjQMIOFrlO.OQUc0KGyVzr8h3nwfWi1' analog; \
	usermod -p '\$6\$xx\$OCk/lHkXahf1zu7kG4wzEic75NlaPVNtK8uwW3Ytjas229MmjVA.x/WFjQMIOFrlO.OQUc0KGyVzr8h3nwfWi1' root; \
	usermod -a -G audio analog; \
	usermod -a -G video analog; \
	groupadd -r aie; \
	usermod -a -G aie analog; \
"
EXTRA_USERS_SUDOERS =+ "analog ALL=(ALL) ALL;"
```
This way the default behaviour is unchanged and the EXTRA_USERS_PARAMS set by petalinux-config is no longer overridden entirely. One can still change root password directly if necessary.

Or better yet, this behaviour can be made toggleable by using overrides:
```
EXTRA_USERS_PARAMS:prepend:xilinx-kuiper-compat = "	\
	useradd -p '\$6\$xx\$OCk/lHkXahf1zu7kG4wzEic75NlaPVNtK8uwW3Ytjas229MmjVA.x/WFjQMIOFrlO.OQUc0KGyVzr8h3nwfWi1' analog; \
	usermod -p '\$6\$xx\$OCk/lHkXahf1zu7kG4wzEic75NlaPVNtK8uwW3Ytjas229MmjVA.x/WFjQMIOFrlO.OQUc0KGyVzr8h3nwfWi1' root; \
	usermod -a -G audio analog; \
	usermod -a -G video analog; \
	groupadd -r aie; \
	usermod -a -G aie analog; \
"
EXTRA_USERS_SUDOERS:prepend:xilinx-kuiper-compat = "analog ALL=(ALL) ALL;"
```

Since yocto's overrides is a well known torture mechanism I'll [leave this here](https://docs.mulgara.org/images/4/45/004-1325-SLIDES-demystifying_the_overrides_mechanism_and_bitbake_operators.pdf) if anyone wants to try understand this. This approach also implies changes in source code for petalinux toolchain. There will have to be an option to enable kuiper compatibility in petalinux-config ncurses menu, when enabled it will add this line to petalinux.conf:
```
OVERRIDES .= ":xilinx-kuiper-compat"
```

It will be on by default so that the default behaviour is unchanged and disabling this option will prevent this bbappend from changing aforementioned variables in petalinux-image-minimal.

The reason for this proposal is that as someone who knows yocto and has never heard of kuiper it seems rather bizarre that the default password for user root is changed without any notice while also silently breaking the ability to add users in petalinux-config. But then again, I have no idea how applicable this is since I don't know the historical reasons that drive this password change in the first place.

nunojsa, 2024-09-04:
> This way the default behaviour is unchanged and the EXTRA_USERS_PARAMS set by petalinux-config is no longer overridden entirely. One can still change root password directly if necessary.

Well, but I would argue that the default behavior when meta-adi is inclueded is having the analog user and password for root (so it's the same behavior as kuiper in the sd card release). Now, I do understand it's cumbersome not being able to change it using the normal petalinux interfaces. So having more than one user by default might be acceptable (need to sleep over it)...

>  This approach also implies changes in source code for petalinux toolchain

And that is a no go for me as I'm not keen on the effort of supporting this directly in petalinux. It would only be sane if upstreamed to AMD and I have doubts something so specific to ADI like this would be accepted at all...

sumguytho, 2024-09-13:
> Well, but I would argue that the default behavior when meta-adi is inclueded is having the analog user and password for root (so it's the same behavior as kuiper in the sd card release).

I have never heard of kuiper so, naturally, I have no idea why the behaviour should remain in line with it.

I'm looking for a way to reverse this behaviour and as it is it's impossible without resorting to nasty stuff like patching the meta-adi layer or adding a placeholder recipe that runs a script at launch that reverses this. The reason for this is that the EXTRA_USERS_PARAMS varible is modified in bbappend with priority higher than user layer in petalinux which is also the last thing to be parsed meaning it can't be changed. I gotta say, I'm not the only one who came across this. I've seen at least a couple forum posts employing similar solutions.

I can think of another way to make default behaviour suppressable. Make a separate variable for usermod commands in bbappend for image and prepend it to EXTRA_USERS_PARAMS like so:

```
KUIPER_COMPAT_USERADD ??= "	\
	useradd -p '\$6\$xx\$OCk/lHkXahf1zu7kG4wzEic75NlaPVNtK8uwW3Ytjas229MmjVA.x/WFjQMIOFrlO.OQUc0KGyVzr8h3nwfWi1' analog; \
	usermod -p '\$6\$xx\$OCk/lHkXahf1zu7kG4wzEic75NlaPVNtK8uwW3Ytjas229MmjVA.x/WFjQMIOFrlO.OQUc0KGyVzr8h3nwfWi1' root; \
	usermod -a -G audio analog; \
	usermod -a -G video analog; \
	groupadd -r aie; \
	usermod -a -G aie analog; \
"
KUIPER_COMPAT_SUDOERS ??= "analog ALL=(ALL) ALL;"
EXTRA_USERS_PARAMS:prepend = "${KUIPER_COMPAT_USERADD}"
EXTRA_USERS_SUDOERS:prepend = "${KUIPER_COMPAT_SUDOERS}"
```

This way the default behaviour can be reversed by setting KUIPER_COMPAT_USERADD and KUIPER_COMPAT_SUDOERS to "" in petalinuxbsp.conf. Does this look acceptable?

Sorry for late response.

nunojsa, 2025-01-14:
> I have never heard of kuiper so, naturally, I have no idea why the behaviour should remain in line with it. 

It's basically a raspbian based image with support for ADI related projects and EVBs (same as meta-adi but more complete). We try to provide, as much as possible, the same experience between the images.

> This way the default behaviour can be reversed by setting KUIPER_COMPAT_USERADD and KUIPER_COMPAT_SUDOERS to "" in petalinuxbsp.conf. Does this look acceptable?

Hmm, like this we also end up with multiple users right? Default ADI user plus petalinux one? Anyways, I do prefer the second option over the first one. Feel free to send a PR for supporting this.

Sorry for __very__ late reply!

sumguytho, 2025-01-15:
> It's basically a raspbian based image with support for ADI related projects and EVBs (same as meta-adi but more complete). We try to provide, as much as possible, the same experience between the images.

I see. I would argue that yocto is a very complicated technology and anyone who goes as far as to use it does so to make everything cater to their needs specifically, like create different users, tweak kernel options, tweak system configuration in other ways. People who just want to tinker with ADI will probably stick to kuiper.

> Hmm, like this we also end up with multiple users right? Default ADI user plus petalinux one?

Yes. I'm not sure there is a petalinux user at all with meta-adi prior to this change since it just overwrites the command used to create user petalinux. Now, the changes made by user through `petalinux-config -c rootfs` are no longer overwritten, instead the changes made by this layer are always applied over whatever the user has changed, possibly overriding user configuration.

nunojsa, 2025-01-16:
> see. I would argue that yocto is a very complicated technology and anyone who goes as far as to use it does so to make everything cater to their needs specifically, like create different users, [...]

Agreed. Exactly why I'm not opposing to this :)

> Yes. I'm not sure there is a petalinux user at all with meta-adi prior to this change since it just overwrites the command used to create user petalinux

Yes, it was just overwritten...

nunojsa, 2025-01-21:
Solved by #159

closed 2025-01-24 by nunojsa