analogdevicesinc/meta-adi #153: A KBUILD_DEFCONFIG 'adi_zynqmp_defconfig' was specified ERROR while building petalinux

mbaykenar, 2024-09-03:
**Environment (please complete the following information):**
 - Project to build: zynqmp-zcu102-rev10-ad9361-fmcomms2-3
 - Petalinux version: 2019.2
 - HDL branch: 2019_R2
 - META-ADI branch: 2019_R2
 - Using meta-adi petalinux docker image? Not sure

**Describe the bug**
I have generated an xsa file from Vivado, having custom modules on fmcomms3 project of ADI HDL. Before adding Analog Devices Yocto layers, petalinux project is built successfully. Then I followed these instructions from 2019_R2 branch of meta-adi:

```
source  <path-to-installed-PetaLinux>/settings.sh
petalinux-create -t project --template <PLATFORM> --name <PROJECT_NAME>, where <PLATFORM> is:
	* zynqMP (for UltraScale + MPSoC)
	* zynq (for Zynq)
	* microblaze (for MicroBlaze)
git clone https://github.com/analogdevicesinc/meta-adi.git
cd <path-to-project>
# Check the Building HDL link below
petalinux-config --get-hw-description=<path to hdf file>
# Select the devicetree that fits the project being built
echo "KERNEL_DTB=${dts_to_use}" >> project-spec/meta-user/conf/petalinuxbsp.conf
cd build
petalinux-build
```

However, I got an error, the log says:

```
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['/home/aykenar/opt/pkg/petalinux/2019.2/components/yocto/source/aarch64/layers/core/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot']
NOTE: Installed into sysroot: ['kern-tools-native']
NOTE: Skipping as already exists in sysroot: ['quilt-native']
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing shell function do_kernel_metadata
ERROR: A KBUILD_DEFCONFIG 'adi_zynqmp_defconfig' was specified, but not present in the source tree
WARNING: /home/aykenar/Desktop/PetalinuxProjects/xilinx-zcu102-2019.2/build/tmp/work/zcu102_zynqmp-xilinx-linux/linux-xlnx/4.19-adi_2019_R2+git999-r0/temp/run.do_kernel_metadata.13625:1 exit 1 from 'exit 1'
ERROR: Function failed: do_kernel_metadata (log file is located at /home/aykenar/Desktop/PetalinuxProjects/xilinx-zcu102-2019.2/build/tmp/work/zcu102_zynqmp-xilinx-linux/linux-xlnx/4.19-adi_2019_R2+git999-r0/temp/log.do_kernel_metadata.13625)
```

I also get the same error petalinux-config -c kernel.

Regards,

nunojsa, 2024-09-04:
Hi,

You get the error when building? That is odd since the file does exist in the [2019_R2 branch](https://github.com/analogdevicesinc/linux/tree/2019_R2/arch/arm64/configs). Can you run `petalinux-build -c kernel -x devshell` and see if the file is in there (I think this is the right command from the top of my head)?

Also note that you're running a very old release (not supported anymore) so while I'll try to support you as best as I can, I won't/can't go that far (as building/trying that project myself).

mbaykenar, 2024-09-04:
Thanks @nunojsa 

I have to use this old branch since the project I am working on already using this one.

Another note is that when I first add AD Yocto layers in configuration and then try to build this time I get these errors:

```
Error: /home/aykenar/Desktop/connected/petalinuxproject/xilinx-zcu102-2019.2/build/tmp/work/zcu102_zynqmp-xilinx-linux/device-tree/xilinx-v2019.2+gitAUTOINC+a8b39cf536-r0/pl-delete-nodes-zynqmp-zcu102-rev10-ad9361-fmcomms2-3.dtsi:6.15-28 Label or path psu_ctrl_ipi not found
Error: /home/aykenar/Desktop/connected/petalinuxproject/xilinx-zcu102-2019.2/build/tmp/work/zcu102_zynqmp-xilinx-linux/device-tree/xilinx-v2019.2+gitAUTOINC+a8b39cf536-r0/pl-delete-nodes-zynqmp-zcu102-rev10-ad9361-fmcomms2-3.dtsi:7.15-35 Label or path psu_message_buffers not found
FATAL ERROR: Syntax error parsing input tree
```

and 

```
In file included from /home/aykenar/Desktop/connected/petalinuxproject/xilinx-zcu102-2019.2/components/plnx_workspace/sources/linux-xlnx/arch/arm64/boot/dts/xilinx/zynqmp-zcu102-revB.dts:10,
                 from /home/aykenar/Desktop/connected/petalinuxproject/xilinx-zcu102-2019.2/components/plnx_workspace/sources/linux-xlnx/arch/arm64/boot/dts/xilinx/zynqmp-zcu102-rev1.0.dts:10:
/home/aykenar/Desktop/connected/petalinuxproject/xilinx-zcu102-2019.2/components/plnx_workspace/sources/linux-xlnx/arch/arm64/boot/dts/xilinx/zynqmp-zcu102-revA.dts:11:10: fatal error: pl.dtsi: No such file or directory
 #include "pl.dtsi"
          ^~~~~~~~~
compilation terminated.
```

I tried petalinux-build -c kernel -x devshell and it takes too long time, more than 1 hour, I realized it opened another terminal tab and when I close the other tab build process finished. The other tab was:

~/Desktop/connected/petalinuxproject/xilinx-zcu102-2019.2/components/plnx_workspace/sources/linux-xlnx# 

The log file says

```
NOTE: Resolving any missing task queue dependencies
Sstate summary: Wanted 4 Found 0 Missed 8 Current 45 (0% match, 91% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Running task 372 of 372 (/home/aykenar/opt/pkg/petalinux/2019.2/components/yocto/source/aarch64/layers/meta-xilinx/meta-xilinx-bsp/recipes-kernel/linux/linux-xlnx_2019.2.bb:do_devshell)
NOTE: recipe linux-xlnx-4.19-adi_2019_R2+git999-r0: task do_devshell: Started
NOTE: recipe linux-xlnx-4.19-adi_2019_R2+git999-r0: task do_devshell: Succeeded
NOTE: Tasks Summary: Attempted 372 tasks of which 371 didn't need to be rerun and all succeeded.
[INFO] successfully built kernel
```
But when I check images/linux directory there is only system.bit file generated.

Now I restarted the process and this time I will wait and do not close other terminal tab to see what's gonna happen

Regards,

nunojsa, 2024-09-04:
> I have to use this old branch since the project I am working on already using this one.

I see but still note that support will be very limited then...

> Another note is that when I first add AD Yocto layers in configuration and then try to build this time I get these errors:

It seems those  IPs were removed from the project so it should be just a matter of removing those lines from the .dtsi file.

> I tried petalinux-build -c kernel -x devshell and it takes too long time, more than 1 hour, I realized it opened another terminal tab and when I close the other tab build process finished. The other tab was:

That's odd. It looks like there are some issues with your build environment. 

I would suggest to do a clean build...

nunojsa, 2024-09-05:
The last part is odd. Anyways, glad it worked out... Closing it for now

closed 2024-09-05 by nunojsa