analogdevicesinc/meta-adi #154: How to avoid do_sanity_check() error using a custom devicetree which is not supported by default in meta-adi?

DarrenChengdu, 2024-09-13:
- Project to build: Custom zc706 carrier board with an adrv9009 FMC
 - Petalinux version: 2018.3
 - HDL branch: 2019.1
 - META-ADI branch: 2019.1
 - Using meta-adi petalinux docker image? Not clear! Does it means adding meta-adi-core and meta-adi-xilinx folds path in petalinux yocto settings?

**Describe the bug**
I  copied zynq-zc706-adv7511-adrv9009.dts  as zynq-customboard-adrv9009.dts under <path-to-linux-2019_R1>/arch/arm/boot/dts, and  copied pl-delete-nodes-zynq-zc706-adv7511-adrv9009.dtsi as pl-delete-nodes-zynq-customboard-adrv9009.dtsi under <path-to-meta-adi>/meta-adi-xilinx/recipes-bsp/device-tree/files. I deleted all hdmi related nodes in pl-delete-nodes-zynq-customboard-adrv9009.dtsi because them didn't exist on the customboard. At the end of running petalinux-build, it faild to do sanity check.  How to set do_sanity_check  to be false?
[image]

nunojsa, 2024-09-20:
Hi,

You need to properly add your devicetree into the list:

https://github.com/analogdevicesinc/meta-adi/blob/2019_R1/meta-adi-xilinx/recipes-bsp/device-tree/device-tree.bbappend#L99

nunojsa, 2024-10-10:
Closing due to inactivity