analogdevicesinc/meta-adi #156: analogdevicesinc/linux updated to 6.6 causing meta adi to not build correctly in 2023.1

Jcrash29, 2024-12-05:
When attempting to build I get the following error:

```
ERROR: linux-xlnx-6.1-adi-main+gitAUTOINC+machine-r0 do_kernel_version_sanity_check: Package Version (6.1-adi-main+gitAUTOINC+machine) does not match of kernel being built (6.6). Please update the PV variable to match the kernel source or set KERNEL_VERSION_SANITY_SKIP="1" in your recipe.
ERROR: linux-xlnx-6.1-adi-main+gitAUTOINC+machine-r0 do_kernel_version_sanity_check: ExecutionError('/home/pdw/2023.1/petalinux/build/tmp/work/zynqmp_generic_xczu15eg-xilinx-linux/linux-xlnx/6.1-adi-main+gitAUTOINC+machine-r0/temp/run.do_kernel_version_sanity_check.12387', 1, None, None)
ERROR: Logfile of failure stored in: /home/pdw/2023.1/petalinux/build/tmp/work/zynqmp_generic_xczu15eg-xilinx-linux/linux-xlnx/6.1-adi-main+gitAUTOINC+machine-r0/temp/log.do_kernel_version_sanity_check.12387
ERROR: Task (/home/pdw/2023.1/petalinux/components/yocto/layers/meta-xilinx/meta-xilinx-core/recipes-kernel/linux/linux-xlnx_2023.1.bb:do_kernel_version_sanity_check) failed with exit code '1'
```

I believe this is due to  a recent update to analogdevicesinc/linux

pgsys, 2024-12-05:
I get the same problem with petalinux 2023.2.  Looks like analogdevicesinc/linux moved to 2024.1?

Jcrash29, 2024-12-05:
@pgsys The fix I used:

Modify `meta-adi-xilinx/recipes-kernel/linux/linux-xlnx_%.bbappend`

Change the KBRANCH to something other than main.

In my case I used `adi-6.1.0`

Good luck!

nunojsa, 2024-12-09:
This should be fixed by #157

nunojsa, 2025-01-14:
This should now be fixed

closed 2025-01-14 by nunojsa