analogdevicesinc/meta-adi #162: 2023_R2 - libiio SRCREV Incorrect

BrentK-ADI, 2025-03-06:
The explicit SRCREV in libiio_%.bbappend for 2023_R2 is an invalid hash for the libiio repo.  This is causing build failures when attempting to build a Petalinux project in offline mode. https://github.com/analogdevicesinc/meta-adi/blob/dd884c9265966e970c6525d322f0986a96ecc203/meta-adi-xilinx/recipes-support/libiio/libiio_%25.bbappend#L3

It looks like the hash specified (54eb23f) is actually a hash for ADI's Linux repo. https://github.com/analogdevicesinc/linux/commit/54eb23f4b5c6093916f208772627f7b68f495559

Please update the bbappend file with the correct hash for 2023_R2 offline builds.

nunojsa, 2025-03-07:
> The explicit SRCREV in libiio_%.bbappend for 2023_R2 is an invalid hash for the libiio repo. This is causing build failures when attempting to build a Petalinux project in offline mode.

Ups... Likely copy past error.

nunojsa, 2025-03-18:
fixed by #163

PR #163 (2025-03-11): meta-adi-xilinx: libiio: fix offline build
The offline git hash for libiio was wrong and did not existed. Fix it...

Should fix #162
closed 2025-03-18 by nunojsa